liquidctl/liquidctl #349: Smart Device v2 Change speed

ARKseal, 2021-06-12:
I'm trying to setup my fan curves that are connected to my NZXT Smart Device but it doesn't seem to work.
OS: Ubuntu 20.04
Command: sudo liquidctl set sync speed 20 30  30 50  34 80  40 90  50 100
Response: ERROR: Error: operation not supported by NZXT Smart Device V2

jonasmalacofilho, 2021-06-12:
Devices like the Smart Device do not natively support speed profiles.  This is mostly because they have no temperature sensors they can access.

Even though only support static fan speeds are supported at the hardware level, it's possible to have a program or script that implements the profile in software on your host's CPU, by monitoring the desired sensors and constantly sending adjusted speeds to the device.

You can make your own with bash or Python (in the latter case, using the liquidctl Python APIs), or play with the experimental [`yoda`](https://github.com/liquidctl/liquidctl/blob/main/extra/yoda) script that lives in `extra/`.

closed 2021-06-13 by jonasmalacofilho

ARKseal, 2021-06-17:
If anyone has the same question, here's my code which I run as a service:

`#!/usr/bin/python3

import signal

is_shutdown = False

def stop(sig, frame):
  print("SIGTERM")
  global is_shutdown
  is_shutdown = True

def ignore(sig, frsma):
  print("SIGHUP")

signal.signal(signal.SIGTERM, stop)
signal.signal(signal.SIGHUP, ignore)

if __name__ == '__main__':

    from time import sleep
    from math import ceil

    try:
        from psutil import sensors_temperatures
        from liquidctl import find_liquidctl_devices
    except:
        print("psutil and/or liquidctl are not installed, please install it using 'pip install psutil' and 'pip install liquidctl'")
        exit(-1)

    def change_temp(dev, speed):
        with dev.connect():
            dev.set_fixed_speed('sync', speed)

    temp_to_speed = {
        20:  40,
        25:  40,
        30:  40,
        35:  40,
        40:  40,
        45:  40,
        50:  40,
        55:  80,
        60: 100,
        65: 100,
        70: 100,
        75: 100,
        80: 100,
        85: 100,
        90: 100,
        95: 100
    }

    # get all devices
    devices = find_liquidctl_devices()

    smart_device_v2 = None
    curr_speed = None

    for dev in devices:

        # conenct to device
        with dev.connect():

            # initialize the device after bootup
            init_status = dev.initialize()

            # check if the current device is the Smart Device V2
            if 'Smart Device V2' in dev.description:
                smart_device_v2 = dev
    
    # start loop
    while not is_shutdown:
        
        # get temps
        temps = sensors_temperatures()

        #get temp for the sensor named CPUTIN
        cpu_temp = [temp.current for temp in temps['nct6797'] if temp.label == 'CPUTIN'][0]

        # round to nearest 5
        cpu_temp_rounded = 5 * ceil(cpu_temp/5)

        # check if current speed is the same speed as the one to change to
        if curr_speed != temp_to_speed[cpu_temp_rounded]:
            change_temp(smart_device_v2, temp_to_speed[cpu_temp_rounded])
            curr_speed = temp_to_speed[cpu_temp_rounded]

        sleep(1)`

jonasmalacofilho, 2021-06-17:
I'm curious, did `yoda` not work properly for you?