liquidctl/liquidctl #350: kraken: fan speed stays constant no matter the temperature

tcn, 2021-06-13:
liquidctl v1.6.1 (Arch Linux; 1.6.1-1), Kraken X62v2.

The fan (and possibly pump) speed simply doesn't respond to temperature but stays constant.

Debug output is all looking fine but still temps approach max within seconds with the fan staying at 50% (Ryzen, narrow temp range, idle is basically in the 30+ °C, max is 60):

`# liquidctl --debug --match kraken set fan speed 30 50  40 60  50 75  55 100 2>&1 | grep INFO
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 50% for liquid temperature >= 20°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 50% for liquid temperature >= 21°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 50% for liquid temperature >= 22°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 50% for liquid temperature >= 23°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 50% for liquid temperature >= 24°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 50% for liquid temperature >= 25°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 50% for liquid temperature >= 26°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 50% for liquid temperature >= 27°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 50% for liquid temperature >= 28°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 50% for liquid temperature >= 29°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 50% for liquid temperature >= 30°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 51% for liquid temperature >= 31°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 52% for liquid temperature >= 32°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 53% for liquid temperature >= 33°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 54% for liquid temperature >= 34°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 55% for liquid temperature >= 35°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 56% for liquid temperature >= 36°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 57% for liquid temperature >= 37°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 58% for liquid temperature >= 38°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 59% for liquid temperature >= 39°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 60% for liquid temperature >= 40°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 62% for liquid temperature >= 41°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 63% for liquid temperature >= 42°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 64% for liquid temperature >= 43°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 66% for liquid temperature >= 44°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 68% for liquid temperature >= 45°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 69% for liquid temperature >= 46°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 70% for liquid temperature >= 47°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 72% for liquid temperature >= 48°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 74% for liquid temperature >= 49°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 75% for liquid temperature >= 50°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 85% for liquid temperature >= 52°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 95% for liquid temperature >= 54°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 100% for liquid temperature >= 56°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 100% for liquid temperature >= 58°C
[INFO] liquidctl.driver.kraken2: setting fan PWM duty to 100% for liquid temperature >= 60°C
`

Any idea? Any advise how to further debug this?

jonasmalacofilho, 2021-06-14:
The only sensor that is supported by that device is its internal coolant temperature probe.  The fact that the CPU temperature reaches 60°C under load doesn't appear inherently unexpected to me.

Try to monitor the liquid temperature under sustained load and high power output, you may be able to make your fan and pump curves a bit more aggressive.  I imagine your liquid temperature never goes above 45°C.

But if even after the fluid soaks some heat your fans still stay at ~50%, please send me the complete output of `liquidctl --debug status`, plus the rest of the output from your previous message (without limiting it to INFO level).

tcn, 2021-06-14:
Ugh, alright, the temperature is the one of the coolant not the CPU's one.
Makes actually sense -.- My bad, works fine.
Thanks for the quick response, case closed.

closed 2021-06-14 by tcn