liquidctl/liquidctl #355: Fan/pump/LED settings not saved to flash/nonvol on EVGA CLC 280

genpfault, 2021-06-25:
Observed behavior/repro instructions:

1. Set fan/pump/LED settings with liquidctl on an EVGA CLC 280, for instance:
    ```
    liquidctl initialize
    liquidctl set logo color blackout
    liquidctl set fan speed 70
    ```
2. Verify new settings are applied.
3. Shut down host machine.
4. Power host machine back on.
5. Observe that the fan/pump/LED settings have reverted to the last ones set by EVGA Flow Control.

Expected behavior:

1. Fan/pump/LED settings are saved to the AIO's onboard flash/nonvol and are restored on next power-up without using liquidctl to re-set them.

Settings set by EVGA's Flow Control software (I used the current v2.0.11) are saved to the device and survive power-cycling.

Hardware list (snipped unrelated HID devices):
```
C:\liquidctl-1.6.1-bin-windows-x86_64>liquidctl --debug list
[DEBUG] __main__: running liquidctl v1.6.1 (AppVeyor; liquidctl-1.6.1-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver, CorsairAsetekProDriver)
[DEBUG] liquidctl.driver.usb: found USB device 2433:b200
[DEBUG] liquidctl.driver.usb: instanced driver for Asetek 690LC (assuming EVGA CLC)
Device #0: Asetek 690LC (assuming EVGA CLC)
├── Vendor ID: 0x2433
├── Product ID: 0xb200
├── Release number: 0x0100
├── Serial number: CCVI_1.0
├── Bus: usb1
├── Address: 9
├── Port: 7
└── Driver: Modern690Lc
[DEBUG] __main__: hierarchy: _CommonAsetekDriver, UsbDriver, BaseUsbDriver, BaseDriver
```

Firmware version:

```
C:\liquidctl-1.6.1-bin-windows-x86_64>liquidctl status
Asetek 690LC (assuming EVGA CLC)
├── Liquid temperature        27.1  °C
├── Fan speed                 1500  rpm
├── Pump speed                2760  rpm
└── Firmware version      2.10.0.0
```

Additional information about the system:

- Affected device: EVGA CLC 280
- Operating system: Windows 10 64-bit & Kubuntu 20.04
- Liquidctl version: 1.6.1

jonasmalacofilho, 2021-06-25:
I think Flow Control wasn't able to save the settings to non-volatile storage when I originally figured out the protocol and implemented that driver.

I don't think I'll be able to work on this for at least a few weeks.  Any change you can give it a try?  There is some useful documentation in [`docs/developer/`](https://github.com/liquidctl/liquidctl/tree/main/docs/developer), and you can also ask me questions (here, via email or on Discord).

P.S. Thanks for the detailed bug report.

genpfault, 2021-06-26:
PROTIP: The Zadig driver replacement breaks the EVGA Flow Control app.  [Uninstall the 280 from Device Manager (check "Delete the driver software for this device") and unplug/replug it to get things back to normal](https://github.com/pbatard/libwdi/wiki/FAQ#help-zadig-replaced-the-driver-for-the-wrong-device-how-do-i-restore-it).

genpfault, 2021-06-26:
OK, looking at [the capture](https://github.com/liquidctl/liquidctl/files/6719482/flow-control.pcapng.gz) it seems the Flow Control app sends out a single `0x21` ("[`_CMD_STORE_SETTINGS`](https://github.com/liquidctl/liquidctl/blob/9410125129f73febe37a287897fdc872bd139fde/liquidctl/driver/asetek.py#L36)"?) message/command when you close the app.  Zero-length, other than the command byte:

```
USB URB
    [Source: host]
    [Destination: 1.9.2]
    USBPcap pseudoheader length: 27
    IRP ID: 0xffff9e8ddf56da20
    IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)
    URB Function: URB_FUNCTION_BULK_OR_INTERRUPT_TRANSFER (0x0009)
    IRP information: 0x00, Direction: FDO -> PDO
    URB bus id: 1
    Device address: 9
    Endpoint: 0x02, Direction: OUT
        0... .... = Direction: OUT (0)
        .... 0010 = Endpoint number: 2
    URB transfer type: URB_BULK (0x03)
    Packet Data Length: 1
    [Response in: 42912]
    [bInterfaceClass: Vendor Specific (0xff)]
Leftover Capture Data: 21
```

Probably don't want to emit that after *every* set-command to avoid burning through flash write-cycles...maybe a new zero-argument `save`/`store` verb?  Ex:

```
liquidctl store
```

jonasmalacofilho, 2021-06-26:
> OK, looking at the capture it seems the Flow Control app sends out a single 0x21 ("_CMD_STORE_SETTINGS"?) message/command when you close the app. Zero-length, other than the command byte:

Ok, that makes sense.

> Probably don't want to emit that after every set-command to avoid burning through flash write-cycles...maybe a new zero-argument save/store verb?

Instead of a new command, I think you can reuse the `--non-volatile` flag to allow `initialize()`, `set_speed_profile()`, `set_fixed_speed()` and `set_color()` to optionally persist the settings before they return.

The fact that all state will be persisted (not just what was passed to the specific command/function called) is indeed a bit weird, but something that the documentation can clear up.  The documentation should also explicitly hint that it's more efficient to set up almost everything without `--non-volatile`, and only use that flag when changing the very last (to be persisted) setting.

commit a01cabfa (2022-01-14): Asetek: Persist settings with --non-volatile

Fixes #355
closed 2022-01-14 by jonasmalacofilho