liquidctl/liquidctl #359: Release detected as malware by Windows defender

ZRRero, 2021-07-16:
When downloaded the release file for version `1.7.1` it was detected by Windows Defender, uploading the .zip to VirusTotal also raises alarms on malware.
[image]

jonasmalacofilho, 2021-07-16:
Unless AppVeyor has been pwned, those are false positives.

I'll get in touch with MS... last time they resolved it fairly quickly.

jonasmalacofilho, 2021-07-17:
I have reported this as a possible false positive to MS.  While I wait to hear back from them, let me document a bit of how our Windows build works.

Our `liquidctl.exe` executable for Windows, and the associated `.zip` file that is part of a release, are not manually built.  They are built on [AppVeyor](https://www.appveyor.com/),<sup>1</sup> a continuous integration service that supports Windows and is free for open-source projects.  [Build history and logs][appveyor-history] are publicly accessible.

The build recipe is public – [`./appveyor.yml`] – and, apart from Microsoft Windows and Microsoft MSVC, we only use open-source components.  Most things are also build from source, with the exceptions being things like the libusb DLL; that one we take from their official release package.

After the executable and ZIP package are built, they are checksumed on AppVeyor, and manually verified for correct functionality by me.  I then sign the ZIP package with my PGP key, and publish both as part of a release.  The checksums generated on AppVeyor are part of build log.

_<sup>1</sup> We will probably migrate from AppVeyor to GitHub Actions at some point; we already prefer it the latter the Linux and Mac tests._  

[`./appveyor.yml`]: https://github.com/liquidctl/liquidctl/blob/main/appveyor.yml
[appveyor-history]: https://ci.appveyor.com/project/jonasmalacofilho/liquidctl/history

---

_Technical notes_

Detected by Defender as [`Program:Win32/Wacapew.C!ml`] (on VirusTotal) or [`Trojan:Win32/Wacatac.B!ml`] (locally).  Technical details are not available.  Local definition version is `1.343.1162.0`.

Reported to Microsoft as a possible false positive:

> This executable is publicly built on AppVeyor from open-source components.[1]  While certain DLLs are used as provided by their corresponding projects, they are fairly common (like libusb) and, as far as I know, they have not been identified as malicious by any AV software.  Additionally, the executable is checksumed when built on AppVeyor, and has not been modified after that.
> 
> Based on these points, I believe it is being incorrectly flagged, possibly because it is a Python package bundled into an exe with PyInstaller.  Still, I am available to answer any questions and assist in any way that might be helpful.
> 
> [1] https://ci.appveyor.com/project/jonasmalacofilho/liquidctl/builds/40015929

Submission ID: [`c405f4c5-b767-4f1d-8eef-a59a1f7d6155`] (restricted)

Other links:

 - [VirusTotal `liquidctl-1.7.1-bin-windows-x86_64.zip` (hash `5f35d4ac8ad6...`)](https://www.virustotal.com/gui/file/5f35d4ac8ad6da374877d17c7a36bbb202b0a74bd773ebe45444f0089daba27b/detection)
 - [VirusTotal `liquidctl.exe` (v1.7.1, hash `487759d4f0d3...`)](https://www.virustotal.com/gui/file/487759d4f0d3dc4291d056cada8474a301376808d19c5582e1cadab032042c39/detection)
 - [Log for AppVeyor build #1483 (v1.7.1)](https://ci.appveyor.com/project/jonasmalacofilho/liquidctl/builds/40015929)

P.S. In the future, it would be helpful if the Defender definition version was included in reports like these ([instructions from MS][definition-version]).  When reporting a possible false positive to MS, I am highly encouraged to supply that information.

[`Program:Win32/Wacapew.C!ml`]: https://www.microsoft.com/en-us/wdsi/threats/malware-encyclopedia-description?Name=Program:Win32/Wacapew.C!ml
[`Trojan:Win32/Wacatac.B!ml`]: https://www.microsoft.com/en-us/wdsi/threats/malware-encyclopedia-description?Name=Trojan:Win32/Wacatac.B!ml
[`c405f4c5-b767-4f1d-8eef-a59a1f7d6155`]: https://www.microsoft.com/en-us/wdsi/submission/c405f4c5-b767-4f1d-8eef-a59a1f7d6155
[definition-version]: https://docs.microsoft.com/en-us/microsoft-365/security/defender-endpoint/microsoft-defender-security-center-antivirus?view=o365-worldwide#definition-version

jonasmalacofilho, 2021-07-17:
Resolved:

> We have removed the detection.  Please follow the steps below to clear cached detection and obtain the latest malware definitions.
> 
>      1. Open command prompt as administrator and change directory to c:\Program Files\Windows Defender 
>      2. Run “MpCmdRun.exe -removedefinitions -dynamicsignatures”
>      3. Run "MpCmdRun.exe -SignatureUpdate"
> 
> Alternatively, the latest definition is available for download here: https://www.microsoft.com/en-us/wdsi/definitions
> 
> Thank you for contacting Microsoft.

closed 2021-07-17 by jonasmalacofilho