liquidctl/liquidctl #361: Error after upgrade to  `python-hidapi 0.10.1-1`

ti-mo, 2021-07-20:
Hi @jonasmalacofilho, thanks for the great tool! After upgrading my machine today, liquidctl no longer works. The existing install ( `1.6.1.r67.2268784-1`) throws the following error:

```
λ liquidctl initialize --match fan
AttributeError: module 'hid' has no attribute 'device'
```

Running the pkgbuild for `liquidctl-git 1.7.1.r2.f7ddd99-1` doesn't get very far through the test suite. One of the failures reads as follows:

```
========================================== FAILURES ==========================================
________________________________ test_modified_readme_example ________________________________

capsys = <_pytest.capture.CaptureFixture object at 0x7f8c0f009760>

    def test_modified_readme_example(capsys):
        from liquidctl import find_liquidctl_devices

        first = True

        # find all connected and supported devices on pseudo bus 'virtual'
        devices = find_liquidctl_devices(bus='virtual')

>       for dev in devices:

tests/test_api.py:33:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
liquidctl/driver/__init__.py:59: in find_liquidctl_devices
    for dev in bus_cls().find_devices(**kwargs):
liquidctl/driver/usb.py:526: in find_devices
    for handle in handles:
liquidctl/driver/usb.py:483: in enumerate
    yield cls(api, info)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <liquidctl.driver.usb.HidapiDevice object at 0x7f8c0f3419d0>
hidapi = <module 'hid' from '/usr/lib/python3.9/site-packages/hid/__init__.py'>
hidapi_dev_info = {'interface_number': 0, 'manufacturer_string': 'NZXT. - Inc.', 'path': b'/dev/hidraw1', 'product_id': 8201, ...}

    def __init__(self, hidapi, hidapi_dev_info):
        self.api = hidapi
        self.hidinfo = hidapi_dev_info
>       self.hiddev = self.api.device()
E       AttributeError: module 'hid' has no attribute 'device'

liquidctl/driver/usb.py:380: AttributeError
------------------------------------ Captured stdout call ------------------------------------
Virtual Bus Device (experimental) at virtual:virtual_address:
Firmware version: 3.14.16
Temperature: 30.4 °C
Fan control mode: VirtualControlMode.QUIET
Animation: None
Uptime: 18:23:12
Hardware mode: True
setting pump to radical red
```

Additional information about the system:

- Affected device: /
- Operating system: Arch Linux
- Liquidctl version: liquidctl-git 1.6.1.r67.2268784-1
- `python-hidapi 0.10.1-1`

Any clues as to what I might've done wrong? I've gone through the hidapi git history to see if I could find anything related, but I'm too unfamiliar with the library. Could this be a compatibillity issue?

jonasmalacofilho, 2021-07-20:
I seems that you have another library besides [`cython-hidapi`](https://github.com/trezor/cython-hidapi) ([`community/python-hidapi`](https://archlinux.org/packages/community/x86_64/python-hidapi/) in Arch) using the `hid` namespace, which creates a conflict and makes `liquidctl` pull a `hid` module that isn't what it expects.

The offending library is probably [`pyhidapi`](https://github.com/apmorton/pyhidapi) ([`community/python-hid`](https://archlinux.org/packages/community/any/python-hid/) in Arch), which has just recently been added to Arch without a necessary `conflicts=('python-hidapi')`.

(If you can,) try uninstalling `python-hid` and seeing if the problem is resolved (you may also need to reinstall `python-hidapi`).  Assuming this fixes the issue, can you report it on the Arch bug tracker?

ti-mo, 2021-07-20:
Thank you, that did it! `python-hid` is a dependency of [`community/qmk`](https://archlinux.org/packages/community/any/qmk/), so that's a bit of an issue. All 3 packages are maintained by Foxboron, that makes things easier. :slightly_smiling_face: I'll file a bug and I'll run qmk in a venv for now.

Thanks again for the help!

closed 2021-07-20 by ti-mo

jonasmalacofilho, 2021-07-20:
> Thank you, that did it! `python-hid` is a dependency of [`community/qmk`](https://archlinux.org/packages/community/any/qmk/), so that's a bit of an issue. All 3 packages are maintained by Foxboron, that makes things easier. slightly_smiling_face I'll file a bug and I'll run qmk in a venv for now.

It seems that qmk itself only uses `hid` in a couple of places,[1] and in one of those it looks like it actually expects cython-hidapi, not pyhidapi.[2]  So proposing a patch for qmk that drops the latter in favor of the former may be viable.

[1] https://github.com/search?l=Python&q=org%3Aqmk+hid&type=Code
[2] https://github.com/qmk/qmk_firmware/blob/9dea6f772077/keyboards/dekunukem/duckypad/keymaps/m4cs/sysinfo.py#L35

But let's see what Foxboron has to say first.

> Bug report here: https://bugs.archlinux.org/task/71555.

Thanks, I've already added my self to be notified on updates on that issue.