liquidctl/liquidctl #362: Systemctl giving errors

Lumene98, 2021-07-21:
I can't manage to make the systemctl service working, this is my service

```
[Unit]
Description=AIO startup service

[Service]
Type=oneshot
ExecStart=liquidctl initialize all
ExecStart=liquidctl --match kraken set pump speed 25 60 28 70 30 80
ExecStart=liquidctl --match kraken set fan speed  20 30  30 80  34 100

[Install]
WantedBy=default.target
```
And these are commands I run:

```
/usr/local/bin ⌚ 10:09:13
$ systemctl start liquidcfg
Job for liquidcfg.service failed because the control process exited with error code.
See "systemctl status liquidcfg.service" and "journalctl -xe" for details.

/usr/local/bin ⌚ 10:09:20
$ systemctl status liquidcfg.service
● liquidcfg.service - AIO startup service
     Loaded: loaded (/etc/systemd/system/liquidcfg.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2021-07-21 10:09:20 CEST; 10s ago
    Process: 4675 ExecStart=liquidctl initialize all (code=exited, status=1/FAILURE)
   Main PID: 4675 (code=exited, status=1/FAILURE)

lug 21 10:09:20 lumene-PC systemd[1]: Starting AIO startup service...
lug 21 10:09:20 lumene-PC liquidctl[4675]: Traceback (most recent call last):
lug 21 10:09:20 lumene-PC liquidctl[4675]:   File "/usr/local/bin/liquidctl", line 5, in <module>
lug 21 10:09:20 lumene-PC liquidctl[4675]:     from liquidctl.cli import main
lug 21 10:09:20 lumene-PC liquidctl[4675]: ModuleNotFoundError: No module named 'liquidctl'
lug 21 10:09:20 lumene-PC systemd[1]: liquidcfg.service: Main process exited, code=exited, status=1/FAILURE
lug 21 10:09:20 lumene-PC systemd[1]: liquidcfg.service: Failed with result 'exit-code'.
lug 21 10:09:20 lumene-PC systemd[1]: Failed to start AIO startup service.

```

It works normally outside of systemctl

Additional information about the system:

- Affected device: (e.g. Corsair Hydro H100i Pro XT) Kraken Z
- Operating system: (e.g. Arch Linux) Ubuntu 20.10
- Liquidctl version: (e.g. Arch Linux `liquidctl` package, version `1.4.2-2`) 1.7.1

MarshallAsch, 2021-07-21:
does any of the suggestions from this stack overflow question help? https://stackoverflow.com/questions/35641414/python-import-of-local-module-failing-when-run-as-systemd-systemctl-service

I have not had that error before, but I have seen it where the absolute path to `liquidctl` is needed (ie. `/bin/local/liquidctl` instead of just `liquidctl`, you can use the `which liquidctl` command to find the actual path).

Lumene98, 2021-08-06:
I thought it was fixed because if I do `systemctl start liquidcfg.service` it works fine but after `systemctl enable liquidcfg.service`, if I reboot my system it gives an error of insufficient permission

jonasmalacofilho, 2021-08-06:
The selected `User=<user>` doesn't have permission to talk to the devices.

Are you using udev rules to allow non-root access to them?  Maybe the issue is that when the service runs udev hasn't yet managed to apply those rules yet and, consequently, the devices are still only accessible by root.

---

More importantly, I would argue that your initial problem was that you hadn't been able to install liquidctl in a super system-friendly way.  It's in `/usr/local/bin/`, which normally isn't part of the `PATH`.

If you can, install it on `/usr/bin/` and it should™ just work© (with the default `User=root`).

Another alternative would be to include `/usr/local/bin` in the `PATH` for that service and/or for the root user.

By the way, it doesn't look like it, but if were using a Python virtual environment, a bit extra care would also be necessary to handle it.

Lumene98, 2021-08-06:
> The selected `User=<user>` doesn't have permission to talk to the devices. (EDIT: I'm assuming the permission error is related to the devices, but it would be better if you showed the exact error).
> 
> Are you using udev rules to allow non-root access to them? Maybe the issue is that when the service runs udev hasn't yet managed to apply those rules yet and, consequently, the devices are still only accessible by root.
> 
> More importantly, I would argue that your initial problem was that you hadn't been able to install liquidctl in a super system-friendly way. It's in `/usr/local/bin/`, which normally isn't part of the `PATH`.
> 
> If you can, install it on `/usr/bin/` and it should™ just work© (with the default `User=root`).
> 
> Another alternative would be to include `/usr/local/bin` in the `PATH` for that service and/or for the root user.
> 
> By the way, it doesn't look like it, but if were using a Python virtual environment, a bit extra care would also be necessary to handle it.

Yeah I realized it is actually in `/usr/local/bin`, can I just move it to `/usr/bin` or should I reinstall it?

jonasmalacofilho, 2021-08-06:
I think it's _probably_ fine to symlink (or move) to `/usr/bin/liquidctl`, but I could be missing something.

In the end, `/usr/[...]/bin/liquidctl` is itself just a script, automatically created by `setuptools` during the installation of liquidctl.  It finds and calls the CLI entry point from the liquidctl Python library/package.  I think this should just work, provided the liquidctl has been installed on the system Python environment, but there could be weird corner cases in more complex scenarios.

Lumene98, 2021-08-06:
> I think it's _probably_ fine to symlink (or move) to `/usr/bin/liquidctl`, but I could be missing something.
>
> In the end, `/usr/[...]/bin/liquidctl` is itself just a script, automatically created by [...]

I move liquidctl to `/usr/bin` and added `User=root`, but I'm getting the old `No module named 'liquidctl'`

jonasmalacofilho, 2021-08-06:
My bad, I shouldn't answer to issues while awake with insomnia.

---

In hindsight, the issue obviously wasn't with finding `liquidctl` in `/usr/bin/local/`.

The real problem probably has to do with:

> In the end, `/usr/[...]/bin/liquidctl` is itself just a script, automatically created by `setuptools` during the installation of liquidctl. It finds and calls the CLI entry point from the liquidctl [...]

Basically, the service can't find the root `liquidctl` Python _module._  How did you install liquidctl, do you remember?  And are you using your system `python3` or some other Python installation?

Lumene98, 2021-08-06:
> My bad, I shouldn't answer to issues while awake with insomnia.
> 
> (EDIT: I'm still doing it now, so take everything bellow with a grain of salt).
> 
> In hindsight, the issue obviously wasn't with finding `liquidctl` in `/usr/bin/local/`.
> 
> The real problem probably has to do with:
> 
> > In the end, `/usr/[...]/bin/liquidctl` is itself just a script, automatically created by `setuptools` during the installation of liquidctl. It finds and calls the CLI entry point from the liquidctl Python library/package. I think this should just work, provided the liquidctl has been installed on the system Python environment, but there could be weird corner cases in more complex scenarios.
> 
> Basically, the service can't find the root `liquidctl` Python _module._ How did you install liquidctl, do you remember? And are you using your system `python3` or some other Python installation?

No worries! I used python3 to install it

jonasmalacofilho, 2021-08-10:
> No worries! I used python3 to install it

I was not able to reproduce a scenario, based on the information you provided, where the `liquidctl` modules end up somewhere the root user later fails to find them.  I tried both `pip3 install liquidctl` and `python3 setup.py install`, with the distro version of `python3` on fresh Ubuntu 20.10 containers.

I'm guessing Ubuntu does something a bit different with Python libraries than the distros I'm more used to.  For a quick fix, I would try the suggestions in [this specific answer on SO](https://stackoverflow.com/a/35641919/2801377).  But, personally, I think you try to find some documentation on how (and why) Ubuntu handles Python libraries.

You could also try checking how the Debian/Ubuntu `liquidctl` package is set up.

Lumene98, 2021-08-11:
> I was not able to reproduce a scenario, based on the information you provided, where the `liquidctl` modules end up somewhere the root user later fails to find them. I tried both `pip3 install [...]

I think I solved it by doing `pip3 install liquidctl` in root

closed 2021-08-11 by Lumene98