liquidctl/liquidctl #363: NZXT Kraken X73 rgb support

MuhammadIbnAlAbbas, 2021-07-28:
NZXT Kraken X73 is in support list, but is NZXT Kraken X73 RGB supported?
like that model ( https://www.amazon.com/NZXT-Kraken-X73-360mm-RL-KRX73-01/dp/B08MB7WQW9?th=1 ) ?

jonasmalacofilho, 2021-07-28:
It's very likely.  In fact, at the OS level, they are probably indistinguishable (the standard X73 can control RGB fans).

But we'll only know for sure once someone tests it and let us know.

jonasmalacofilho, 2021-08-11:
@MuhammadIbnAlAbbas did you end up getting the X73 RGB, and (if so) does our driver work with it as is?

jonasmalacofilho, 2021-08-15:
Closing as, based on the lack of bug reports on the three major projects that support the X73, there is no indication that a different USB product ID is used for the variant that comes with RGB fans.

P.S. The lack of response so far also suggests a significant probability that the author would have never told me if the driver was working and the issue could be closed.

closed 2021-08-15 by jonasmalacofilho