liquidctl/liquidctl #366: Homebrew --HEAD installation is not working

phiilu, 2021-08-01:
I just tried to install the latest `liquidctl` using the following command, but it failed:

```
❯ brew install liquidctl --HEAD
==> Downloading https://files.pythonhosted.org/packages/07/d4/ac5127f7d7e022caf740b9f624e5b9fe9a69fefc0f4f9c047b1e9298c87a/colorlog-5.0.1.tar.g
Already downloaded: /Users/florian/Library/Caches/Homebrew/downloads/e37f454f150cd9b3b5800993626b11200b67309fd9e07f3e06ad34d981d8b5ef--colorlog-5.0.1.tar.gz
==> Downloading https://files.pythonhosted.org/packages/a2/55/8f8cab2afd404cf578136ef2cc5dfb50baa1761b68c9da1fb1e4eed343c9/docopt-0.6.2.tar.gz
Already downloaded: /Users/florian/Library/Caches/Homebrew/downloads/cae543cf1748375c2e56c2a4984cfa49a5858a0a55e6b55a819dfe969ce443d9--docopt-0.6.2.tar.gz
==> Downloading https://files.pythonhosted.org/packages/99/9b/5c41756461308a5b2d8dcbcd6eaa2f1c1bc60f0a6aa743b58cab756a92e1/hidapi-0.10.1.tar.gz
Already downloaded: /Users/florian/Library/Caches/Homebrew/downloads/a35090ea6ded9a8cb8af6025f28eea9dde4992c8e118d2734b26cb0f60dc8f94--hidapi-0.10.1.tar.gz
==> Downloading https://files.pythonhosted.org/packages/d9/6e/433a5614132576289b8643fe598dd5d51b16e130fd591564be952e15bb45/pyusb-1.2.1.tar.gz
Already downloaded: /Users/florian/Library/Caches/Homebrew/downloads/e34b8c197df27ea358bb20b554183616eb7d38ee383c4f254133e71f8370816b--pyusb-1.2.1.tar.gz
==> Cloning https://github.com/jonasmalacofilho/liquidctl.git
Cloning into '/Users/florian/Library/Caches/Homebrew/liquidctl--git'...
fatal: Remote branch master not found in upstream origin
Error: Failed to download resource "liquidctl"
Failure while executing; `git clone --branch master -c advice.detachedHead=false https://github.com/jonasmalacofilho/liquidctl.git /Users/florian/Library/Caches/Homebrew/liquidctl--git` exited with 128. Here's the output:
Cloning into '/Users/florian/Library/Caches/Homebrew/liquidctl--git'...
fatal: Remote branch master not found in upstream origin
```

It seems it trying to default to the `master` branch, but this has now changed to `main`. I guess the fix for this would be to update this line in the [Homebrew Formula](https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/liquidctl.rb#L9) like this:

```ruby
 head "https://github.com/jonasmalacofilho/liquidctl.git", branch: "main"
```

jonasmalacofilho, 2021-08-01:
The homepage and head URLs are outdated as well.  I just sent a PR to homebrew-core (linked above) to fix both problems.

Thanks!

jonasmalacofilho, 2021-08-01:
Fix has been merged.

closed 2021-08-01 by jonasmalacofilho