liquidctl/liquidctl #367: Corsair Lighting Node Core

Collbrothers, 2021-08-01:
Running `liquidctl --match Core set sync color fixed 0080ff` will only affect one pixel of one fan (see img below)
https://imgur.com/a/Ro9jO8i
Additional information about the system:

- Affected device: Corsair QL 120 RGB & Node Core
- Operating system: Arch Linux
- Liquidctl version: AUR 1.7.0

jonasmalacofilho, 2021-08-01:
@MarshallAsch,

I think `--maximum-leds` should default to all the LEDs, so that the lighting modes also default to all LEDs/fans when neither this nor the `--start-led` option has been specified.

What do you think?  Is it at least feasible to implement this change?

(The simplest way to implement it would be to default to the maximum number of LEDs that a channel can support.  But does the controller accept that without locking up, even when there aren't as many connected accessories/LED, or when `--start-led` is greater than one?)

MarshallAsch, 2021-08-02:
It does not mess up the device if too many leds are specified.

I think that that would be a good idea and I can get that implemented tomorrow morning.

MarshallAsch, 2021-08-02:
@Collbrothers do you have 6 of the QL fans connected? And if you do can you confirm that the command `liquidctl --match Core set sync color fixed 0080ff --maximum-leds 204` does indeed turn on all of the LEDs, @jonasmalacofilho identified a potential off by one error that might result in the last LED to not light up.

jonasmalacofilho, 2021-08-02:
I think that test can also be adapted to 4 or any other number of fans.  Which brings another question: @MarshallAsch , do you want @Collbrothers to test with the version he already has installed (1.7.0, apparently[1]) or with the latest commit in #368?

---

@Collbrothers, you said you have liquidctl AUR 1.7.0; that's odd, on Arch liquidctl is on the official repositories (community) at 1.7.0, or on AUR as a `-git` package.  There's no 1.7.0 on the AUR.  Can you clarify what version you have installed (maybe show us the output of `liquidctl --version`)?

MarshallAsch, 2021-08-03:
I actually think the new code is fine I tested it by scaling the Max leds to 3 and the edge behaviour worked as expected, so it may have been wrong before (it would have only been an issue for the edge case)

PR #368 (2021-08-03): changed the default number of the maximum LEDs to use for the commander node deviced
Describe what the changes are meant to address.

This will change the default value for the `maximum-leds` flag to be all LED's and not just `1` LED. 

Fixes: #367 

---

Checklist:

- [x] Adhere to the [development process]
- [x] Conform to the [style guide]
- [x] Verify that the changes work as expected on real hardware
- [ ] Add automated tests cases
- [x] Verify that all (other) automated tests (still) pass
- [x] Update the README and other applicable documentation pages
- [ ] Update the `liquidctl.8` Linu [... cut]
commit e2ec2d34 (2021-08-03): Merge #368: change `--maximum-leds` to default to all LEDs

Fixes: #367
closed 2021-08-03 by jonasmalacofilho