liquidctl/liquidctl #369: Aqua-Computer Aquaero 6 PRO USB fan controller

TyphoonNL, 2021-08-04:
Aqua-Computer Aquaero 6 PRO USB fan controller

Software that you need is called aquasuite (download link: https://aquacomputer.de/software.html?file=tl_files/aquacomputer/downloads/aquasuite/aquasuite_setup.exe)

The device appears to support the following features:

 - [X] monitoring of temperatures or other environment sensors
 - [X] monitoring of fan or pump speeds or duty cycles
 - [X] monitoring of voltages, currents or power
 - [X] configurable fan or pump speeds
 - [ ] configurable voltages or current/power limits
 - [X] configurable lighting of embedded LEDs
 - [X] configurable lighting of accessories like RGB fans or LED strips

The device is connected to the host system using:

- [X] USB
- [ ] PCI-E
- [ ] onboard (the motherboard)
- [ ] other (please elaborate)
- [ ] unknown

The device communicates with the host system using:

- [X] HID
- [X] USB
- [ ] I²C or SMBus
- [ ] other (please elaborate)
- [ ] unknown

> Bus 001 Device 003: ID 0c70:f001 MCT Elektronikladen aquaero
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0c70 MCT Elektronikladen
  idProduct          0xf001 
  bcdDevice            0.02
  iManufacturer           1 Aqua Computer GmbH & Co. KG
  iProduct                2 aquaero
  iSerial                 3 06338-06467
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              5 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     101
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval              32
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              6 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.12
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      56
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              32
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              7 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     156
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5

---

Finally, I can help with:¹

- [X] testing changes on Linux
- [ ] testing changes on Windows
- [ ] testing changes on Mac OS
- [ ] attempting to capture USB traffic
- [ ] attempting to capture I²C/SMBus traffic
- [ ] analyzing traffic data
- [ ] documenting the protocol
- [ ] implementing the driver

¹ Assuming documentation and/or assistance will be provided.

jonasmalacofilho, 2021-08-11:
Gently pinging @TyphoonNL.

jonasmalacofilho, 2021-08-15:
Closing due to lack of feedback.

closed 2021-08-15 by jonasmalacofilho

soenkeliebau, 2025-03-04:
Hey there, I'd be interested in trying to capture a few traces and providing them here, if there is still an appetite to try and get this device supported?

jonasmalacofilho, 2025-03-05:
Ping @aleksamagicka .

aleksamagicka, 2025-03-09:
They indeed have the same offsets. I've opened #773, it's a draft at this point but can be tested.