liquidctl/liquidctl #370: Should we continue publishing a pre-built exe for Windows?

jonasmalacofilho, 2021-08-18:
The pre-built exe for Windows is becoming a hassle due to _(what are very likely all false)_ positive detentions on anti-virus tools.

This first happened with 1.5.0 on Windows Defender, and has come back with 1.7.1 on Defender and a bunch of other AV tools.

- Confirmed false positive for v1.7.1 from Windows Defender:
    * #359
- Confirmed false positive for v1.5.0 from Windows Defender:
   * [Commit 5e1b26cfe370 ("docs: readme: temporarily remove link to 1.5.0 executable")](https://github.com/liquidctl/liquidctl/commit/5e1b26cfe3700f7517527a7a2a65518266385c26)
   * [Commit 1ee690375351 ('Revert "docs: readme: temporarily remove link to 1.5.0 executable"')](https://github.com/liquidctl/liquidctl/commit/1ee69037535136dc6a8a9dd637ff738087b4f620)
- Current reports from VirusTotal:
    * [`liquidctl.exe` v1.7.1](https://www.virustotal.com/gui/file/487759d4f0d3dc4291d056cada8474a301376808d19c5582e1cadab032042c39/detection)
    * [`liquidctl-1.7.1-bin-windows-x86_64.zip`](https://www.virustotal.com/gui/file/5f35d4ac8ad6da374877d17c7a36bbb202b0a74bd773ebe45444f0089daba27b/detection)

The issue linked above has a bit more information, but the gist of why I think these are probably all false positives is:

> The executable is not built on my machine, but rather on a public continuous integration service, from a public recipe and mostly only using trustworthy open-source components (the only exceptions to the open-source part are Microsoft Windows and MSVC).
> —from a recent comment of mine on Reddit

Current reddit threads about this:

- [r/NZXT: Is LiquidCTL safe?](https://www.reddit.com/r/NZXT/comments/p6eryq/is_liquidctl_safe/)
- [r/buildapc: Is LiquidCTL a safe alternative to CAM?](https://www.reddit.com/r/buildapc/comments/p6gn13/is_liquidctl_a_safe_alternative_to_cam/)

I also answered someone else about this recently, but I don't remember where.

---

Even if they are all false positives, they put me in a tough spot: I feel pressured to spend hours reporting every single one of these instances (which may require that I first install all of these tools).  But that's not something I can or want to do.

The other alternative I see is shaming the silly heuristics these tools use.  But I'm not a security expert, nor do I want to encourage everyone to disregard all warnings that look like that: as much I think some of these heuristics are worse than a random filter, they are probably still beneficial for those users that just click on anything they find on the internet.

So, I've been considering no longer publishing the exe (and removing the ones for previous releases).  The downsides of this approach is that our Windows users may:

- simply stop using liquidctl;
- use some random liquidctl.exe they find on the internet;
- fill our issue tracker with problems related to the installation (on Windows) of Python, the dependencies, or liquidclt itself;
- or all of the above.

What are the thoughts of our community on this topic?

reelmccoy, 2021-08-18:
I've used ESET NOD32 for eons and have always had good luck with it's detection. I've been using 1.7.1 since it was released with ESET never flagging it. One run with Emisoft's Emergency Kit flagged the executable which had me flinging the executable at Virus Total which had a number of (unknown to me) AV solutions flagging it as well. In the end, I deemed the detection as a false positive and left it at that. 

I can understand the reasoning behind wanting to stop releasing the Windows executable as any amount of time I spent researching this false positive has to be amplified to deal with countless sites asking if it's safe. 

The downfall is that the Windows EXE release really makes it easy to use LiquidCTL in the Windows realm without having to figure out how to get it to work otherwise. So I would wonder how many Windows users wouldn't use LiquidCTL as a result? Who knows. I would think those using LiquidCTL are more technical than those using brand's specific monitoring software.

In my case, I'll figure it out. I'd rather not use NZXT's CAM and take my chance at pump/fan USB detection roulette. I'll spend some time getting LiquidCTL running without the executable distribution. :)

sushibait, 2021-09-08:
Please do not stop publishing :)

You give a lot of users with NZXT devices a much, much better alternative to CAM.

gringan, 2021-09-09:
Please do not stop publishing. I use liquid instead of NZXT CAM. 

Thx for all

jonasmalacofilho, 2021-10-05:
Unfortunately it happened again with 1.7.2; this time MS Defender reports it as `Trojan:Win32/Sabsik.FL.B!ml`.[^1]

While that's probably still just a false positive, it's no longer obviously so.  Either our dependencies and/or AppVeyor are no
longer trustworthy, or the false positive issue is truly out of control.[^2]

Either way, I don't see any option other than to yank the executables (e91fcc44ec37, 0ab403f8f661).

[^1]: https://www.virustotal.com/gui/file/c806283ac8da8c597742786dd0a835ea447eeb1066c6bd442f15294067c917ab
[^2]: I'm guessing it's the latter.

closed 2021-10-05 by jonasmalacofilho

caesay, 2022-05-12:
I would love to benefit from some of the recent updates, but unfortunately can't use this project unless it ships a windows binary. Would you consider adding build instructions or a providing the windows build script you used?

I looked at https://github.com/liquidctl/liquidctl/commit/ed28fb9373b6a7f04e2235c2723b64caedb5e6ad and found the old appveyor (https://ci.appveyor.com/project/jonasmalacofilho/liquidctl/builds/42722264) but I can't spot anything obvious in the appveyor logs, or the github actions script that shed any light.

jonasmalacofilho, 2022-05-12:
@caesay,

> I would love to benefit from some of the recent updates, but unfortunately can't use this tool unless it ships a windows binary (not a flexible requirement). Would you consider adding build instructions or a providing the windows build script you used?

Can I ask why the executable is necessary?
(Given that the alternative is simply installing Python and running `pip install liquidctl`).

> I looked at [ed28fb9](https://github.com/liquidctl/liquidctl/commit/ed28fb9373b6a7f04e2235c2723b64caedb5e6ad) and found the old appveyor (https://ci.appveyor.com/project/jonasmalacofilho/liquidctl/builds/42722264) but I can't spot anything obvious in the appveyor logs, or the github actions script that shed any light on how to produce a windows binary.

I think the Appveyor log you're looking at postdates the disabling of Windows `.exe` executables.

Anyway, [we used to](https://github.com/liquidctl/liquidctl/blob/e23a5cd14cfec032ede88847119892a2721903c4/appveyor.yml#L48) use PyInstaller:

```
pyinstaller --add-data VS2019\MS64\dll\*;. --clean -F liquidctl\cli.py --name liquidctl --onefile --distpath .
```

Today should should be able to leave out the manual embedding of the LibUSB DLLs (the `--add-data` part).

P.S. Unfortunately some malware reportedly also uses PyInstaller (to build their executables), and AV software "AI" heuristics are really dumb: so expect a ton of warnings after you build your executable...

caesay, 2022-05-13:
The way it's being distributed I can not rely on python/pip being installed, or using pip as a package manager. The link to the old appveyor script was very useful, thanks for that. 

> P.S. Unfortunately some malware reportedly also uses PyInstaller (to build their executables), and AV software "AI" heuristics are really dumb: so expect a ton of warnings after you build your executable...

This is just a common/expected part of building software for windows. Even a empty C# dotnet 6 binary, with zero code, is detected as malware by default - and dotnet is a Microsoft technology.

The best way to avoid AV detections (if you were interested in ever shipping a windows binary ever again) is to codesign your binary. This is often all that is necessary to avoid detections. 

It is also possible to build up binary reputation over time, this applies to codesigned and non-codesigned binaries. You can build reputation by:
 - Always hosting the binary in the same location (eg. `https://yoursite.com/latest/some-static-download-location.zip`).
 - Having users use it and tell windows defender it's wrong when it gets detected.
 - Submit every release to Microsoft for malware analysis in advance.

But honestly, all of that stuff is optional. If your new release is detected by a virus, and you do nothing about it, it will very likely stop being detected after you have a few dozen people who have decided to use it anyways.

jonasmalacofilho, 2022-05-13:
@caesay,

> The way it's being distributed [...]

So you're redistributing it.  I would love to know more about how/where it's being used!  (Please contact me privately if necessary).

> This is just a common/expected part of building software for windows. [...]

I suppose, although I've had more issues than usual with PyInstaller.  Anyway, I'm aware of the workarounds, but they are either expensive or annoying.  Are you signing the liquidctl executable you're distributing?

caesay, 2022-05-13:
> So you're redistributing it. 

Not really, sorry for the confusion. Just using it privately on my own computers / servers. I have a home grown management utility which does all scripting / scheduled tasks / checking for tool updates from github etc. I can't install python for various reasons on some of the machines, and my utility does not support delegating installation/updating to pip in any case. 

> I suppose, although I've had more issues than usual with PyInstaller. 

Lucky then, because pretty much every binary I've compiled in the last 3 years has been flagged as a virus :S.

> Are you signing the liquidctl executable you're distributing?

I don't need to sign it, as I mentioned just using it myself - and virus detections are not an issue in any case. just par for the course.

caesay, 2022-05-13:
But I would add, that in general most windows users probably will not have or want python, so I can imagine a lot of people struggling with this for far more common reasons.

AeliusSaionji, 2022-06-08:
Absent a windows binary, could explicit support for msys2 be added? Real python has always been an awkward thing on Windows, and I hate to use that when msys2 is an option.

Either publish directly to msys2 repos, or list msys2 specific install instructions, or do whatever is necessary to fix the issue I'm seeing after following the generic build instructions:

```sh
aelius@Pyrrha UCRT64 ~
$ liquidctl/bin/liquidctl.exe status --verbose
INFO: status reports not available from Gigabyte RGB Fusion 2.0 8297 Controller
NZXT Kraken X (X53, X63 or X73)
INFO: detailed error: unexpected error with NZXT Kraken X (X53, X63 or X73): UnicodeEncodeError('charmap', '\u251c\u2500\u2500 Liquid temperature    28.2  ▒C', 0, 3, 'character maps to <undefined>')
Traceback (most recent call last):
  File "C:/tools/msys64/home/aelius/liquidctl/lib/python3.10/site-packages/liquidctl/cli.py", line 467, in main
    _print_dev_status(dev, status)
  File "C:/tools/msys64/home/aelius/liquidctl/lib/python3.10/site-packages/liquidctl/cli.py", line 272, in _print_dev_status
    print(f'\u251c\u2500\u2500 {k:<{kcols}}    {v:>{vcols}}  {u}')
  File "C:/tools/msys64/ucrt64/lib/python3.10/encodings/cp1252.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode characters in position 0-2: character maps to <undefined>
ERROR: unexpected error with NZXT Kraken X (X53, X63 or X73): UnicodeEncodeError('charmap', '\u251c\u2500\u2500 Liquid temperature    28.2  ▒C', 0, 3, 'character maps to <undefined>')
```

thanks

jonasmalacofilho, 2022-06-08:
@AeliusSaionji ,

> Absent a windows binary, could explicit support for msys2 be added?

As long as someone figures out what's necessary, yes.

That said, one issue I foresee is that the version of Python that ends up being executed within MSYS2 could be the stock CPython for Windows, _or_ a MSYS2 patched version of it...

> Publish directly to msys2 repos?

I'm not sure how their packaging works, but usually we don't push directly to downstream.  Instead, downstream maintains a build script and chooses which versions (and patches) they want to distribute and support.

Also, while I'm very familiar with Pacman, I don't use Windows or MSYS2 much.  In other words, I can help with the packaging part, but I wont be of much use debugging platform specific issues, especially those that _aren't_ specific to liquidctl...

(Also confirming the issue you reported), the following fails on MSYS2 (the one bundled with Git):

```
$ python -c "import locale; print(locale.getpreferredencoding()); print('\u2514 26.5\u00B0C')"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\dad\AppData\Local\Programs\Python\Python310\lib\encodings\cp1252.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode character '\u2514' in position 0: character maps to <undefined>
cp1252
```

But Windows Command Prompt has no issue with it:

```
> python -c "import locale; print(locale.getpreferredencoding()); print('\u2514 26.5\u00B0C')"
cp1252
└ 26.5°C
```

Since in both cases the preferred encoding is the same and does _not_ support Unicode, I'm guessing Python has a special workaround for Command Prompt (or Windows as whole), but that's not enabled and/or working on MSYS2.

> List msys2 specific install instructions?

If you can help figuring out what's necessary...

> Incidentally, the generic instructions work under windows terminal (provided you install libusb manually),

Actually, manually installing LibUSB should no longer be necessary since 1.9.x: the `libusb_package` dependency should already included a copy of `libusb-1.0.dll`.

AeliusSaionji, 2022-06-09:
> Actually, manually installing LibUSB should no longer be necessary since 1.9.x: the libusb_package dependency should already included a copy of libusb-1.0.dll.

I used the ucrt64 environment of msys2, and it simply didn't work until I installed the libusb package in ucrt64/msys2.
I tried the mingw64 environment, but liquidctl wouldn't detect libusb in the mingw64/msys2 environment at all.

> If you can help figuring out what's necessary [for msys2]

I had started composing the message before I got liquidctl to work at all. I was assuming I wouldn't be able to resolve this alone.

In hindsight it was relatively simple: the only four notes:
1. Use UCRT64 instead of mingw64 (haven't tested clang, don't know why mingw64 didn't work)
2. Install ucrt64/msys2's gcc
3. Install ucrt64/msys2's libusb (don't know why `pip`ing liquidctl didn't "just work")
4. Once compiled, only try to execute it from windows consoles (again, no idea).

Hopefully there are no further issues

jonasmalacofilho, 2022-06-09:
> I used the ucrt64 environment of msys2, and it simply didn't work until I installed the libusb package in ucrt64/msys2.
I tried the mingw64 environment, but liquidctl wouldn't detect libusb in the mingw64/msys2 environment at all.

Oh, when you previously said "under windows terminal", I assumed `cmd`...

Anyway, on MSYS2 that makes sense, since the auto-bundling goes through a bunch of platform checks, and, statistically, I'm sure many of them don't consider MSYS2 or are very conservative for regarding Windows-but-not-Windows environments.

> In hindsight it was relatively simple [...]

In other for us to add any of this to our documentation, we would need to know _why_ it's only works in this very specific configuration.  Otherwise we might actually create more problems than we're trying to solve for anyone not already running a very similar configuration.

Also, it feels overly complicated in itself... like, there (usually) should a simpler way to make it work.

> only try to execute it from windows consoles (again, no idea).

On the MSYS2 terminal, have you tried setting the encoding to a Unicode-capable one?

On my Git-bundled MSYS2 it was actually _blank_ by default.  But I didn't try to change it (yet)...

AeliusSaionji, 2024-03-11:
2024 update:

Been a while since I used liquidctl, since I am just not a fan of python wrangling. However, someone recently showed me pipx, which greatly reduces the friction of having to deal with python.

liquidctl seems to run just fine under msys2 with the following instructions:

1: I've only tested ucrt64
2: Install from msys2 ucrt64 repos cython0 and python-pillow: `pacman -S mingw-w64-ucrt-x86_64-cython0 mingw-w64-ucrt-x86_64-python-pillow`
3: Pipx will handle the rest: `pipx install --system-site-packages liquidctl`

For whatever reason, pipx will fail to build pillow, and will fail without cython0.

caesay, 2024-03-11:
I'm still using a very old windows exe/build of liquidctl, I wasn't able to build one myself using the linked AppVeyor script when I tried back in 2022, and there is no real alternative available. Very poor instructions for Windows in any case, and requiring msys2 or python etc. is just not possible for me. Still hoping a windows build will return one day, or at least some build instructions.

AeliusSaionji, 2024-03-14:
I tried to package my build for you, but one big reason I use msys2 is precisely because it shields me from the idiosyncrasies of python... I immediately ran into a dependency hell :tm: problem with pyinstaller :/