liquidctl/liquidctl #373: Insufficient permissions despite udev rules

deags, 2021-08-22:
Fedora 33
Corsair HX750i
liquidctl 1.7.1

Hi,
I've added the .rules from the repo to /etc/udev/rules.d/ and they appear to be applied correctly, however I can only get status when running liquidctl as root.

I am wanting to historise metrics constantly, so running as root is not a good solution.

I suspect this is not a liquidctl issue, but I am not sure how to troubleshoot further and suspect this might be an issue for other liquidctl users?

lsusb
`Bus 001 Device 003: ID 1b1c:1c05 Corsair HX750i Power Supply`

ls -lA /dev/bus/usb/001/
```
[username@hostname ~]$ ls -lA /dev/bus/usb/001/003*
crw-rw-r--. 1 root root 189, 2 Aug 22 13:48 /dev/bus/usb/001/003
```

liquidctl status
```
[username@hostname ~]$ liquidctl status
ERROR: Error: could not open Corsair HX750i, possibly due to insufficient permissions
```

sudo liquidctl status
```
[username@hostname ~]$ sudo liquidctl status
[sudo] password for username: 
Corsair HX750i
├── Current uptime               7 days, 3:57:46  
├── Total uptime             1406 days, 21:57:46  
├── Temperature 1                           42.0  °C
├── Temperature 2                           35.2  °C
├── Fan control mode                    Hardware  
├── Fan speed                                  0  rpm
├── Input voltage                         230.00  V
├── +12V OCP mode                     Multi rail  
├── +12V output voltage                    12.16  V
├── +12V output current                     5.00  A
├── +12V output power                      60.00  W
├── +5V output voltage                      4.97  V
├── +5V output current                      3.31  A
├── +5V output power                       16.00  W
├── +3.3V output voltage                    3.30  V
├── +3.3V output current                    0.56  A
├── +3.3V output power                      1.50  W
├── Total power output                     76.00  W
├── Estimated input power                  86.00  W
└── Estimated efficiency                      88  %
```

deags, 2021-08-22:
Issue resolved by changing the included .rules file as follows:
```
# Corsair HX750i
#SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="1c05", TAG+="uaccess"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="1c05", MODE="0666"
```
Contrary to the above the relevent /dev/ file seems to be /dev/hidraw0 which now appears with 0666 as wanted:
```
[root@hostname liquidctl]# ls -lA /dev/ | grep hid
crw-rw-rw-.  1 root root    242,   0 Aug 22 22:31 hidraw0
```
I haven't been able to find conclusive info but it appears that TAG+="uaccess" only applies for 'physical' log ins. I have seen this emphasised on some pages, it appears that physical is distinct from remote (ssh) sessions? This is a headless unit which is reasonably atypical usage for liquidctl I guess.

Thanks

jonasmalacofilho, 2021-08-31:
> I haven't been able to find conclusive info but it appears that TAG+="uaccess" only applies for 'physical' log ins. I have seen this emphasised on some pages, it appears that physical is distinct [...]

Yes, that's correct (btw, I think that in some places the relevant upstream documentation hints to this by using the word "seat").

I updated the comment on top of the rules file to mention headless sessions.

---

I think `GROUP="<your group>" MODE="0660"` might be a bit more appropriate/secure for your use case; after all, you're no longer restricting access to physical sessions, and I don't think that you'll ever want (for example) `www-data` to have access to the device.

closed 2021-08-31 by jonasmalacofilho