liquidctl/liquidctl #377: Permission not work for Corsair Hydro H150i Pro XT  inspite of  setting udev rules

dai2bungeibu, 2021-09-06:
I would like to use Corsair Hydro H150i Pro XT on Ubuntu. I set udev rules using 71-liquidctl.rules to resolve permission issue, but error occurs: could not open it, possibly due to insufficient permissions. I firstly used ubuntu 20.04LTS and lquidctl  v1.7.1 installed by snap. Next I upgraded to Ubuntu 20.10 and used liquidctl v1.5.1, but this was not supported Corsair Hydro H150i Pro XT. Now I am using Ubuntu 20.10 and liquidctl v1.7.1 installed snap. Can I resolve permission issue?

> leokun@rosachinensis:~$ sudo liquidctl initialize -g
[DEBUG] cli (main): running liquidctl v1.7.1 (6295354afb8d)
[DEBUG] smbus (find_devices): searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] smbus (find_devices): found I²C bus i2c-3
[DEBUG] smbus (find_devices): found I²C bus i2c-1
[DEBUG] smbus (find_devices): found I²C bus i2c-4
[DEBUG] smbus (find_devices): found I²C bus i2c-2
[DEBUG] smbus (find_devices): found I²C bus i2c-0
[DEBUG] usb (find_devices): searching HidapiBus (CommanderCore, CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] usb (find_devices): found HID device 046d:c246
[DEBUG] usb (find_devices): found HID device 046d:c246
[DEBUG] usb (find_devices): found HID device 046d:c246
[DEBUG] usb (find_devices): found HID device 046d:c246
[DEBUG] usb (find_devices): found HID device 0b05:1939
[DEBUG] usb (find_devices): found HID device 1b1c:0c22
[DEBUG] hydro_platinum (probe): instanced driver for Corsair Hydro H150i Pro XT (experimental)
[DEBUG] usb (find_devices): found HID device 0853:0200
[DEBUG] usb (find_devices): searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc, CorsairAsetekProDriver)
[DEBUG] usb (find_devices): found USB device 1d6b:0003
[DEBUG] usb (find_devices): found USB device 1d6b:0002
[DEBUG] usb (find_devices): found USB device 1d6b:0003
[DEBUG] usb (find_devices): found USB device 0853:0200
[DEBUG] usb (find_devices): found USB device 1b1c:0c22
[DEBUG] usb (find_devices): found USB device 05e3:0610
[DEBUG] usb (find_devices): found USB device 0b05:1939
[DEBUG] usb (find_devices): found USB device 046d:c246
[DEBUG] usb (find_devices): found USB device 1d6b:0002
[DEBUG] cli (main): device: Corsair Hydro H150i Pro XT (experimental)
[INFO] cli (log_error): open failed
Traceback (most recent call last):
  File "/snap/liquidctl/3/lib/python3.8/site-packages/liquidctl/cli.py", line 420, in main
    with dev.connect(**opts):
  File "/snap/liquidctl/3/lib/python3.8/site-packages/liquidctl/driver/hydro_platinum.py", line 178, in connect
    ret = super().connect(**kwargs)
  File "/snap/liquidctl/3/lib/python3.8/site-packages/liquidctl/driver/usb.py", line 118, in connect
    self.device.open()
  File "/snap/liquidctl/3/lib/python3.8/site-packages/liquidctl/driver/usb.py", line 384, in open
    self.hiddev.open_path(self.hidinfo['path'])
  File "hidraw.pyx", line 74, in hidraw.device.open_path
OSError: open failed
[ERROR] cli (log_error): Error: could not open Corsair Hydro H150i Pro XT (experimental), possibly due to insufficient permissions

> leokun@rosachinensis:~$ sudo liquidctl list -g
[DEBUG] cli (main): running liquidctl v1.7.1 (6295354afb8d)
[DEBUG] smbus (find_devices): searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] smbus (find_devices): found I²C bus i2c-3
[DEBUG] smbus (find_devices): found I²C bus i2c-1
[DEBUG] smbus (find_devices): found I²C bus i2c-4
[DEBUG] smbus (find_devices): found I²C bus i2c-2
[DEBUG] smbus (find_devices): found I²C bus i2c-0
[DEBUG] usb (find_devices): searching HidapiBus (CommanderCore, CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] usb (find_devices): found HID device 046d:c246
[DEBUG] usb (find_devices): found HID device 046d:c246
[DEBUG] usb (find_devices): found HID device 046d:c246
[DEBUG] usb (find_devices): found HID device 046d:c246
[DEBUG] usb (find_devices): found HID device 0b05:1939
[DEBUG] usb (find_devices): found HID device 1b1c:0c22
[DEBUG] hydro_platinum (probe): instanced driver for Corsair Hydro H150i Pro XT (experimental)
[DEBUG] usb (find_devices): found HID device 0853:0200
[DEBUG] usb (find_devices): searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc, CorsairAsetekProDriver)
[DEBUG] usb (find_devices): found USB device 1d6b:0003
[DEBUG] usb (find_devices): found USB device 1d6b:0002
[DEBUG] usb (find_devices): found USB device 1d6b:0003
[DEBUG] usb (find_devices): found USB device 0853:0200
[DEBUG] usb (find_devices): found USB device 1b1c:0c22
[DEBUG] usb (find_devices): found USB device 05e3:0610
[DEBUG] usb (find_devices): found USB device 0b05:1939
[DEBUG] usb (find_devices): found USB device 046d:c246
[DEBUG] usb (find_devices): found USB device 1d6b:0002
Device #0: Corsair Hydro H150i Pro XT (experimental)
├── Vendor ID: 0x1b1c
├── Product ID: 0x0c22
├── Release number: 0x0100
├── Bus: hid
├── Address: /dev/hidraw2
└── Driver: HydroPlatinum
[DEBUG] cli (_list_devices_human): hierarchy: UsbHidDriver, BaseUsbDriver, BaseDriver

>leokun@rosachinensis:~$ ls -l /dev/hidraw2
crw-rw----+ 1 root root 237, 2  9月  6 19:10 /dev/hidraw2

> leokun@rosachinensis:~$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0853:0200 Topre Corporation RealForce Compact Keyboard
Bus 001 Device 005: ID 1b1c:0c22 Corsair iCUE H150i RGB PRO XT Liquid CPU Cooler
Bus 001 Device 003: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 002: ID 0b05:1939 ASUSTek Computer, Inc. AURA LED Controller
Bus 001 Device 006: ID 046d:c246 Logitech, Inc. Gaming Mouse G300
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

> leokun@rosachinensis:~$ lsusb -s 1:5 -v
Bus 001 Device 005: ID 1b1c:0c22 Corsair iCUE H150i RGB PRO XT Liquid CPU Cooler
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c22 iCUE H150i RGB PRO XT Liquid CPU Cooler
  bcdDevice            1.00
  iManufacturer           1 Corsair Inc.
  iProduct                2 H150iRGBPROXT
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               7
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered

jonasmalacofilho, 2021-09-10:
Are you encountering this problem in interactive or remote sessions?

I recently added a [note](https://github.com/liquidctl/liquidctl/blob/6295354afb8d/extra/linux/71-liquidctl.rules#L25-L28) about remote/headless sessions to the rules file (related: #373):

```
# These rules also assume that only physical sessions (or "seats") need
# unprivileged access to the devices.  In case headless sessions are also
# expected to interactively run liquidctl, using GROUP and MODE might be
# preferable to TAG+="uaccess".
```

---

The fact that you're installing through `snap` could also play a role.  Note that the snap [documentation](https://snapcraft.io/liquidctl) states that:

> To allow this snap access to your device(s), issue the following connection commands:
> 
> ```
>  $ sudo snap connect liquidctl:raw-usb
>  $ sudo snap connect liquidctl:hardware-observe
> ```

dai2bungeibu, 2021-09-12:
After that, I installed liquidctl by using pip. In this result, it can be working with Corsair Hydro H150i Pro XT without permission issue. Thank you.

Maybe the snap package of liqudctl includes bug.

> Are you encountering this problem in interactive or remote sessions?

I used interactive session.

> > ```
> >  $ sudo snap connect liquidctl:raw-usb
> >  $ sudo snap connect liquidctl:hardware-observe
> > ```

Yes, the snap connect commands had been input before liquidctl initializing.

jonasmalacofilho, 2021-09-15:
In that case it might be helpful to report the issue to the snap package.

closed 2021-09-15 by jonasmalacofilho