liquidctl/liquidctl #379: Freeze while initializing with H150i AIO

RolandWarburton, 2021-09-11:
I have run into an issue where liquidctl previously was able to start (liquidctl initialize all), and no longer works now, this issue presents itself as soon as I run liquidctl and freezes the terminal indefinitely.

When running `liquidctl initialize all` or `liquidctl status` the program freezes while (presumably) getting data from my corsair AIO device.

I have a corsair commander pro which seems to work.

I also have a Corsair H150i Pro which seems to not be readable. I have the USB plugged directly into the motherboard and `lsusb` can see it.

**When does the issue occur:**
When running `liquidctl initialize all`

**OS/Distribution:**
Debian 11 5.10.0-8-amd64

**Installed via:**
python package, and tested cloning repo directly (neither fixes the issue)

**version of liquidctl (liquidctl --version):**
liquidctl v1.7.1 (6295354afb8d)

**Output from lsusb:**
See line 9.

```none
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 0951:1666 Kingston Technology DataTraveler 100 G3/G4/SE9 G2/50
Bus 004 Device 006: ID 0499:172f Yamaha Corp. Steinberg UR22C
Bus 004 Device 005: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 004 Device 004: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 004 Device 003: ID 174c:3074 ASMedia Technology Inc. ASM1074 SuperSpeed hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 1b1c:0c22 Corsair iCUE H150i RGB PRO XT Liquid CPU Cooler
Bus 003 Device 007: ID 046d:082b Logitech, Inc. Webcam C170
Bus 003 Device 013: ID 1532:0016 Razer USA, Ltd DeathAdder 3.5G
Bus 003 Device 012: ID 04d9:0169 Holtek Semiconductor, Inc. USB Keyboard
Bus 003 Device 011: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 003 Device 010: ID 056e:010d Elecom Co., Ltd M-HT1DRBK HUGE Wireless Optical TrackBall
Bus 003 Device 009: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 003 Device 008: ID 13ba:0018 PCPlay Barcode PCP-BCG4209
Bus 003 Device 006: ID 14cd:8601 Super Top 4-Port hub
Bus 003 Device 005: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 003 Device 004: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 003 Device 002: ID 174c:2074 ASMedia Technology Inc. ASM1074 High-Speed hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 8087:0029 Intel Corp. AX200 Bluetooth
Bus 001 Device 006: ID 0b05:18f3 ASUSTek Computer, Inc. AURA LED Controller
Bus 001 Device 005: ID 1b1c:0c10 Corsair Commander PRO
Bus 001 Device 004: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

**Output from liquidctl initialize all:**

```none
❯ liquidctl initialize all --debug
[DEBUG] cli (main): running liquidctl v1.7.1 (6295354afb8d)
[DEBUG] smbus (find_devices): searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] smbus (find_devices): found I²C bus i2c-3
[DEBUG] smbus (find_devices): found I²C bus i2c-1
[DEBUG] smbus (find_devices): found I²C bus i2c-6
[DEBUG] smbus (find_devices): found I²C bus i2c-4
[DEBUG] smbus (find_devices): found I²C bus i2c-2
[DEBUG] smbus (find_devices): found I²C bus i2c-0
[DEBUG] smbus (find_devices): found I²C bus i2c-5
[DEBUG] usb (find_devices): searching HidapiBus (CommanderCore, CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] usb (find_devices): found HID device 1b1c:0c10
[DEBUG] usb (probe): instanced driver for Corsair Commander Pro
[... 85 lines cut ...]
[DEBUG] keyval (__init__): data in /run/user/1000/liquidctl/vid1b1c_pid0c22/loc0
[DEBUG] keyval (store): stored leds_enabled=0 (in /run/user/1000/liquidctl/vid1b1c_pid0c22/loc0/leds_enabled)
[DEBUG] keyval (store): stored pump_mode=1 (in /run/user/1000/liquidctl/vid1b1c_pid0c22/loc0/pump_mode)
[DEBUG] keyval (load): loaded fan1_mode=2 (from /run/user/1000/liquidctl/vid1b1c_pid0c22/loc0/fan1_mode)
[DEBUG] keyval (load): loaded fan1_duty=100 (from /run/user/1000/liquidctl/vid1b1c_pid0c22/loc0/fan1_duty)
[INFO] hydro_platinum (_generate_cooling_payload): setting fan1 to 100% duty cycle
[DEBUG] keyval (load): loaded fan2_mode=2 (from /run/user/1000/liquidctl/vid1b1c_pid0c22/loc0/fan2_mode)
[DEBUG] keyval (load): loaded fan2_duty=100 (from /run/user/1000/liquidctl/vid1b1c_pid0c22/loc0/fan2_duty)
[INFO] hydro_platinum (_generate_cooling_payload): setting fan2 to 100% duty cycle
[DEBUG] keyval (load): loaded fan3_mode=2 (from /run/user/1000/liquidctl/vid1b1c_pid0c22/loc0/fan3_mode)
[DEBUG] keyval (load): loaded fan3_duty=100 (from /run/user/1000/liquidctl/vid1b1c_pid0c22/loc0/fan3_duty)
[INFO] hydro_platinum (_generate_cooling_payload): setting fan3 to 100% duty cycle
[DEBUG] keyval (load): loaded pump_mode=1 (from /run/user/1000/liquidctl/vid1b1c_pid0c22/loc0/pump_mode)
[INFO] hydro_platinum (_send_set_cooling): setting pump mode to balanced
[DEBUG] keyval (load_store): loaded sequence=13 (from /run/user/1000/liquidctl/vid1b1c_pid0c22/loc0/sequence)
[DEBUG] keyval (load_store): replaced with sequence=14 (stored in /run/user/1000/liquidctl/vid1b1c_pid0c22/loc0/sequence)
[DEBUG] usb (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] usb (write): writting report 0x00 with 64 bytes: 3f:73:14:00:ff:05:ff:ff:ff:ff:ff:02:00:00:00:00:ff:00:00:00:00:00:00:ff:00:00:00:00:00:07:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:fe
```

Additional information about the system:

- Affected device: iCUE H150i RGB PRO XT Liquid CPU Cooler
- Operating system: Debian 11
- Liquidctl version: liquidctl v1.7.1 (6295354afb8d) (installed via github repo)

jonasmalacofilho, 2021-09-11:
Have you simultaneously accessed the device from different processes/tools?  The Pro XT is susceptible to the problem described in #274.

If that's the case, you should be able to clear the problem by allowing the cooler to reset itself, which usually requires shutting down the computer to mechanical off (=turn the PSU off) for a couple of minutes, because the embedded controller in the cooler may be powered from the +5V Stand-by rail.

RolandWarburton, 2021-09-12:
Thank you and yes a simple reboot seems to have solved it!

I would not be surprised if I was running two liquidctl commands at the same time to have caused this as I woke up this morning and turned my computer on to find it solved.

closed 2021-09-12 by RolandWarburton