liquidctl/liquidctl #385: NZXT Smart Device V2: 'moving-alternating' mode does just 'alternating' since 1.5.0

Hely0n, 2021-09-26:
In November 2020, I installed liquidctl to set the colors of my LEDs at boot. One important mode here was 'moving-alternating-5' for an LED strip, which looked as it should: The two colors move step by step in one direction.

Now I have a little bit of free time and want to write a yoda-like Windows Service, of course using the newest liquidctl build.
The Problem is: since Version 1.5.0, the "moving-alternate" effect doesn't move forward step by step anymore, but instead just  alternates the to colors, which looks terrible. I couldn't figure out, which change made this happen. 

I don't think using an one year old version is a good workaround for this problem.

Additional information about the system:

- Affected device: NZXZ Smart Device V2, HUE LED Strip
- Operating system: Windows 10 x64
- Liquidctl version: >=1.5.0

MarshallAsch, 2021-09-27:
Did you test it against any of the versions between 1.5.0 and 1.7.1? 

Would you be able to test some of those versions to see if you can find the specific version that broke the behaviour. I took a quick look at the code and there have been a number of changes to that module but they should not have changed the actual behaviour, narrowing down which set of changes needs to be looked at will help.

Hely0n, 2021-09-27:
Yes, I tested some versions, as I wanted to use the most new version possible for this project. 1.4.2 Is the version where it looks as it should. At 1.5.0 it was already broken. I think there is no versions in-between, isn't it? 
Forunately my code works with 1.4.2 (just without the handy "with dev.connect()" but hey). So for me personally, it's not urgent anymore.
Nevertheless, I think it should be fixed sometime.

BTW
Is there some demand for my Windows Service? Maybe I could contribute the code and/or .exe to /extra . It's very simple but handy, as it controls radiator fans connected to an NZXT Smart Device, based on the liquid temperature from a NZXT Kraken Pump. Everything is configurable via a JSON file. Due to the simplicity, It should be very easy to adapt it to other manufacturers.

MarshallAsch, 2021-09-28:
> 1.4.2 Is the version where it looks as it should. At 1.5.0 it was already broken.

ah I missed that part. For some reason I read that as 'it worked in 1.5.0', Ill try to take a look at it in the morning and see what I can find.

MarshallAsch, 2021-09-28:
This is the relevant diff: https://github.com/liquidctl/liquidctl/compare/v1.4.2...v1.5.0 looking at `liquidctl/driver/smart_device.py`

Can you try with the `moving-alternating-3` mode on v1.5.0? after taking a look at the code it looks like the the `moving-alternating` is not defined for the smart device v2 (just the v1). This seems to be the case for both the 1.4.2 and the 1.5.0 releases, If that other mode works as expected then I think just adding the `moving-alternating` mode back into the list should solve the problem

Hely0n, 2021-09-28:
At 1.5.0, "moving-alternating-3" looks like "alternating-3", so still same issue with that one.
I don't know, how modes are selected, but maybe there is something like an "if string.contains("alternating")", so it picks this mode directly.

MarshallAsch, 2021-09-28:
thats really really weird. It works by pulling all the configurations for a mode from a map of configurations with the mode as the key. There has been no changes to how that block of the code functions since before 1.4.0. (aside from changing the the `backward-<mode>` to be `--direction=backward`

Hely0n, 2021-09-28:
Yeah, I tought the same when I was looking for the changes. I tried --direction backward and explicitly forward, but it looks the same

MarshallAsch, 2021-09-28:
what about explicitly calling the `--speed` flag?

Hely0n, 2021-09-28:
.\liquidctl.exe -m smart set led1 color moving-alternating-3 "hsv(200,100,1)" "hsv(200,80,1)" --speed slower --verbose --debug

`[DEBUG] __main__: running liquidctl v1.5.0 (AppVeyor; liquidctl-1.5.0-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
...
[DEBUG] __main__: device: NZXT Smart Device V2
[DEBUG] liquidctl.driver.usb: writting report 0x28 with 63 bytes: 03:01:00:05:01:10:00:02:00:02:00:03:02:01:03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00`

Hely0n, 2021-09-28:
For comparison with "alternating-3"

.\liquidctl.exe -m smart set led1 color alternating-3 "hsv(200,100,1)" "hsv(200,80,1)" --speed slower --verbose --debug

> [DEBUG] liquidctl.driver.usb: writting report 0x28 with 63 bytes: 03:01:00:05:01:00:00:02:00:02:00:03:02:01:03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00

Hely0n, 2021-09-28:
After introducing liquidctl in November 2020, I uninstalled NZXT Cam and never installed it again, so there shouldn't be any firmware updates installed

closed 2021-09-28 by jonasmalacofilho

Hely0n, 2021-09-28:
It works! Thank you. I'm fascinated how quick you solved the problem :) .

jonasmalacofilho, 2021-09-28:
You both did the hard part of confirming and isolating the problem, I just came in with a fresh set of eyes (and possibly a rubber duck).

commit b94b597a (2021-09-28): smart_device: fix out-of-bounds moving flag in SD2/HUE2 devices

Fixes: 327a72d7208d ("changed nzxt devices to use the direction flag")
Fixes: #385 ("NZXT Smart Device V2: 'moving-alternating' mode does just
'alternating' since 1.5.0")