liquidctl/liquidctl #386: Windows XP support?

Wasmachineman-NL, 2021-09-26:
I have a NZXT E500 in my Pentium M XP rig and I was wondering: can I use liquidctl to check what amount of power my E500 is using, under XP with something like HWiNFO?

jonasmalacofilho, 2021-09-28:
> I have a NZXT E500 in my Pentium M XP rig and I was wondering: can I use liquidctl to check what amount of power my E500 is using, under XP

Probably, although some of our dependencies may not support XP anymore.

> with something like HWiNFO?

What do you mean?  Using the [LQiNFO script](https://github.com/liquidctl/liquidctl/blob/main/extra/windows/LQiNFO.py) for supplying HWiNFO with liquidctl data?

Wasmachineman-NL, 2021-09-30:
I guess so, NZXT CAM doesn't run under XP obviously.

jonasmalacofilho, 2021-10-01:
Well, it turns out that our minimum Python 3 required version (3.6) already does _not_ run on Windows XP.

---

Can I ask what are you using a desktop/home server Pentium M platform running Windows XP for these days?

MarshallAsch, 2021-10-01:
> Can I ask what you are using a desktop/home server Pentium M platform running Windows XP for these days?

Im curious about the answer to this too

Wasmachineman-NL, 2021-10-01:
Partially based on my first computer I had back in 2003 (that had a Pentium 4 though), I mostly use it for older games.

jonasmalacofilho, 2021-10-01:
Oh, cool!

---

Well, sorry we can't support Windows XP.

You could connect the PSU data to another machine (even something like a Raspberry Pi) and monitor it from there...

closed 2021-10-01 by jonasmalacofilho