liquidctl/liquidctl #388: Gigabyte M32U monitor

kelvie, 2021-10-04:
It's a monitor, that uses OEM software (Gigabyte OSD SIdekick, Windows-only of course) over USBHID to control monitor settings, like brightness, contrast, etc.

It has built in KVM functionality, so it works as a USB hub to up to two devices over usb-c and usb 3.1 type b (in both cases, USB 3.1 gen 2, SS10).

The device appears to support the following features:

 - [ ] monitoring of temperatures or other environment sensors
 - [ ] monitoring of fan or pump speeds or duty cycles
 - [ ] monitoring of voltages, currents or power
 - [ ] configurable fan or pump speeds
 - [ ] configurable voltages or current/power limits
 - [ ] configurable lighting of embedded LEDs
 - [ ] configurable lighting of accessories like RGB fans or LED strips

The device is connected to the host system using:

- [ x] USB
- [ ] PCI-E
- [ ] onboard (the motherboard)
- [ ] other (please elaborate)
- [ ] unknown

The device communicates with the host system using:

- [ x] HID
- [ x] USB
- [ ] I²C or SMBus
- [ ] other (please elaborate)
- [ ] unknown

Please also include any useful additional information, such as USB vendor and product IDs, the output of `lsusb -v` (Linux) or `system_profiler SPUSBDataType` (Mac OS), links to external resources or already collected traffic data.

ID 0bda:1100 Realtek Semiconductor Corp. HID Device

I have some notes here:

https://gist.github.com/kelvie/fa562c4643c4abc8d91bb192b325995b

And wrote a quick and dirty CLI command in go:
https://github.com/kelvie/gbmonctl/blob/main/main.go

But it would be cleaner to integrate with a tool like this. Not sure if we want to add "monitors" to the list of things this tool supports, though, as it's pretty esoteric to control monitors over HID.

---

Finally, I can help with:¹

- [x] testing changes on Linux
- [ ] testing changes on Windows
- [ ] testing changes on Mac OS
- [x] attempting to capture USB traffic
- [ ] attempting to capture I²C/SMBus traffic
- [x] analyzing traffic data
- [x] documenting the protocol
- [x] implementing the driver

¹ Assuming documentation and/or assistance will be provided.

jonasmalacofilho, 2021-10-05:
I'm not opposed to it, especially since a few monitors with [ambient/accent] RGB [lighting] would probably work in a similar way.

That said, I don't think we currently have a good way of taking these types of parameters (brightness, contrast, sharpness) in our CLI.  We might need something like #297.

kelvie, 2021-10-05:
Ah alright. I haven't looked at what the CLI would look like yet. Anyway, once this is figured out, you can ping me, and I can help get this all done.

There are actually a lot of other parameters that can be set over HID, so it would be good to have a generic way of adding custom parameters.

jonasmalacofilho, 2022-03-03:
Have you considered looking if OpenRGB supports this monitor, and maybe contributing it there?

It would probably be a bit more suited for this type of stuff.

closed 2022-04-05 by jonasmalacofilho