liquidctl/liquidctl #392: Can't figure out syntax for NZXT accessories

wolfgangrumpf, 2021-10-29:
I have an NZXT Kraken X53 pump connected to an NZXT AER RGB fan and and two NZXT lightstrips.  When I initialize all, I get:

`NZXT Kraken X (X53, X63 or X73)
├── Firmware version                     1.8.0  
├── LED accessory 1           AER RGB 2 140 mm  
├── LED accessory 2     HUE 2 LED Strip 300 mm  
├── LED accessory 3     HUE 2 LED Strip 300 mm  
├── Pump Logo LEDs                    detected  
└── Pump Ring LEDs                    detected  
`
I can control the pump logo and ring separately pretty easily using:

`sudo liquidctl --match kraken set ring color rainbow-flow --speed faster
sudo liquidctl --match kraken set logo color fixed 03fcec
`
But when I try to control the fan and/or light strips I don't know how to tell it which device.  I've tried LED1 and that doesn't work.  I can set them all (two lightstrips and the fan) with:

`sudo liquidctl --match kraken set external color spectrum-wave # or super-rainbow
`
But what is the synatax for controlling JUST the fan (accessory 1) or the lightstrips (accessories 2 and 3)?

jonasmalacofilho, 2021-10-29:
All accessories in a HUE2 channel share the same lighting mode. To clarify, that's a hardware&firmware limitation, and there's nothing we can do to mitigate it.

That said, the `super-fixed` and `super-breathing` modes allow you to set each individual LED in the channel, effectively giving you almost arbitrary control over each accessory. Combining those with some scripting (either using the Python APIs or through your shell) can allow fairly complex effects.

wolfgangrumpf, 2021-10-29:
Okay, that makes sense.  Thanks for the quick response!

closed 2021-10-29 by jonasmalacofilho