liquidctl/liquidctl #393: How to use liquidctl with corsair h100i

MrMcyeet, 2021-11-13:
This is less of a bug, and probably just me being dense, but when I install Liquidctl, my h100i isn't detected. 
I see on the install part of the readme it says 
>  [...] Requires replacing the device driver [on Windows](https://github.com/liquidctl/liquidctl#installing-on-windows).

about the h100i, but when clicking on that, I cant really make much sense of what I'm supposed to do, could someone help me understand what I'm supposed to do?

I think its saying that as of issue #370 you aren't supplying prebuilt executables, so I need to build it myself on windows, then do what?

jonasmalacofilho, 2021-11-13:
Can you first show me the output of `liquidctl list --debug`?

Corsair uses very similar names, and only a few models require these special drivers.

---

> > [...] Requires replacing the device driver [on Windows](https://github.com/liquidctl/liquidctl#installing-on-windows).
> 
> about the h100i, but when clicking on that, I cant really make much sense of what I'm supposed to do, could someone help me understand what I'm supposed to do?

That part means that, besides installing liquidctl itself and the LibUSB DLL, you also need to go through the following steps to replace the system driver used by that device:

> Additionally, products that are not Human Interface Devices (HIDs), or that do not use the Microsoft HID Driver, require a libusb-compatible driver, see notes in [Supported devices](#supported-devices)).  In most cases Microsoft WinUSB is recommended, which can easily be set up for a device with [Zadig](https://zadig.akeo.ie/):¹ open the application, click `Options`, `List All Devices`, then select your device from the dropdown list, and click "Replace Driver".  Note that replacing the driver for devices that do not require it will likely cause them to disappear from liquidctl.

MrMcyeet, 2021-11-13:
Sure, the output of that command shows
<details>
  <summary>Command output</summary>

```
mrmcyeet@voidium:~$ liquidctl list --debug
[DEBUG] liquidctl.cli: running liquidctl v1.5.1
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: ignoring 0-0052, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-3
[DEBUG] liquidctl.driver.smbus: ignoring 0-0050, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-1
[DEBUG] liquidctl.driver.smbus: ignoring 0-0037, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: ignoring 0-0053, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-4
[DEBUG] liquidctl.driver.smbus: ignoring 0-0051, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-2
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0
[DEBUG] liquidctl.driver.ddr4: instanced driver for Corsair CMW32GX4M4Z3200C16 DIMM1 (experimental)
[DEBUG] liquidctl.driver.ddr4: instanced driver for Corsair CMW32GX4M4Z3200C16 DIMM2 (experimental)
[DEBUG] liquidctl.driver.ddr4: instanced driver for Corsair CMW32GX4M4Z3200C16 DIMM3 (experimental)
[DEBUG] liquidctl.driver.ddr4: instanced driver for Corsair CMW32GX4M4Z3200C16 DIMM4 (experimental)
[DEBUG] liquidctl.driver.smbus: ignoring 0-0036, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-5
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1b7e
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1b7e
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1b7e
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1b7e
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1b7e
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1b7e
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0a45
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0a45
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0a3e
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0a3e
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0a3e
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0a3e
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0a3e
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1b7f
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1b7f
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1b7f
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1b7f
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1b7f
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1b7f
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1b7f
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1b7f
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1b33
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1b33
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1b33
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1b33
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:1b33
[DEBUG] liquidctl.driver.usb: found HID device 2833:0051
[DEBUG] liquidctl.driver.usb: found HID device 2833:0051
[DEBUG] liquidctl.driver.usb: found HID device 2833:0051
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 2833:2051
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 2833:0051
[DEBUG] liquidctl.driver.usb: found USB device 2833:2052
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:1b33
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:1b7f
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0a3e
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0a45
[DEBUG] liquidctl.driver.usb: found USB device 8087:0a2a
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c15
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0608
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:1b7e
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
Device ID 0: Corsair CMW32GX4M4Z3200C16 DIMM1 (experimental)
├── Bus: i2c-0
├── Address: 0x50
└── Driver: Ddr4Temperature
[DEBUG] liquidctl.cli: hierarchy: SmbusDriver, BaseDriver

Device ID 1: Corsair CMW32GX4M4Z3200C16 DIMM2 (experimental)
├── Bus: i2c-0
├── Address: 0x51
└── Driver: Ddr4Temperature
[DEBUG] liquidctl.cli: hierarchy: SmbusDriver, BaseDriver

Device ID 2: Corsair CMW32GX4M4Z3200C16 DIMM3 (experimental)
├── Bus: i2c-0
├── Address: 0x52
└── Driver: Ddr4Temperature
[DEBUG] liquidctl.cli: hierarchy: SmbusDriver, BaseDriver

Device ID 3: Corsair CMW32GX4M4Z3200C16 DIMM4 (experimental)
├── Bus: i2c-0
├── Address: 0x53
└── Driver: Ddr4Temperature
[DEBUG] liquidctl.cli: hierarchy: SmbusDriver, BaseDriver
```
</details>

Some additional context that I think might be useful, If I run the same command on [OpenCorsairLink](https://github.com/audiohacked/OpenCorsairLink), it outputs 
<details>
  <summary>Command output</summary>

```
mrmcyeet@voidium:~$ OpenCorsairLink.elf --debug
Checking USB device 0 (1d6b:0003)...
Checking USB device 1 (1d6b:0002)...
Checking USB device 2 (2833:2051)...
Checking USB device 3 (1d6b:0003)...
Checking USB device 4 (2833:0051)...
Checking USB device 5 (2833:2052)...
Checking USB device 6 (1b1c:1b33)...
Checking USB device 7 (1b1c:1b7f)...
Checking USB device 8 (1b1c:0a3e)...
Checking USB device 9 (1b1c:0a45)...
Checking USB device 10 (8087:0a2a)...
Checking USB device 11 (1b1c:0c15)...
Corsair product detected. Checking if device is H100i Pro... Could not open device 6940:3093.Checking USB device 12 (05e3:0608)...
Checking USB device 13 (1b1c:1b7e)...
Checking USB device 14 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: selected device_number = -1

```
</details>

jonasmalacofilho, 2021-11-13:
> ```
> [DEBUG] liquidctl.cli: running liquidctl v1.5.1
> ```

You need at least liquidctl 1.6.x for support for the H100i **Pro.**

Another thing I can see is that you're on Linux...  
...are you also dual booting Windows, and is that why you asked the Windows driver?

---

> I have 4x corsair RGB memory sticks install, along with a corsair kb and mouse. I have no expectation of those being controllable with this, but I thought itd be useful, since they are also corsair.

The keyboard and mouse aren't supported, but liquidctl should be able to read the temperature from those memory modules.  But for that you would need to enable an [unsafe feature](https://github.com/liquidctl/liquidctl/blob/main/docs/ddr4-guide.md#inherent-unsafeness-of-i2c-and-smbus).

```
$ liquidctl status --unsafe smbus,ddr4_temperature
```

MrMcyeet, 2021-11-13:
> 
> 
> > ```
> > [DEBUG] liquidctl.cli: running liquidctl v1.5.1
> > ```
> 
> You need at least liquidctl 1.6.x for support for the H100i **Pro.** If you can, please use the latest version (1.7.2).
> 
> Another thing I can see is that you're on Linux... ...are you also dual booting Windows, and is that why you asked about the Windows driver?
> 
> > I have 4x corsair RGB memory sticks install, along with a corsair kb and mouse. I have no expectation of those being controllable with this, but I thought itd be useful, since they are also corsair.
> 
> The keyboard and mouse aren't supported, but liquidctl should be able to read the temperature from those memory modules. But for that you would need to enable an [unsafe feature](https://github.com/liquidctl/liquidctl/blob/main/docs/ddr4-guide.md#inherent-unsafeness-of-i2c-and-smbus).
> 
> ```
> $ liquidctl status --unsafe smbus,ddr4_temperature
> ```

I am indeed dual booting windows atm, I recently switched over to using Kubuntu, but decieded to keep windows installed for games like forza and cod, but eventually I want to just visualize windows instead of dual booting, but ill figure that all out another day haha. Anyway, if doing something with the device on windows is required, I have no issues switching over and doing whatever it is.

And I didnt know that the H100i Pro required a different version, I installed this version because of
[image]
I think that this shows that for ubuntu (Kubuntu is just ubuntu with the KDE Plasma desktop, I think) that I need version 1.5.1.

I will try it with the latest version and report back here with my findings

jonasmalacofilho, 2021-11-13:
> I am indeed dual booting windows atm [...]

Oh, ok then.

> Anyway, if doing something with the device on windows is required, I have no issues switching over and doing whatever it is.

But just to clarify, the procedure with Zadig is only necessary if you plan to use liquidctl on Windows.

> I think that this shows that for ubuntu (Kubuntu is just ubuntu with the KDE Plasma desktop, I think) that I need version 1.5.1.

That table simply shows the liquidctl version that already is in those distributions' official repositories.

We currently have no "version less than" type of requirements.  (That would be very unusual... especially within the same major version number "1.x.x").

jonasmalacofilho, 2021-11-15:
> I will try it with the latest version and report back here with my findings

How did it work out?

closed 2021-11-19 by jonasmalacofilho