liquidctl/liquidctl #394: how shuld i run it on ubuntu?

PsycholabLove, 2021-11-19:
i already install the package, when i run the command `liquidctl list` 
i was told 

Traceback (most recent call last):
  File "/usr/lib/command-not-found", line 27, in <module>
    from CommandNotFound.util import crash_guard
ModuleNotFoundError: No module named 'CommandNotFound'
root@psycholab:/home/liu# liquidctl initialize

what should I do?

jonasmalacofilho, 2021-11-19:
Hi,

```
/usr/lib/command-not-found
```

seems to be broken on your system.   (It's a Ubuntu thing).

---

Regardless, it also seems that there's no `liquidctl` executable in (a any of the directories in) your `PATH` (environment variable).  (This triggers the "command not found" convenience handler that's broken).

I think the most likely causes are:

- you haven't installed liquidctl in the global environment, and the virtual environment is not in the `PATH`;
- you're trying to execute `liquidctl` as `root`, and `root` has a different `PATH` than you/what you're used to;
- liquidctl hasn't actually been installed (e.g. you missed an error message).

PsycholabLove, 2021-11-19:
hello thanks for your response, I try to figure it out now.

closed 2021-11-23 by jonasmalacofilho