liquidctl/liquidctl #395: Fan speed stuck after waking PC from sleep

nekato, 2021-11-29:
If I enter ```liquidctl set fan speed 25```, fan speed will always be around 532 rpm. But when I wake PC from sleep it shows 473 rpm. If I enter it again, rpm doesn't change. It's the same with ```liquidctl set fan speed 30```. Only on ```liquidctl set fan speed 35``` will change and show 624 rpm. Then I can enter ```liquidctl set fan speed 25``` and it will show 532 rpm again.

Additional information about the system:

- Affected device: Kraken X62
- Operating system: Windows 11
- Liquidctl version: v1.7.2

nekato, 2021-11-29:
I wanted to check if it happens with CAM. I first tried with version 3.7.8 and it does. But when I tried the latest CAM (4.30.2) it doesn't, it goes to 530rpm immediately. Weird.

jonasmalacofilho, 2021-11-29:
Hi @nekato,

I tried but couldn't reproduce this behavior.  Note: I'm on Linux and using aftermarket fans.

So, some things come to mind...

That duty cycle is fairly low for the stock fans: usually, when it's too low, the fans just wont be able to start spinning again, but maybe these NZXT ones behave a bit differently.  Also, can you see if _both_ of them are spinning when stuck at ~470 rpm?

Another possible cause is an invisible interaction between fan and pump speeds (this can happens because how the cooler requires us to encode these settings, at a low level).  Can you try, instead of fan speed 35, setting some _pump_ speed (to whatever value you're comfortable with) and see if that allows a subsequent fan speed 25 to work as expected?

jonasmalacofilho, 2021-11-29:
Can you also report the liquidctl and firmware versions?

nekato, 2021-11-29:
Liquidctl version is 1.7.2 and firmware version is 6.0.2. That is in liquidctl status, but in CAM it says 6.2.

I observed fans when waking up PC and both start spinning. Also tried setting pump speed to every value from 50 to 100 but it didn't affect the results. Pump speed stays the same after waking.

I noticed that setting fan speed at 35 sounds completely the same as at 25%, so I'll start my curve with that. So it's not a problem. 

Thank you Jonas for this great tool.

jonasmalacofilho, 2021-11-29:
> Liquidctl version is 1.7.2 and firmware version is 6.0.2.

Good, same here.

> That is in liquidctl status, but in CAM it says 6.2.

That's ok: a while back they changed how they display these version numbers, but I decided to *not* follow them.

> Pump speed stays the same after waking.

But only while the fan speed is also stuck, right?  Afterward it does respond to new settings, correct?

---

> Thank you Jonas for this great tool.

Thanks!

nekato, 2021-11-29:
> > Pump speed stays the same after waking.
> 
> But only while the fan speed is also stuck, right? Afterward it does respond to new settings, correct?

Pump speed is not affected like fan speed. Speed is not reduced after wake up, while fan speed is, and responds to any new settings.

jonasmalacofilho, 2021-11-29:
> Pump speed is not affected like fan speed. Speed is not reduced after wake up, while fan speed is, and responds to any new settings.

Got it.  Thanks!

---

I'm glad that this issue isn't super pressing for you, but I'll keep it open as there seems to be a bug somewhere.

nekato, 2021-11-30:
Sorry, I got one more question. I created a batch file to use for login but when I try to test the file by double clicking it, cmd just keeps looping the first line.

jonasmalacofilho, 2021-11-30:
I'm not sure what you mean my cmd being stuck in a loop.

Can you enable `--debug` on all `liquidctl` incantations, and also call the batch file from a previously open shell so we can see what's going on?  Also, what does your batch file look like?

But please open a new issue for it.

jonasmalacofilho, 2022-02-13:
At the time, what would be a typical coolant temperature just after resuming from a extended sleep?

jonasmalacofilho, 2022-02-13:
Another question: could there have been any CAM service running at the same time as liquidctl?

jonasmalacofilho, 2022-02-27:
I looked into this a couple of weeks ago while working on other stuff, and I did find one bug that could maybe explain this behavior.  Can you give the latest code in the main branch a try?

Note that now it's essential to `initialize` the device after the system boots or resumes from sleeping.  (And the bug/fix has to do with taking into account that the cooler may have reset to a configuration not totally compatible with how we normally set speeds).

jonasmalacofilho, 2022-03-04:
@nekato did you get a chance to test the latest code in `main` (with calls/invocations of `initialization` is the appropriate times)?

closed 2022-03-25 by jonasmalacofilho