liquidctl/liquidctl #396: Kraken X42 not recognized in liquidctl > v1.3.2

JojoBie, 2021-12-03:
I recognized some while ago newer versions of liquidctl does not recognize my Kraken x42 anymore.
version v1.3.2 did well.

python3 -m /liquidctl.cli --version; python3 -m liquidctl.cli list --verbose 

liquidctl v1.3.2 (fb9ec1fdb0ce-dirty):
Device ID 0: NZXT Kraken X (X42, X52, X62 or X72)
├── Vendor ID: 0x1e71
├── Product ID: 0x170e
├── Release number: 0x0200
├── Serial number: XXXYYYYZZZZ
├── Bus: usb1
├── Address: 4
├── Port: 7
└── Driver: KrakenTwoDriver using module usb

but no device found in v.1.7.2!!!

liquidctl v1.7.2 (2b4376891c2d):
\<no output\>

jonasmalacofilho, 2021-12-03:
Are you on Windows and, if you so, did you ever use Zadig (or some other tool) to replace the stock Windows driver?

JojoBie, 2021-12-03:
----------------------------------------------------------------------------------------------------------------
# /usr/src/liquidctl_1_3_2# python3 -m /liquidctl.cli list --debug

[DEBUG] __main__: running liquidctl v1.3.2 (fb9ec1fdb0ce-dirty)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=usb, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x174c:0x3074
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x170e
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b1c:0x1b2d
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b1c:0xa18
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x174c:0x2074
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b1c:0x1b3e
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xb05:0x1867
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x174c:0x3074
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b1c:0x1b2d
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b1c:0xa18
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x174c:0x2074
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b1c:0x1b3e
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xb05:0x1867
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
Device ID 0: NZXT Kraken X (X42, X52, X62 or X72)
├── Vendor ID: 0x1e71
├── Product ID: 0x170e
├── Release number: 0x0200
├── Serial number: 5D85146B343
├── Bus: usb1
├── Address: 4
├── Port: 7
└── Driver: KrakenTwoDriver using module usb
[DEBUG] __main__: hierarchy: UsbHidDriver, BaseUsbDriver, BaseDriver; PyUsbHid

-----------------------------------------------------------------------------------------------------------------
# >/usr/src/liquidctl# python3 -m /liquidctl.cli list --debug

[DEBUG] cli (main): running liquidctl v1.7.2 (2b4376891c2d)
[DEBUG] smbus (find_devices): searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] smbus (find_devices): found I²C bus i2c-3
[DEBUG] smbus (find_devices): found I²C bus i2c-10
[DEBUG] smbus (find_devices): found I²C bus i2c-1
[DEBUG] smbus (find_devices): found I²C bus i2c-17
[DEBUG] smbus (find_devices): found I²C bus i2c-8
[DEBUG] smbus (find_devices): found I²C bus i2c-15
[DEBUG] smbus (find_devices): found I²C bus i2c-6
[DEBUG] smbus (find_devices): found I²C bus i2c-13
[DEBUG] smbus (find_devices): found I²C bus i2c-4
[DEBUG] smbus (find_devices): found I²C bus i2c-11
[DEBUG] smbus (find_devices): found I²C bus i2c-2
[DEBUG] smbus (find_devices): found I²C bus i2c-0
[DEBUG] smbus (find_devices): found I²C bus i2c-9
[DEBUG] smbus (find_devices): found I²C bus i2c-16
[DEBUG] smbus (find_devices): found I²C bus i2c-7
[DEBUG] smbus (find_devices): found I²C bus i2c-14
[DEBUG] smbus (find_devices): found I²C bus i2c-5
[... 295 lines cut ...]
            ID 1b1c:0a18 Corsair
            /sys/bus/usb/devices/1-6.2  /dev/bus/usb/001/005
        |__ Port 2: Dev 5, If 3, Class=Human Interface Device, Driver=usbhid, 12M
            ID 1b1c:0a18 Corsair
            /sys/bus/usb/devices/1-6.2  /dev/bus/usb/001/005
        |__ Port 2: Dev 5, If 1, Class=Audio, Driver=snd-usb-audio, 12M
            ID 1b1c:0a18 Corsair
            /sys/bus/usb/devices/1-6.2  /dev/bus/usb/001/005
        |__ Port 2: Dev 5, If 2, Class=Audio, Driver=snd-usb-audio, 12M
            ID 1b1c:0a18 Corsair
            /sys/bus/usb/devices/1-6.2  /dev/bus/usb/001/005
        |__ Port 3: Dev 7, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            ID 1b1c:1b2d Corsair K95 RGB Platinum Keyboard [RGP0056]
            /sys/bus/usb/devices/1-6.3  /dev/bus/usb/001/007
        |__ Port 3: Dev 7, If 1, Class=Human Interface Device, Driver=usbhid, 12M
            ID 1b1c:1b2d Corsair K95 RGB Platinum Keyboard [RGP0056]
            /sys/bus/usb/devices/1-6.3  /dev/bus/usb/001/007
    |__ Port 7: Dev 4, If 0, Class=Human Interface Device, Driver=, 12M
        ID 1e71:170e NZXT Kraken X
        /sys/bus/usb/devices/1-7  /dev/bus/usb/001/004
    |__ Port 14: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 0b05:1867 ASUSTek Computer, Inc.
        /sys/bus/usb/devices/1-14  /dev/bus/usb/001/006  

-----------------------------------------------------------------------------------------------------------------

JojoBie, 2021-12-03:
i'am sorry... just saw i patched the 1.3.2 version using the deprecated option "--hid=usb".
so it's the version-change using libhidapi-hidraw you mentioned!

JojoBie, 2021-12-03:
any suggestion how i can get this with libhidapi-hidraw running?

JojoBie, 2021-12-03:
X42 doesn't bind to an usbhid driver... Don't know why. Did not find any udev-rules that might block this.
:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    /sys/bus/usb/devices/usb1  /dev/bus/usb/001/001
    |__ ...
    |__ Port 5: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 1b1c:1b3e Corsair
        /sys/bus/usb/devices/1-5  /dev/bus/usb/001/002
    |__ Port 6: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
		...
		...
        |__ Port 3: Dev 7, If 1, Class=Human Interface Device, Driver=usbhid, 12M
            ID 1b1c:1b2d Corsair K95 RGB Platinum Keyboard [RGP0056]
            /sys/bus/usb/devices/1-6.3  /dev/bus/usb/001/007
    |__ Port 7: Dev 4, If 0, Class=Human Interface Device, Driver=, 12M
        ID 1e71:170e NZXT Kraken X
        /sys/bus/usb/devices/1-7  /dev/bus/usb/001/004

jonasmalacofilho, 2021-12-03:
> any suggestion how i can get this with libhidapi-hidraw running?

Bind the device to the `usbhid` driver and/or prevent it from being left unbound by other tools or even old liquidctl.

---

> X42 doesn't bind to an usbhid driver... Don't know why. Did not find any udev-rules that might block this.

Can you attach (or send me) your kernel log for one session?

JojoBie, 2021-12-03:
changed usb-port and cable. no effect. perhaps a problem with the x42-usb-interface itself. log via email.

jonasmalacofilho, 2021-12-03:
> log via email.

Got it, thanks.

> perhaps a problem with the x42-usb-interface itself

Doesn't look like it.  Perhaps a kernel bug?

This could also be related to the use of Proxmox; I also maintain PyUSB, and a weird issue has popped up there which appears to be caused by Proxmox.

I'm not sure how detailed the debug output from the usb and hid subsystems are by default, but can you try enabling it?  There's the `debug=1` boot parameter, and you also probably have access to [dynamic debug](https://www.kernel.org/doc/html/latest/admin-guide/dynamic-debug-howto.html).

jonasmalacofilho, 2021-12-03:
Of course, I'm assuming you haven't blacklisted any modules that could affect that device.

Also, I would expect the `nzxt_kraken2` driver to bind to the device as well on Linux 5.13+.  But maybe it has been configured out of that particular kernel build?

closed 2022-01-14 by jonasmalacofilho