liquidctl/liquidctl #397: Unable to set fan speed on Corsair H100i Pro

karlsolomon, 2021-12-04:
On Corsair H100i Pro Fan speeds do not reflect changes to Fan duty cycle/pwm.

Initially, I've set I've set Fan1 speed to 20 and Fan2 speed to 100. However, Fan1 and Fan2 speeds are nearly identical (1192 vs 1187 rpm)
```
$ liquidctl status
Corsair Hydro H100i Pro XT (experimental)
├── Liquid temperature    33.0  °C
├── Fan 1 speed           1192  rpm
├── Fan 1 duty              20  %
├── Fan 2 speed           1187  rpm
├── Fan 2 duty             100  %
├── Pump speed            1951  rpm
└── Pump duty               55  %
```

I then set Fan1 speed to 100 and Fan2 speed to 20, and wait ~1 minute.
```
$ liquidctl set fan1 speed 100
$ liquidctl set fan2 speed 20
// wait ~ 1 minute
```

After this, although Fan1/2 duty cycles show change, the Fan1/2 speeds are unchanged.
```
$ liquidctl status
Corsair Hydro H100i Pro XT (experimental)
├── Liquid temperature    33.0  °C
├── Fan 1 speed           1192  rpm
├── Fan 1 duty             100  %
├── Fan 2 speed           1188  rpm
├── Fan 2 duty              20  %
├── Pump speed            1951  rpm
└── Pump duty               55  %
```

Setup: installed latest release (1.7.2 from pacman)
```
$ liquidctl --version
liquidctl v1.7.2 (Arch Linux; 1.7.2-1)
$ liquidctl list --verbose
Device #0: Corsair Hydro H100i Pro XT (experimental)
├── Vendor ID: 0x1b1c
├── Product ID: 0x0c20
├── Release number: 0x0100
├── Bus: hid
├── Address: /dev/hidraw1
└── Driver: HydroPlatinum
```

For what it's worth, I uninstalled liquidctl and cloned the repo to try the same sequence on a few older releases (1.7.1, 1.6.1, and 1.4.1) w/ `git checkout v1.6.1` and I was unable to change fan speed on any of these versions (I saw similar/same behavior as in v1.7.2).

I am, however, able to see change to pump speed/duty when using `liquidctl initialize --pump-mode`.

jonasmalacofilho, 2021-12-05:
H @karlsolomon,

Can you re-run these commands with `--debug` and post the output?  And can you also try with profiles and see if the issue still manifests that way?

Also, just to check, did you remember to `initialize` the device before trying to set the fan speeds?

karlsolomon, 2021-12-05:
Hi @jonasmalacofilho, thanks for following up so quickly!

did you remember to initialize
>Yes, I usually do `liquidctl initialize --pump-mode=quiet` just before trying to set fan speeds
My typical initialization script looks something like this.
```
#!/bin/bash

liquidctl initialize --pump-mode quiet
liquidctl set fan speed  30 30  40 50  45 80  50 90  60 100
```

Can you re-run these commands
>Please see attached liquidctl_397.log with `--debug` output.
The commands run in-order are as follows
```
$ liquidctl --version
$ liquidctl status --debug
$ liquidctl initialize all --debug
$ liquidctl status --debug
$ liquidctl set fan1 speed 100 --debug
$ liquidctl set fan2 speed 20 --debug
$ liquidctl status --debug
$ liquidctl status --debug
$ liquidctl list --verbose --debug
```
[liquidctl_397.log](https://github.com/liquidctl/liquidctl/files/7656560/liquidctl_397.log)

jonasmalacofilho, 2021-12-05:
Thanks!  Unfortunately nothing obviously wrong popped up there.

---

Does the issue still manifest when using the profile from your initialization script, or is it specific to the fixed speed mode?

Can you try iCue (in a VM, second system or by dual booting) and see if the issue still happens with it?

Oh, and are you by any chance using aftermarket fans?

jonasmalacofilho, 2021-12-05:
One more question: can you hear the fans spinning up from 20% to 100%?

karlsolomon, 2021-12-05:
Does the issue still manifest when using the profile from your initialization script, or is it specific to the fixed speed mode?
> same issue, fan speed is independent of fan duty cycle/pwm

Can you try iCue
> yes, I'll follow up w/ results. Don't have a windows distribution/virtualbox installed at the moment

aftermarket fans
> Not to my knowledge, ordered from [this](https://www.amazon.com/gp/product/B0829RTMKY/ref=ppx_yo_dt_b_asin_title_o06_s01?ie=UTF8&psc=1) link on amazon

can you hear the fans spinning
> Using [this](https://bafxpro.com/products/bafx-products-decibel-meter-sound-level-reader-w-battery-advanced-sound-meter) to measure. Read ~42-42.5dB at 1 meter with both fans at 20% pwm. Same dB ~42.5 with both fans at 100% pwm. I am, audibly, also unable to distinguish a difference.

jonasmalacofilho, 2021-12-06:
Thanks for those additional details.

It's possible that the protocol has changed in more recent firmware (I'm assuming your device may be newer than what our driver was tested on), or that you've hit a yet unknown corner case.

Finding out whether iCue has the same issue will determinate the next steps; if it doesn't I'll ask you to [take some traffic captures](https://github.com/liquidctl/liquidctl/blob/main/docs/developer/capturing-usb-traffic.md) so I can try to find out what needs to be fixed.

_(Also, it's a longshot, but merely using iCue once could maybe solve the problem... for example, if it's some type of firmware bug/quirk that has already been patched/changed by Corsair)._

---

P.S. I'm not sure if you're aware and intentionally doing it differently, but in Markdown (and conventions for plain text email on which Markdown was based on) `>` is normally used for quotations (you're using it for your responses).

jonasmalacofilho, 2022-03-12:
Hi @karlsolomon,

Can you still reproduce this issue with a current version of liquidctl (preferably the code in our main branch)?

Additionally, are you using stock or third-party fans, and have you double checked their electrical connections?

jonasmalacofilho, 2022-07-03:
Triage: closing due to lack of feedback.

closed 2022-07-03 by jonasmalacofilho