liquidctl/liquidctl #400: Corsair memory protocol error

wael444, 2021-12-10:
```
~ $ liquidctl --address 0x53 set led color breathing 000000 --unsafe=smbus,vengenance_rgb --debug
[DEBUG] cli (main): running liquidctl v1.7.2 (Arch Linux; 1.7.2-1)
[DEBUG] smbus (find_devices): searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] smbus (find_devices): found I²C bus i2c-3
[DEBUG] smbus (find_devices): found I²C bus i2c-1
[DEBUG] ddr4 (probe): instanced driver for Corsair CMW16GX4M2D3600C18 DIMM4
[DEBUG] smbus (find_devices): ignoring 1-0037, cannot infer bus number
[DEBUG] smbus (find_devices): ignoring 1-0053, cannot infer bus number
[DEBUG] smbus (find_devices): found I²C bus i2c-6
[DEBUG] smbus (find_devices): ignoring 1-0051, cannot infer bus number
[DEBUG] smbus (find_devices): found I²C bus i2c-4
[DEBUG] smbus (find_devices): found I²C bus i2c-2
[DEBUG] smbus (find_devices): found I²C bus i2c-0
[DEBUG] smbus (find_devices): ignoring 1-0036, cannot infer bus number
[DEBUG] smbus (find_devices): ignoring 0-0008, cannot infer bus number
[DEBUG] smbus (find_devices): found I²C bus i2c-5
[DEBUG] usb (find_devices): searching HidapiBus (CommanderCore, CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] usb (find_devices): searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc, CorsairAsetekProDriver)
[DEBUG] cli (main): device: Corsair CMW16GX4M2D3600C18 DIMM4
[INFO] cli (log_error):
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/liquidctl/cli.py", line 436, in main
    _device_set_color(dev, args, **opts)
  File "/usr/lib/python3.9/site-packages/liquidctl/cli.py", line 275, in _device_set_color
    dev.set_color(args['<channel>'].lower(), args['<mode>'].lower(), color, **opts)
  File "/usr/lib/python3.9/site-packages/liquidctl/driver/ddr4.py", line 242, in set_color
    raise NotSupportedByDriver()
liquidctl.error.NotSupportedByDriver
[ERROR] cli (log_error): Error: operation not supported by driver for Corsair CMW16GX4M2D3600C18 DIMM4
```
```
$ liquidctl status --unsafe=smbus,ddr4_temperature,vengenance_rgb --debug
...
...
[DEBUG] cli (main): device: Corsair CMW16GX4M2D3600C18 DIMM2
[INFO] cli (log_error): [Errno 71] Protocol error
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/liquidctl/cli.py", line 428, in main
    status = dev.get_status(**opts)
  File "/usr/lib/python3.9/site-packages/liquidctl/driver/ddr4.py", line 217, in get_status
    treg = self._read_temperature_register()
  File "/usr/lib/python3.9/site-packages/liquidctl/driver/ddr4.py", line 234, in _read_temperature_register
    return self._smbus.read_block_data(self._ts_address, self._REG_TEMPERATURE)
  File "/usr/lib/python3.9/site-packages/liquidctl/driver/smbus.py", line 129, in read_block_data
    data = self._smbus.read_block_data(address, register)
OSError: [Errno 71] Protocol error
[ERROR] cli (log_error): Unexpected OS error with Corsair CMW16GX4M2D3600C18 DIMM2: OSError: [Errno 71] Protocol error
...
```
seems to me this is a user error or something with my system.

Additional information about the system:
- Affected device: Corsair CMW16GX4M2D3600C18
- Operating system: Artix Linux
- Liquidctl version: v1.7.2 (Arch Linux; 1.7.2-1)

jonasmalacofilho, 2021-12-11:
That model number is for a Vengeance RGB **PRO,** which  uses a different protocol and is not yet supported by liquidctl (or, more specifically, by the `VengeanceRgb` driver).

> ```
> ~ $ liquidctl --address 0x53 set led color breathing 000000 --unsafe=smbus,vengenance_rgb --debug
> [ ... ]
> [ERROR] cli (log_error): Error: operation not supported by driver for Corsair CMW16GX4M2D3600C18 DIMM4
> ```

At that point, everything is still working as designed: the `Ddr4Temperature` was instantiated because the module's SPD advertises a temperature sensor, but that driver does not support RGB control.

> ```
> $ liquidctl status --unsafe=smbus,ddr4_temperature,vengenance_rgb --debug
> [ ... ]
> [ERROR] cli (log_error): Unexpected OS error with Corsair CMW16GX4M2D3600C18 DIMM2: OSError: [Errno 71] Protocol error
> ```

However, the second error was _not_ expected.  Can you give 7d2f1100ad15 a try?

---

P.S. Since only the `Ddr4Temperature` driver is used you don't need the `vengeance_rgb` unsafe feature.

jonasmalacofilho, 2021-12-11:
Actually, tests are failing, I messed something else in the process.  Please wait : )

jonasmalacofilho, 2021-12-11:
Please give dc6be89aa0e5 a try.

wael444, 2021-12-11:
```
$ liquidctl status --unsafe=smbus,ddr4_temperature,vengenance_rgb --debug
[DEBUG] cli (main): device: Corsair CMW16GX4M2D3600C18 DIMM2
[DEBUG] smbus (read_word_data): read word data @ 0x19:0x05 0x76c2
Corsair CMW16GX4M2D3600C18 DIMM2
└── Temperature    39.2  °C

[DEBUG] cli (main): device: Corsair CMW16GX4M2D3600C18 DIMM4
[DEBUG] smbus (read_word_data): read word data @ 0x1b:0x05 0x90c2
Corsair CMW16GX4M2D3600C18 DIMM4
└── Temperature    41.0  °C
...
```

```
$ liquidctl --address 0x53 set led color breathing 000000 --unsafe=smbus,vengenance_rgb,ddr4_temperature --debug
[DEBUG] cli (main): running liquidctl v1.7.2 (dc6be89aa0e5)
[DEBUG] smbus (find_devices): searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] smbus (find_devices): found I²C bus i2c-3
[DEBUG] smbus (find_devices): found I²C bus i2c-1
[DEBUG] ddr4 (probe): instantiated Ddr4Temperature driver for Corsair CMW16GX4M2D3600C18 DIMM4
[DEBUG] smbus (find_devices): ignoring 1-0037, cannot infer bus number
[DEBUG] smbus (find_devices): ignoring 1-0053, cannot infer bus number
[DEBUG] smbus (find_devices): found I²C bus i2c-6
[DEBUG] smbus (find_devices): ignoring 1-0051, cannot infer bus number
[DEBUG] smbus (find_devices): found I²C bus i2c-4
[DEBUG] smbus (find_devices): found I²C bus i2c-2
[DEBUG] smbus (find_devices): found I²C bus i2c-0
[DEBUG] smbus (find_devices): ignoring 1-0036, cannot infer bus number
[DEBUG] smbus (find_devices): ignoring 0-0008, cannot infer bus number
[DEBUG] smbus (find_devices): found I²C bus i2c-5
... 
[DEBUG] cli (main): device: Corsair CMW16GX4M2D3600C18 DIMM4
[INFO] cli (log_error):
Traceback (most recent call last):
  File "/home/wael/.local/lib/python3.9/site-packages/liquidctl/cli.py", line 436, in main
    _device_set_color(dev, args, **opts)
  File "/home/wael/.local/lib/python3.9/site-packages/liquidctl/cli.py", line 275, in _device_set_color
    dev.set_color(args['<channel>'].lower(), args['<mode>'].lower(), color, **opts)
  File "/home/wael/.local/lib/python3.9/site-packages/liquidctl/driver/ddr4.py", line 249, in set_color
    raise NotSupportedByDriver()
liquidctl.error.NotSupportedByDriver
[ERROR] cli (log_error): Error: operation not supported by driver for Corsair CMW16GX4M2D3600C18 DIMM4
```
also thank you for your help.

jonasmalacofilho, 2021-12-11:
Thanks for testing the fix (and for reporting the bug in the first place)!

Regarding support for RGB control of the Vengeance RGB Pro, it's not something I'm personally working on.  But if you're willing to work on it, let me know and I can help get you started.

On the other hand, if you just need something that already works, I think OpenRGB might support those modules.

commit 70630c6d (2021-12-11): ddr4: read word instead of SMBus Block Read for all TS SPDs

While JEDEC 21-C 4.1.6 uses the term "block read", it has little to do
with the SMBus Block Read protocol; instead, it is closer to the SMBus
Read Word protocol, except in big endianess.

Fixes: #400 ("Corsair memory protocol error")
closed 2021-12-11 by jonasmalacofilho