liquidctl/liquidctl #401: NZXT 510i case fan cannot go less than 1000 rpm

hifi25nl, 2021-12-11:
In my NZXT 510i with case fans _connected to NZXT controller_ I cannot set a speed lower than around 1000

/usr/bin/liquidctl --match "smart" set sync speed 0 -> stop fans
/usr/bin/liquidctl --match "smart" set sync speed 50  -> around 1000 rpm
/usr/bin/liquidctl --match "smart" set sync speed 100 -> max speed

With values less than 50 down to 1 the speed stays at around 1000 rpm

I don't understand if this is a limitation of liquidctl or of the NZXT controller

Additional information about the system:

- Affected device: NZXT 510i case controller
- Operating system: Arch Linux
- Liquidctl version: Arch Linux `liquidctl` package, version 1.7.2

jonasmalacofilho, 2021-12-11:
The "case" version fans are only controllable through variable DC, not PWM, so the limitation around ~50% is expected (and of the controller).

That said, and given that NZXT states their maximum speed (at 12 V) as 1200±200 rpm, 1000 rpm seems a bit high at ~50%.

Can you show me the output from `status` at both 50% and 100%?

hifi25nl, 2021-12-11:
`[audiolinux@archlinux ~]$ /usr/bin/liquidctl --match "smart" set sync speed 50
[audiolinux@archlinux ~]$ /usr/bin/liquidctl status
NZXT Smart Device V2
├── Fan 2 duty       50  %
├── Fan 2 speed    1136  rpm
├── Fan 3 duty       50  %
├── Fan 3 speed    1056  rpm
└── Noise level       0  dB

[audiolinux@archlinux ~]$ /usr/bin/liquidctl --match "smart" set sync speed 100
[audiolinux@archlinux ~]$ /usr/bin/liquidctl status
NZXT Smart Device V2
├── Fan 2 duty      100  %
├── Fan 2 speed    1503  rpm
├── Fan 3 duty      100  %
├── Fan 3 speed    1408  rpm
└── Noise level       0  dB
`

jonasmalacofilho, 2021-12-11:
Well, those are significant reductions of around 25%.  And it would be somewhat sensible for the controller to limit the minimum DC voltage to ~9 V (25% reduction from 12 V), if a bit conservative (in my experience most DC fans will still reliably start at ~7V).

And while it's a bit odd that the controller apparently maps 50% to 9V (so really 75%), it isn't unheard of: many PWM fans do this type of thing.

I guess there's nothing wrong here.  But could you also show me the output at 75%?

hifi25nl, 2021-12-11:
[audiolinux@archlinux ~]$ /usr/bin/liquidctl --match "smart" set sync speed 75
[audiolinux@archlinux ~]$ /usr/bin/liquidctl status
NZXT Smart Device V2
├── Fan 2 duty       75  %
├── Fan 2 speed    1315  rpm
├── Fan 3 duty       75  %
├── Fan 3 speed    1221  rpm
└── Noise level       0  dB

So no way to arrive at 500-600?

jonasmalacofilho, 2021-12-11:
> So no way to arrive at 500-600?

Definitively not that low, the fans would most likely fail to start with such low voltage; they could also randomly stall.  This is a limitation of variable DC/non-PWM fans.

At something like 700–800 rpm the fans would probably already work reliably, but I don't know any way of achieving that with the NZXT RGB & Fan Controller / Smart Device v2.[^1]  You may have better luck with the controller on your motherboard.

Closing the issue as everything seems to be working within spec.

[^1]: https://github.com/liquidctl/liquidctl/issues/58#issuecomment-546157049

closed 2021-12-11 by jonasmalacofilho