liquidctl/liquidctl #406: Corsair Vengeance PRO RGB not detected, "unknown" vendor

soundscape84, 2022-01-04:
My 3600Mhz Corsair Vengeance PRO RGB RAM isn't detected by liquidctl, and cpu-x shows the vendor of the RAM as "unknown". hwinfo and dmidecode show similar results. The RAM otherwise works fine, and the OS detects the correct amount (32GB).

soundscape84, 2022-01-04:
Output of `sudo demidecode --type 17`:

```
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0013, DMI type 17, 92 bytes
Memory Device
	Array Handle: 0x000B
	Error Information Handle: 0x0012
	Total Width: Unknown
	Data Width: Unknown
	Size: No Module Installed
	Form Factor: Unknown
	Set: None
	Locator: DIMM 0
	Bank Locator: P0 CHANNEL A
	Type: Unknown
	Type Detail: Unknown
	Speed: Unknown
	Manufacturer: Unknown
	Serial Number: Unknown
	Asset Tag: Not Specified
	Part Number: Unknown
	Rank: Unknown
	Configured Memory Speed: Unknown
	Minimum Voltage: Unknown
	Maximum Voltage: Unknown
	Configured Voltage: Unknown
	Memory Technology: Unknown
	Memory Operating Mode Capability: Unknown
	Firmware Version: Unknown
	Module Manufacturer ID: Unknown
	Module Product ID: Unknown
	Memory Subsystem Controller Manufacturer ID: Unknown
	Memory Subsystem Controller Product ID: Unknown
	Non-Volatile Size: None
	Volatile Size: None
	Cache Size: None
	Logical Size: None

Handle 0x0015, DMI type 17, 92 bytes
Memory Device
	Array Handle: 0x000B
	Error Information Handle: 0x0014
	Total Width: 64 bits
	Data Width: 64 bits
	Size: 16 GB
	Form Factor: DIMM
	Set: None
	Locator: DIMM 1
	Bank Locator: P0 CHANNEL A
	Type: DDR4
	Type Detail: Synchronous Unbuffered (Unregistered)
	Speed: 3600 MT/s
	Manufacturer: Unknown
	Serial Number: 00000000
	Asset Tag: Not Specified
	Part Number: CMW32GX4M2Z3600C18
	Rank: 1
	Configured Memory Speed: 3600 MT/s
	Minimum Voltage: 1.2 V
	Maximum Voltage: 1.2 V
	Configured Voltage: 1.2 V
	Memory Technology: DRAM
	Memory Operating Mode Capability: Volatile memory
	Firmware Version: Unknown
	Module Manufacturer ID: Bank 3, Hex 0x9E
	Module Product ID: Unknown
	Memory Subsystem Controller Manufacturer ID: Unknown
	Memory Subsystem Controller Product ID: Unknown
	Non-Volatile Size: None
	Volatile Size: 16 GB
	Cache Size: None
	Logical Size: None

Handle 0x0018, DMI type 17, 92 bytes
Memory Device
	Array Handle: 0x000B
	Error Information Handle: 0x0017
	Total Width: Unknown
	Data Width: Unknown
	Size: No Module Installed
	Form Factor: Unknown
	Set: None
	Locator: DIMM 0
	Bank Locator: P0 CHANNEL B
	Type: Unknown
	Type Detail: Unknown
	Speed: Unknown
	Manufacturer: Unknown
	Serial Number: Unknown
	Asset Tag: Not Specified
	Part Number: Unknown
	Rank: Unknown
	Configured Memory Speed: Unknown
	Minimum Voltage: Unknown
	Maximum Voltage: Unknown
	Configured Voltage: Unknown
	Memory Technology: Unknown
	Memory Operating Mode Capability: Unknown
	Firmware Version: Unknown
	Module Manufacturer ID: Unknown
	Module Product ID: Unknown
	Memory Subsystem Controller Manufacturer ID: Unknown
	Memory Subsystem Controller Product ID: Unknown
	Non-Volatile Size: None
	Volatile Size: None
	Cache Size: None
	Logical Size: None

Handle 0x001A, DMI type 17, 92 bytes
Memory Device
	Array Handle: 0x000B
	Error Information Handle: 0x0019
	Total Width: 64 bits
	Data Width: 64 bits
	Size: 16 GB
	Form Factor: DIMM
	Set: None
	Locator: DIMM 1
	Bank Locator: P0 CHANNEL B
	Type: DDR4
	Type Detail: Synchronous Unbuffered (Unregistered)
	Speed: 3600 MT/s
	Manufacturer: Unknown
	Serial Number: 00000000
	Asset Tag: Not Specified
	Part Number: CMW32GX4M2Z3600C18
	Rank: 1
	Configured Memory Speed: 3600 MT/s
	Minimum Voltage: 1.2 V
	Maximum Voltage: 1.2 V
	Configured Voltage: 1.2 V
	Memory Technology: DRAM
	Memory Operating Mode Capability: Volatile memory
	Firmware Version: Unknown
	Module Manufacturer ID: Bank 3, Hex 0x9E
	Module Product ID: Unknown
	Memory Subsystem Controller Manufacturer ID: Unknown
	Memory Subsystem Controller Product ID: Unknown
	Non-Volatile Size: None
	Volatile Size: 16 GB
	Cache Size: None
	Logical Size: None

```

jonasmalacofilho, 2022-01-04:
Only _non-pro_ Vengeange RGB DDR4 modules are specifically supported by liquidctl.

DDR4 modules that advertise a temperature sensor through the SPD are also supported with our generic driver (on select platforms), but yours don't appear to do that.

closed 2022-01-04 by jonasmalacofilho