liquidctl/liquidctl #407: extra/yoda failing with TypeError

Tloe, 2022-01-11:
I am trying to run liquidctl with extra/yoda and getting a TypeError. Seems to be docopt failing.. I tired to print the args before yoda calls_make_opts in cli.py and --pick is set to a empty []
```
    ~  sudo yoda control fan1 with '(30,10),(62,10),(65,30),(70,40),(71,70),(75,100)' on k10temp-pci-00c3                                                       
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'list'
    ~   
```
- Affected device: Corsair Hydro H100i Pro XT
- Operating system: Manjaro
- Liquidctl version: Manjaro liquidctl v1.8.0

jonasmalacofilho, 2022-01-12:
Can you run it with `--debug` so we can see the stack trace?

Tloe, 2022-01-12:
When I add --debug it just prints the help. It seems to be docopt that is failing so I guess its not able to read the --debug option either?

```
    ~  sudo yoda --debug control fan1 with '(30,10),(62,10),(65,30),(70,40),(71,70),(75,100)' on k10temp-pci-00c
Usage:
  yoda [options] show-sensors
  yoda [options] control (<channel> with <profile> on <sensor> [and])...
  yoda --help
  yoda --version
Options:
  --interval <seconds>     Update interval in seconds [default: 2]
  -m, --match <substring>  Filter devices by description substring
  -n, --pick <number>      Pick among many results for a given filter
  --vendor <id>            Filter devices by vendor id
  --product <id>           Filter devices by product id
  --release <number>       Filter devices by release number
  --serial <number>        Filter devices by serial number
  --bus <bus>              Filter devices by bus
  --address <address>      Filter devices by address in bus
  --usb-port <port>        Filter devices by USB port in bus
  -v, --verbose            Output additional information
  -g, --debug              Show debug information on stderr
  --hid <module>           Override API for USB HIDs: usb, hid or hidraw
  --legacy-690lc           Use Asetek 690LC in legacy mode (old Krakens)
  --version                Display the version number
  --help                   Show this message
Examples:
  yoda --match grid show-sensors
  yoda --match grid control fan1 with '(20,20),(35,100)' on nct6793.systin
  yoda --match kraken show-sensors
  yoda --match kraken control pump with '(20,50),(50,100)' on istats.cpu and fan with '(20,25),(34,100)' on _internal.liquid
Requirements:
  all platforms  liquidctl, including the Python APIs (pip install liquidctl)
  Linux/FreeBSD  psutil (pip install psutil)
  macOS          iStats (gem install iStats)
  Windows        none, system sensors not yet supported
Changelog:
  0.0.4  Fix casing of log and error messages
  0.0.3  Remove duplicate option definition
  0.0.2  Add low-pass filter and basic error handling.
  0.0.1  Generalization of krakencurve-poc 0.0.2 to multiple devices.
Copyright (C) 2020–2021  Jonas Malaco
SPDX-License-Identifier: GPL-3.0-or-later
```

jonasmalacofilho, 2022-01-12:
Possibly.

I can't reproduce this behavior with my distro-provided docopt 0.6.2.  What version of it do you have?

Tloe, 2022-01-12:
hum.. where can I see the version? Where are python packages stored? Its a manjaro install with stable. Manjaro could be a couple weeks after arch I think..

jonasmalacofilho, 2022-01-13:
Oh, right, you're on Manjaro.  Then you should have 0.6.2 as well.  But you can check it with `pacman -Qi python-docopt`.

Of course, this assumes you're _not_ using a Python virtual environment for `yoda`; if you are, then you need to check what's installed in that environment with `pip`.

jonasmalacofilho, 2022-01-13:
In the meantime, can you simply replace

https://github.com/liquidctl/liquidctl/blob/73962574632f94050c2a75f517e929a29797b5e2/extra/yoda#L221

with

```py
if true:
```

to manually enable all debug output and get the traceback for the original error?

Tloe, 2022-01-13:
`    ~  pacman -Qi python-docopt                                                      1 ✘  px-admin@brutus 
Name            : python-docopt
Version         : 0.6.2-11
Description     : Pythonic argument parser, that will make you smile
Architecture    : any
URL             : https://github.com/docopt/docopt
Licenses        : MIT
Groups          : None
Provides        : None
Depends On      : python
Optional Deps   : None
Required By     : ueberzug
Optional For    : None
Conflicts With  : None
Replaces        : None
Installed Size  : 79.00 KiB
Packager        : Felix Yan <felixonmars@archlinux.org>
Build Date      : Wed 01 Dec 2021 10:35:42 PM CET
Install Date    : Mon 03 Jan 2022 05:25:18 PM CET
Install Reason  : Installed as a dependency for another package
Install Script  : No
Validated By    : Signature
`

`
[DEBUG] __main__: yoda v0.0.4
[DEBUG] __main__: liquidctl v1.8.0
Traceback (most recent call last):
  File "/usr/local/bin/yoda", line 223, in <module>
    frwd = _borrow._make_opts(args)
  File "/usr/lib/python3.10/site-packages/liquidctl/cli.py", line 295, in _make_opts
    opts[opt] = _PARSE_ARG[arg](val)
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'list'
`

jonasmalacofilho, 2022-01-14:
Ok, I see what you before when mentioning `_make_opts`.  But I'm not sure docopt is the issue, since you appear to have the same version as I (and, in fact, it's the same upstream docopt version since 2014).

That said, have you checked that there aren't any virtual environments and other copies of docopt that could possibly be in use?

---

Another thing I noticed is that the help message when you tried to pass `--debug` is not what it's supposed to be: specifically, some empty lines are missing.  While that may seem minor, it could actually change how docopt parses usage/options from it.

It also suggests that you're not using an exact (current) copy of `extra/yoda`, which means I'm not looking at the right code.  Can you try a fresh copy from what's currently in this repository?

Tloe, 2022-01-14:
My bad.. Not sure why, but when I downloaded the raw file from the repo and copied it in it was fine.. I copy pasted it in before so something must have happened there. Thanks for your help :)