liquidctl/liquidctl #408: Request support for Corsair iCUE H150i ELITE LCD

alfiedotwtf, 2022-01-16:
_Note: Just speculation, but I wouldn't be surprised if the LEDs on this model worked the same as #218, with the only difference here being that it has a programmable LCD._

The Corsair iCUE H150i ELITE LCD is a AIO liquid cooler, and comes with _3_ fans with LCDs. Currently, `liquidctl` can read the LEDs but not write to them. Here's the `initialize`output:

```
  > liquidctl initialize
  Corsair Commander Core (experimental)
  ├── Firmware version            2.6.201
  ├── AIO LED count                   N/A
  ├── RGB port 1 LED count              8
  ├── RGB port 2 LED count              8
  ├── RGB port 3 LED count              8
  ├── RGB port 4 LED count            N/A
  ├── RGB port 5 LED count            N/A
  ├── RGB port 6 LED count            N/A
  ├── Water temperature sensor        Yes
  └── Temperature sensor 1             No
```
I can use OpenRGB to read and write to the LEDs, but am currently using Corsair iCUE within a Windows 10 VM so that I can also program the LCD screen.

The device appears to support the following features:

 - [ x] monitoring of temperatures or other environment sensors
 - [x ] monitoring of fan or pump speeds or duty cycles
 - [ ?] monitoring of voltages, currents or power
 - [x ] configurable fan or pump speeds
 - [ ?] configurable voltages or current/power limits
 - [x ] configurable lighting of embedded LEDs
 - [ x] configurable lighting of accessories like RGB fans or LED strips

The device is connected to the host system using:

- [ x] USB
- [ ] PCI-E
- [ ] onboard (the motherboard)
- [ ] other (please elaborate)
- [ ] unknown

The device communicates with the host system using:

- [ ] HID
- [ x] USB
- [ ] I²C or SMBus
- [ x] other (I also have the pump connected to the motherboard)
- [ ] unknown

Here's the output from `lsusb -v`:

```
Bus 001 Device 003: ID 1b1c:0c1c Corsair CORSAIR iCUE Commander CORE
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c1c 
  bcdDevice            1.00
  iManufacturer           1 Corsair
  iProduct                2 CORSAIR iCUE Commander CORE
  iSerial                 3 e00a20b00e81e3aab4391c061091005f
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0042
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      34
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0300  1x 768 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      21
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0300  1x 768 bytes
        bInterval               1
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      0
Device Status:     0x0001
  Self Powered

Bus 001 Device 002: ID 1b1c:0c33 Corsair LCD Cap for Elite Capellix coolers
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c33 
  bcdDevice            2.00
  iManufacturer           1 Corsair
  iProduct                2 LCD Cap for Elite Capellix coolers
  iSerial                 3 0434214380000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              288mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     381
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               1
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)
```
Finally, I can help with:¹

- [ x] testing changes on Linux
- [ x] testing changes on Windows
- [ ] testing changes on Mac OS
- [ x] attempting to capture USB traffic
- [ ] attempting to capture I²C/SMBus traffic
- [ ] analyzing traffic data
- [ ] documenting the protocol
- [ ] implementing the driver

¹ Assuming documentation and/or assistance will be provided.

jonasmalacofilho, 2022-01-16:
Thanks for the detailed info.

Can you capture some traffic to the `1b1c:0c33` ("Corsair LCD Cap for Elite Capellix coolers") device?  Start with something simple, like setting a single static image.  This will give us some idea of what we're dealing with...

---

@ParkerMc, pinging you as this also involves a Commander Core, and maybe there's something OP can help you there too.

alfiedotwtf, 2022-01-16:
I'd be happy to help capture, but sorry I'll need some guidance.

I'm assuming you want this captured from within the Windows 10 VM? If so, which packet capture software did you want me to use (and maybe any online tutorials I can use to set this up)? Thanks.

ParkerMc, 2022-01-16:
I do want to add that the LCD is actually an upgrade on the cooler that can be bought [separately](https://www.corsair.com/us/en/Categories/Products/Liquid-Cooling/iCUE-ELITE-CPU-Cooler-LCD-Display-Upgrade-Kit/p/CW-9060056-WW). So it is basically an add-on to coolers that already use the commander core. I do think it would be preferable to keep both issues though.

As far as capturing data, what Jonas said would be perfect.
> Start with something simple, like setting a single static image.
There are docs [here](https://github.com/liquidctl/liquidctl/blob/main/docs/developer/creating-vm-for-capture.md) that go over the process.

If you wouldn't mind screen sharing the options iCUE gives you to configure it some time, that would also help a lot in giving me an idea of how it works. There is the liquidctl discord [here](https://discord.gg/GyCBjQhqCd) and we have `#wg-commander-core` for focusing on the commander core.

ParkerMc, 2022-01-16:
Also if you don't mind running this script too. It will dump a lot of what's in the memory into a csv file that you can then upload. I don't think it will include everything for the LCD, but it should include what changes when it's connected.
https://github.com/liquidctl/collected-device-data/blob/master/Corsair Commander Core/dump_memory Script - ParkerMc/dump_memory.py

jonasmalacofilho, 2022-01-16:
> I'd be happy to help capture, but sorry I'll need some guidance.

We have a document on [Capturing USB traffic](https://github.com/liquidctl/liquidctl/blob/main/docs/developer/capturing-usb-traffic.md) in our development docs.  The "Capturing USB traffic on Linux" section should be a good fit for your setup. 

If you already have a VM, you can probably skip the accompanying "Virtual Machine Setup for Traffic Capture" document.

---

And in addition to what Parker said:

> I do want to add that the LCD is actually an upgrade on the cooler that can be bought separately. So it is basically an add-on to coolers that already use the commander core. I do think it would be preferable to keep both issues though.

I agree, we should keep the issues separate, and only focus on the specific LCD Cap device in this issue.

> Also if you don't mind running this script too. It will dump a lot of what's in the memory into a csv file that you can then upload. I don't think it will include everything for the LCD, but it should include what changes when it's connected.

I won't be surprised if there is no interaction between the Commander Code (which is what the script will dump) and the LCD Cap.  But it's good to double check, and Parker's script is incredibly useful.

KittyKatt, 2022-11-12:
Has anyone had any luck with this one? Was considering buying this over the z73 but am possibly thinking different based on being able to control it in Linux.

cloudcell, 2022-11-24:
I also need support for Corsair iCUE H150i Elite Capellix. In my case it's not being detected at all. I investigated the issue and found that the device ID, in my case, is **1b1c:0c32** instead of **...0c33**. Perhaps the manufacturer changed the code. If so, can this be taken into account? 

I hope you find this useful.

Here's the output from running liquidctl with debug/verbose options: https://pastebin.com/D7JX2Y7V

jonasmalacofilho, 2022-11-24:
@cloudcell, it's odd that your output _only_ lists the 1b1c:0c32 device, but no 1b1c:0c1c (Commander Core). Are you sure the (included) fan controller is connected correctly to the motherboard?

cloudcell, 2022-11-24:
@jonasmalacofilho, my motherboard (M12SWA-TF) has only one compatible USB connector and that was previously used by the front panel USB sockets. So I had to order a USB splitter (USB hub). Now the device is connected to the motherboard via that hub, i.e. not directly. I just bought this liquid cooler, so it might be a recent change by the manufacturer. It's actually a Commander Core ST as you can see here https://linux-hardware.org/?id=usb:1b1c-0c32 (this has been produced by the hardware probe). Besides, I have both Linux AND Windows 10 on my system and the device functions perfectly on Windows. In addition, if I modify the line in commander_core.py file as suggested here #511, then liquidctl could produce something similar to the output produced in #511. I did not install it properly (I just replaced the files installed by default with the files from this repo and "hacked" the code so it worked, and it now produces an error after I restarted the computer, while I'd rather wait until the suggestion in #511 gets merged then spend time on "hacking" the installed code again).

As I already said, the Windows iCUE software (which is over 1 gigabyte in size!) functions perfectly in both detecting and managing all the temperature sensors and RGB lights,  reporting RPMs. After installation, it updated the firmware of the Commander to version 2.17.x.x (if I remember correctly), then I set the lights to "temperature based" and "hardware only mode." Everything works perfectly by itself now. However, I wish I did not have to switch to Windows and install that ridiculously enormous bloatware with the size of a whole operating system :) to change the settings if I have to change them at any point.

I hope this helps.

jonasmalacofilho, 2022-11-25:
@cloudcell, oh, I see, and thanks for the additional information. Can you also tell me the SKU (something that looks like `CW-0000000-XX` written on the box)?

cloudcell, 2022-11-25:
You are welcome.
As for the SKU, here it is: "CW-9060048-WW"
Also, just in case, the name of the product on the box says the following: "iCUE H150i ELITE CAPELLIX" exactly as can be seen on this guy's video: https://www.youtube.com/watch?v=7eo3HQnT2vA (the title picture of the video).

jonasmalacofilho, 2022-11-25:
@cloudcell thanks again. And you're right, we can probably just add the new VID:PID to the driver, as there's no LCD screen on that model.

(Yeah, you said from the start it was an Elite Capellix, not Elite LCD... but since this is an issue about the latter, commenting here about the former was a bit confusing).

browserdotsys, 2023-05-21:
Hi – I have some code to support displaying images on this LCD display. The protocol is very simple and involves just sending 480x480 JPEG images over USB HID. Animated GIFs, live temperature/time/etc. display is all done by repeatedly sending individual frames to the display.

Here's the code: https://gist.github.com/browserdotsys/ef1b22c60c31d9c61e18cca30b3ce903

The LCD also supports uploading an image or animation to the cooler. I haven't tried to RE that yet but I have a packet capture of it, as well as a capture of a firmware update (and you can extract the LCD's firmware from that pretty easily). Unfortunately I can't upload these here (they're around 150MB zipped) but I'll try to follow up once I find a place to stash them.

wandervicente, 2023-12-12:
Hello!

I loved knowing that there are other means of control besides Icue, but could you clear up my doubts?

I also have the H150I LCD, and I have the red triangle of death, but when I close Icue it disappears. Does this code that was sent to control the LCD also work on Windows 10? Could you explain to me how I can run it or how I can add an image to the LCD in Windows 10 too?

I'm entering this world now, and I'm having difficulties, thank you for all the help!

Thanks!

pflavio, 2024-07-16:
Has this been implemented? 

Since LEDs on the Capellix cooler with the commander core doesn't work, I circumvented the fan control with a lightening node pro and a single data cable directly to my main board. 

So now I can control everything in my case except the LEDs in the Capellix cooler which is annoying as it blinks yellow all the time. My plan would therefore be to get the LCD screen and either set a static image (which hopefully survives reboots) or, if this is possible, boot into a Windows SSD that I set up specifically to update firmware on Corsair devices to use iCue to set the temperature sensor display (again, hoping this survives rebooting into Linux).

What do you guys think?

Consumedgrub2, 2024-08-15:
Same, bump