liquidctl/liquidctl #409: Support the project

WMcKibbin, 2022-01-21:
Please describe the issue or request as clearly and completely as possible.
I would like to give support to the liquidctl project with a https://www.buymeacoffee.com/ or patreon or github sponsorship. Do you guys have any such mechanism in place? I just bought a nice new liquid cooler which is going to be controlled through this software. Would like to say thankyou somehow. 

Thanks!

jonasmalacofilho, 2022-01-21:
Hey, thanks for wanting to support the project!

At the org level, setting up something (probably through Open Collective) is on my to-do list.

At the personal level I thought GitHub Sponsors would be working for me by now, but it took a while to even be enabled in my region... now my application is "pending review."  In the meantime, I guess I should create a Patreon for myself; once I do, I'll drop a link to it here.

WMcKibbin, 2022-01-22:
Awesome, drop me those links when you can!

jonasmalacofilho, 2022-03-23:
I added a [FUNDING.md](https://github.com/liquidctl/liquidctl/blob/main/FUNDING.md) document with various ways to support the liquidctl maintainers, and also enabled the "Sponsor" button here on GitHub (pointing to that same document).

commit 1ff340fd (2022-03-25): setup: link to funding document from PyPI

Closes: #409
closed 2022-03-25 by jonasmalacofilho