liquidctl/liquidctl #411: liquidctl fails on python 3.10

computersavvy, 2022-01-25:
I recently discovered liquidctl and tried it out.  I have 2 systems, One running Fedora 34 which has python 3.9 and on that system liquidctl installed and works well.
My second system is running Fedora 35 and has python 3.10 installed.  On that system liquidctl installed the same as on the other system and this is the result.

```
$ liquidctl list
Device #0: Corsair Hydro H115i Pro
```
```
$ liquidctl initialize --debug
[DEBUG] (cli) (main): version: liquidctl v1.8.1
[DEBUG] (cli) (main): platform: Linux-5.15.16-200.fc35.x86_64-x86_64-with-glibc2.34
[DEBUG] (smbus) (find_devices): searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] (smbus) (find_devices): found I²C bus i2c-3
[DEBUG] (smbus) (find_devices): found I²C bus i2c-1
[DEBUG] (smbus) (find_devices): found I²C bus i2c-6
[DEBUG] (smbus) (find_devices): found I²C bus i2c-4
[DEBUG] (smbus) (find_devices): found I²C bus i2c-2
[DEBUG] (smbus) (find_devices): found I²C bus i2c-0
[DEBUG] (smbus) (find_devices): found I²C bus i2c-7
[DEBUG] (smbus) (find_devices): found I²C bus i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus (CommanderCore, CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] (usb) (find_devices): found HID device 046d:101b
[DEBUG] (usb) (find_devices): found HID device 046d:101b
[DEBUG] (usb) (find_devices): found HID device 046d:101b
[DEBUG] (usb) (find_devices): found HID device 046d:101b
[DEBUG] (usb) (find_devices): found HID device 046d:101b
[DEBUG] (usb) (find_devices): found HID device 046d:200a
[DEBUG] (usb) (find_devices): found HID device 046d:200a
[DEBUG] (usb) (find_devices): found HID device 046d:200a
[DEBUG] (usb) (find_devices): found HID device 046d:200a
[DEBUG] (usb) (find_devices): found HID device 046d:200a
[DEBUG] (usb) (find_devices): found HID device 046d:200a
[DEBUG] (usb) (find_devices): found HID device 046d:c52b
[DEBUG] (usb) (find_devices): found HID device 046d:c52b
[DEBUG] (usb) (find_devices): found HID device 046d:c52b
[DEBUG] (usb) (find_devices): searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc, CorsairAsetekProDriver)
[DEBUG] (usb) (find_devices): found USB device 1d6b:0002
[DEBUG] (usb) (find_devices): found USB device 1b1c:0c13
[DEBUG] (usb) (probe): instantiated CorsairAsetekProDriver driver for Corsair Hydro H115i Pro
[DEBUG] (usb) (find_devices): found USB device 1d6b:0001
[DEBUG] (usb) (find_devices): found USB device 1d6b:0003
[DEBUG] (usb) (find_devices): found USB device 2109:3431
[DEBUG] (usb) (find_devices): found USB device 1d6b:0002
[DEBUG] (usb) (find_devices): found USB device 1d6b:0001
[DEBUG] (usb) (find_devices): found USB device 1d6b:0002
[DEBUG] (usb) (find_devices): found USB device 046d:c52b
[DEBUG] (usb) (find_devices): found USB device 1d6b:0001
[DEBUG] (usb) (find_devices): found USB device 1d6b:0002
[DEBUG] (usb) (find_devices): found USB device 1d6b:0001
[DEBUG] (usb) (find_devices): found USB device 1d6b:0003
[DEBUG] (usb) (find_devices): found USB device 1d6b:0002
[DEBUG] (cli) (main): device: Corsair Hydro H115i Pro
[INFO] (cli) (log_error): [Errno 13] Access denied (insufficient permissions)
Traceback (most recent call last):
  File "/home/jvian/.local/lib/python3.10/site-packages/liquidctl/cli.py", line 422, in main
    with dev.connect(**opts):
  File "/home/jvian/.local/lib/python3.10/site-packages/liquidctl/driver/asetek.py", line 163, in connect
    ret = super().connect(**kwargs)
  File "/home/jvian/.local/lib/python3.10/site-packages/liquidctl/driver/usb.py", line 118, in connect
    self.device.open()
  File "/home/jvian/.local/lib/python3.10/site-packages/liquidctl/driver/usb.py", line 249, in open
    cfg = self.usbdev.get_active_configuration()
  File "/usr/lib/python3.10/site-packages/usb/core.py", line 875, in get_active_configuration
    return self._ctx.get_active_configuration(self)
  File "/usr/lib/python3.10/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python3.10/site-packages/usb/core.py", line 236, in get_active_configuration
    self.managed_open()
  File "/usr/lib/python3.10/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python3.10/site-packages/usb/core.py", line 120, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "/usr/lib/python3.10/site-packages/usb/backend/libusb1.py", line 786, in open_device
    return _DeviceHandle(dev)
  File "/usr/lib/python3.10/site-packages/usb/backend/libusb1.py", line 643, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "/usr/lib/python3.10/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)
[ERROR] (cli) (log_error): Error: insufficient permissions to access Corsair Hydro H115i Pro
```

I realize that says insufficient permissions, but I cannot identify the device that I need to set permissions for.
On the fedora 34 system the device it is defined as is /dev/hidraw2 and I was able to set permissions for that.
On the fedora 35 system I cannot see in either dmesg or journalctl what device name it is assigned, or even if it gets a device name.
This from fedora 35 
```
Jan 22 12:47:25 raptor systemd[1]: Starting Apply Kernel Variables...
Jan 22 12:47:25 raptor kernel: usb 7-4: New USB device found, idVendor=1b1c, idProduct=0c13, bcdDevice= 1.00
Jan 22 12:47:25 raptor kernel: usb 7-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 22 12:47:25 raptor kernel: usb 7-4: Product: H115i Platinum
Jan 22 12:47:25 raptor kernel: usb 7-4: Manufacturer: Corsair
Jan 22 12:47:25 raptor kernel: usb 7-4: SerialNumber: 7289_2.0
```
and this from fedora 34
```
[    1.828402] usb 1-7.2: new full-speed USB device number 5 using xhci_hcd
[    1.867451] hid-generic 0003:26CE:01A2.0002: input,hidraw1: USB HID v1.10 Device [ASRock LED Controller] on usb-0000:01:00.0-8/input0
[    2.130979] usb 1-7.2: New USB device found, idVendor=1b1c, idProduct=0c21, bcdDevice= 1.00
[    2.130983] usb 1-7.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.130985] usb 1-7.2: Product: H115iRGBPROXT
[    2.130986] usb 1-7.2: Manufacturer: Corsair Inc.
[    2.154143] hid-generic 0003:1B1C:0C21.0003: hiddev97,hidraw2: USB HID v1.11 Device [Corsair Inc. H115iRGBPROXT] on usb-0000:01:00.0-7.2/input0
```
As you can see the H115i is assigned to hidraw2 with fedora 34 but it does not get assigned in fedora 35, yet the device is seen and recognized.
```
$ uname -a
Linux raptor 5.15.16-200.fc35.x86_64 #1 SMP Thu Jan 20 15:38:18 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

$ liquidctl --version
liquidctl v1.8.1
```
I also have different versions of the corsair H115i.  The fedora 35 system has the Corsair H115i Platinum and the fedora 34 system has the Corsair H115i RPB Pro XT.  Both are connected to the mobo USB port with the cable provided by corsair.

I can add additional info if needed, but my initial guess is that the device is not properly configured during boot which would account for not able to access it.  Do you have any suggestions for a solution?

jonasmalacofilho, 2022-01-25:
The Fedora 35 system has a H115i Pro, not Platinum or Pro XT.

This is significant because the Pro is not a HID, and, thus, you won't find a `/dev/hidraw*` node for it.

How are you setting permissions on the Fedore 34 system?  If you're using a udev rule, a small tweak might be all that's needed, and you can take a look at the one we provide as an example.

https://github.com/liquidctl/liquidctl/blob/42d5e19e810cd9d4979bb83cf5be0a44dc4a961c/extra/linux/71-liquidctl.rules#L116-L117

P.S. If I'm reading the spec correctly, the Fedora package already [installs our udev rules](https://src.fedoraproject.org/rpms/liquidctl/blob/rawhide/f/liquidctl.spec#_121).  Maybe you just need to reboot or manually realod & trigger them?

jonasmalacofilho, 2022-01-25:
To clarify: the USB product string descriptor (which you see in the kernel logs) from those devices is wrong and a recurring source of confusion.

```
Jan 22 12:47:25 raptor kernel: usb 7-4: Product: H115i Platinum
```

It's possible that Corsair considered releasing that family with the "Platinum" brand, and later changed their mind.  Regardless, there actually exists a later and completely different Corsair Hydro Platinum family of liquid coolers.

computersavvy, 2022-01-25:
I did not note the udev rules link so I had not added that.
Once the rules were added a reboot showed exactly the same info as previously posted, but I was able to initialize the cooler and then look at the status.  I still have not been able to identify the actual device in /dev but I guess that since the udev rule allows my regular user to manage it that is no longer a critical issue.

Thank you for the feedback and the info that allowed me to get things set up.

Yes, the info from dmesg said platinum and it did not register that what it said in the 'liquidctl list' output was enough different and more accurate so I would not have caught what you told me.

Thanks again.

BTW, this cooler is one that corsair sent me as a warranty replacement for a newer / better one when the pump failed.

jonasmalacofilho, 2022-01-25:
I don't think there will be a `/dev/` node for that device, at least not by default.

We're using libusb-1.0 under the hood, and we/it automatically bind `usbfs` to the device when we need to interact with it; we unbind it once we're done.  Maybe looking into libusb-1.0 and `usbfs`  helps you find the corresponding `/dev/*` device.

closed 2022-01-25 by jonasmalacofilho

computersavvy, 2022-01-25:
One last thing that I presently am struggling with.
Although the 71-liquidctl.rules file contains an entry for the H115i RGB Pro XT, it seems it does not assign access to the hidraw device for a normal user.  Thus on my fedora 34 system at present I am manually using chmod to allow my regular user access to that device.  I use this alias to do that.
```
alias corsair="sudo chmod 666 /dev/hidraw2 && liquidctl initialize"
```

Do you have any hints as to how I could modify the udev rule or any other method to automatically allow access?

jonasmalacofilho, 2022-01-25:
That's expected, as the udev rules use the `TAG+=uaccess` mechanism.

```
$ ls -l /dev/hidraw*
crw-------  1 root root 237, 0 Jan 25 10:43 /dev/hidraw0
crw-------  1 root root 237, 1 Jan 25 10:43 /dev/hidraw1
crw-rw----+ 1 root root 237, 2 Jan 25 10:43 /dev/hidraw2
crw-------  1 root root 237, 3 Jan 25 10:43 /dev/hidraw3
crw-------  1 root root 237, 4 Jan 25 10:43 /dev/hidraw4
crw-rw----+ 1 root root 237, 5 Jan 25 10:43 /dev/hidraw5
crw-------  1 root root 237, 6 Jan 25 10:43 /dev/hidraw6
crw-------  1 root root 237, 7 Jan 25 10:43 /dev/hidraw7
crw-------  1 root root 237, 8 Jan 25 10:43 /dev/hidraw8
crw-------  1 root root 237, 9 Jan 25 10:43 /dev/hidraw9
```

With this tag (which corresponds to the `+` on `hidraw2` and `hidraw5` on my system), all physical user sessions ("seats") will be allowed to access the device.  Unfortunately, this is a not a very well documented feature.[^1]

Anyway, as long as your sessions are physical, `liquidctl` should just work.  But for headless sessions you may need to modify the rules, and the comment at the top of the file suggests one approach.

P.S. As long as you're using `sudo`, you can just `sudo liquidctl`.  No need to use `chmod` first.

[^1]: https://github.com/systemd/systemd/issues/4288

computersavvy, 2022-01-25:
Unfortunately the udev rule appears not to work for me.
```
$ liquidctl list -v
Device #0: Corsair Hydro H115i Pro XT
├── Vendor ID: 0x1b1c
├── Product ID: 0x0c21
├── Release number: 0x0100
├── Bus: hid
├── Address: /dev/hidraw2
└── Driver: HydroPlatinum

```
from the udev rules
```
# Corsair Hydro H115i Pro XT
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c21", TAG+="uaccess"
```
This after I have rebooted several times with the udev rule in place.
```
 $ ls -l /dev/hidraw*
crw-------. 1 root root 239, 0 Jan 25 15:21 /dev/hidraw0
crw-------. 1 root root 239, 1 Jan 25 15:21 /dev/hidraw1
crw-------. 1 root root 239, 2 Jan 25 15:21 /dev/hidraw2
crw-------. 1 root root 239, 3 Jan 25 15:21 /dev/hidraw3
crw-------. 1 root root 239, 4 Jan 25 15:21 /dev/hidraw4
crw-------. 1 root root 239, 5 Jan 25 15:21 /dev/hidraw5
crw-------. 1 root root 239, 6 Jan 25 15:21 /dev/hidraw6
```
So unfortunately udev seems not to be applying that rule to hidraw2 which is the corsair device.

Guess I need to dig deeper and figure out why udev is not applying that rule.

sudo does not work because I installed liquidctl using pip as user so it cannot be found when using sudo.
Guess I could uninstall it and reinstall as root, but did not want to do that.