liquidctl/liquidctl #412: Support for NZXT Kraken 120

pde, 2022-01-27:
This device is an [AIO liquid cooler](https://nzxt.com/product/kraken-120) that runs with NZXT Cam on Windows.

I don't see any NZXT devices in the output of `lspci` or `lsusb`. Running `liquidctl ` shows a `Gigabyte RGB Fusion 2.0 5702 Controller`, which I think is probably listed as `048d:702 Integrated Technolog Express, Inc. ITE Device` in lsusb output; I don't know if that is connected to the Kraken, but `liquidctl` errors when trying to tell it to do anything.

Happy to do a little linux-based hacking / debugging /research on this. Unfortunately I removed windows from this system so can't do much investigation of how NZXT Cam configures it at present :shrug: .

jonasmalacofilho, 2022-01-27:
The cooler may not be showing in `lsusb` as a NZXT device because it's probably too new to be in the USB id database.

Can you show me the complete `lsusb -v` output?

---

The RGB Fusion controller is part of your motherboard ([documentation](https://github.com/liquidctl/liquidctl/blob/main/docs/gigabyte-rgb-fusion2-guide.md)).

jonasmalacofilho, 2022-01-27:
Also, please fill in the first checklist that I added back to the your initial comment.  The rest I have already filled out, or will do from the `lsusb -v` output.

pde, 2022-02-01:
```
sudo lsusb -v

Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            5.10
  iManufacturer           3 Linux 5.10.0-11-amd64 xhci-hcd
  iProduct                2 xHCI Host Controller
  iSerial                 1 0000:0b:00.3
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
        bMaxBurst               0
Hub Descriptor:
  bLength              12
  bDescriptorType      42
  nNbrPorts             4
  wHubCharacteristic 0x000a
    No power switching (usb 1.0)
    Per-port overcurrent protection
  bPwrOn2PwrGood       50 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  bHubDecLat          0.0 micro seconds
  wHubDelay             0 nano seconds
  DeviceRemovable    0x00
 Hub Port Status:
   Port 1: 0000.02a0 lowspeed L1
   Port 2: 0000.02a0 lowspeed L1
   Port 3: 0000.02a0 lowspeed L1
   Port 4: 0000.02a0 lowspeed L1
Binary Object Store Descriptor:
  bLength                 5
[... 1301 lines cut ...]
  bLength              11
  bDescriptorType      41
  nNbrPorts            10
  wHubCharacteristic 0x000a
    No power switching (usb 1.0)
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00 0x00
  PortPwrCtrlMask    0xff 0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
   Port 2: 0000.0100 power
   Port 3: 0000.0103 power enable connect
   Port 4: 0000.0100 power
   Port 5: 0000.0507 highspeed power suspend enable connect
   Port 6: 0000.0103 power enable connect
   Port 7: 0000.0100 power
   Port 8: 0000.0100 power
   Port 9: 0000.0100 power
   Port 10: 0000.0100 power
Device Status:     0x0001
  Self Powered
```

pde, 2022-02-01:
Sorry I haven't filled out more of the form, that's mostly because I don't know the answers. I may get around to re-installing windows on this system, in which case I will have more confidence in the exact controllable feature set of the device and be able to help with Windows diagnostics.

I could also try to physically inspect the device, but this is a prebuilt system and I see cables running from the Kraken 120 behind some shielding so, it would take a bit of fiddling to figure out what it's connected to.

pde, 2022-02-01:
(I do know the device has programmable colours and colour cycles)

pde, 2022-02-01:
I took some guesses at answers to other questions from a quick skim of [this video](https://www.youtube.com/watch?v=gOlqdVEladA).

jonasmalacofilho, 2022-02-02:
Thanks for the additional information, and don't worry about not having initially filled in the form.

---

It turns out that there is no (high-level) controller in this new cooler model.

According to the manual:[^1]

- the pump and fan are to be connected to an external controller (the manual suggests the motherboard);
- the LEDs can be connected to a compatible NZXT controller (I'm guessing: Hue 2, Smart Device V2 or RGB & Fan Controller) or, with an included adapter, to a 5V ARGB controller (e.g. the RGB Fusion 2.0 header on your motherboard).

This means that this cooler isn't a device liquidctl has to (or can) support directly.

In your particular case:

 - pump and fan control will require you to find a tool or driver for your motherboard's controller (probably its super I/O chip); for most people it would already be supported by hwmon/lm-sensors, but I suspect your board may have a chip from ITE, which would make everything a lot harder;[^2]
 - on the other hand, for the LEDs liquidctl [already supports](https://github.com/liquidctl/liquidctl/blob/main/docs/gigabyte-rgb-fusion2-guide.md) the RGB Fusion 2.0 controller on your motherboard; as long as the cooler has been properly connected to it, you should be able to change its color and set some different lighting modes.

[^1]: [Kraken 120 multi-language manual](https://www.datocms-assets.com/34299/1636302137-coolingkraken-120manual-enpdf2110.pdf)
[^2]: Adding support for super I/O controllers on motherboards is currently outside the scope of this project.

closed 2022-02-02 by jonasmalacofilho

pde, 2022-02-09:
Thanks for your help on this @jonasmalacofilho. I was able to implement some nice fading functionality on top of the more limited featureset of the RGB Fusion 2.0, will send you all a (maybe?) useful PR.

jonasmalacofilho, 2022-02-09:
Just saw your PR.  Thanks!