liquidctl/liquidctl #413: Question about Commander Pro on Linux

wildewouter, 2022-01-28:
Hi all,

Thank you for this great tool and all the work you're putting in. I'm struggling a bit in setting fan curves as I don't seem to be getting temperature input. I have four temp probes that list as `No`. I'm using the Corsair Commander Pro. Is this something that can be fixed by installing certain drivers or just some hardware issue that I have to be ok with?

MarshallAsch, 2022-01-28:
Did you run the initialize command first? It needs to be run before the tempature probes will show up in the output

'liquidctl initialize all'

wildewouter, 2022-01-28:
Thank you for your answer! Yes I did run that.
[image]

jonasmalacofilho, 2022-01-30:
Have you checked that all four probes are firmly and properly connected to the Commander Pro?

Otherwise, can you provide the complete output of executing `sudo liquidctl initialize all --debug`?

---

P.S. I'm not familiar with that particular device.  But I think @MarshallAsch is especially busy at the moment, so let's see if maybe I can still help...

MarshallAsch, 2022-01-31:
Yup I'm the one that initially added support for the device (I have one currently hooked up and working successfully with all the probes working.

This is the version of everything I have got running. m running on a slightly outdated OS Fedora 33 so my kernel version is a little old. 

[image]

Have you checked in windows or on a windows VM with the device passed through that the temperature probes do show up there? 

---

>  But I think @MarshallAsch is especially busy at the moment

Yup, I am nearly done my Masters thesis and Im doing the last little bit of edits before it gets sent off to the examination committee some time near the end of next week

Ill try to keep an eye on this issue.

wildewouter, 2022-01-31:
Ok, I'm a complete idiot. I really thought my mobo would be passing temps to the commander pro. I got the commander pro preinstalled with the case and thought this would work out of the box. Seems I don't have any probes installed and need to buy them separately. Sorry for wasting everybody's time but thank you so much for helping people out the way you guys did. Great tool!

jonasmalacofilho, 2022-01-31:
Don't worry about it! : )

closed 2022-01-31 by jonasmalacofilho