liquidctl/liquidctl #417: ICUE H115i RGB PRO XT

galtline, 2022-02-06:
I get this when running  status

ERROR: Error: could not open Corsair H115i Pro XT (experimental), possibly due to insufficient permissions

jonasmalacofilho, 2022-02-06:
Check if the user running the application has permission to access the device.

Otherwise, the [Bug report template](https://github.com/liquidctl/liquidctl/issues/new/choose) can help you give me some basic information that, in turn, allows me to try to help you.

galtline, 2022-02-06:
Cant I run sudo before the commands to get the correct permissions ? 

Also I am running Linux Mint 20.3 Cinnamon

And version 1.5.1 of liquidctl

jonasmalacofilho, 2022-02-06:
> Cant I run sudo before the commands to get the correct permissions ?

Yes, that should do it.[^1]  You can also use the udev rules [we provide](https://github.com/liquidctl/liquidctl/blob/main/extra/linux/71-liquidctl.rules) to automatically grant access to all supported devices from any interactive session.

P.S. There's no need to email me "I am trying to get liquidctl to work. Can you help me out.", I was already trying to do that.

[^1]: The root user may use a different `PATH`, which could make it (and `sudo`) fail to find `liquidctl`, depending on how it's installed.

galtline, 2022-02-06:
I got it.. Thank you

closed 2022-02-06 by jonasmalacofilho