liquidctl/liquidctl #425: MSI Gaming Z Trio 6900 XT

Nolij, 2022-02-11:
Briefly describe the device (e.g. AIO liquid cooler): GPU

Please also mention the first-party software that is used with it (e.g. Corsair iCue), as well as any other commonly used tools: MSI MYSTIC LIGHT

The device appears to support the following features:

 - [x] monitoring of temperatures or other environment sensors
 - [x] monitoring of fan or pump speeds or duty cycles
 - [x] monitoring of voltages, currents or power
 - [x] configurable fan or pump speeds
 - [x] configurable voltages or current/power limits
 - [x] configurable lighting of embedded LEDs
 - [ ] configurable lighting of accessories like RGB fans or LED strips

The device is connected to the host system using:

- [ ] USB
- [x] PCI-E
- [ ] onboard (the motherboard)
- [ ] other (please elaborate)
- [ ] unknown

The device communicates with the host system using:

- [ ] HID
- [ ] USB
- [ ] I²C or SMBus
- [ ] other (please elaborate)
- [x] unknown

Please also include any useful additional information, such as USB vendor and product IDs, the output of `lsusb -v` (Linux) or `system_profiler SPUSBDataType` (Mac OS), links to external resources or already collected traffic data.:
`0d:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 21 [Radeon RX 6900 XT] [1002:73af] (rev c0)
0d:00.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 21 HDMI Audio [Radeon RX 6800/6800 XT / 6900 XT] [1002:ab28]`

---

Finally, I can help with:¹

- [x] testing changes on Linux
- [ ] testing changes on Windows
- [ ] testing changes on Mac OS
- [ ] attempting to capture USB traffic
- [x] attempting to capture I²C/SMBus traffic
- [ ] analyzing traffic data
- [ ] documenting the protocol
- [ ] implementing the driver

¹ Assuming documentation and/or assistance will be provided.

jonasmalacofilho, 2022-02-11:
You should know that at the moment it's unlikely that the maintainer team will work on this:

- we know very little about these devices;
- also, for non-USB protocols (which are generally more sensitive to mistakes) we want to have someone with access to the device writing the driver (so avoid having regular users testing potentially brick-risky code).

---

That said...

What does the Linux kernel driver for this card already expose?  If there's a corresponding hwmon device, we could use it for temperature and fan monitoring (and control, if that's supported).

Also, do you have any I²C devices recognized on that card?  (You might need to modprobe `i2c-dev`).  Those could be useful for lighting control, assuming these cards aren't too different in that regard to those from Nvidia.

P.S. [This](https://github.com/liquidctl/liquidctl/issues/299#issuecomment-788663127) comment about how to capture and analyze the protocol might not be completely useless.

Nolij, 2022-02-11:
> 

How do I check for I2C devices? I've run modprobe i2c-dev.

jonasmalacofilho, 2022-02-12:
I don't know of any particularly good (e.g. with udev) way to do it, but...

What does the following command output?

```
$ grep -rn . /sys/bus/i2c/devices/*/name
```

Nolij, 2022-02-12:
```
/sys/bus/i2c/devices/i2c-0/name:1:SMBus PIIX4 adapter port 0 at 0b00
/sys/bus/i2c/devices/i2c-10/name:1:AMDGPU DM aux hw bus 2
/sys/bus/i2c/devices/i2c-1/name:1:SMBus PIIX4 adapter port 2 at 0b00
/sys/bus/i2c/devices/i2c-2/name:1:SMBus PIIX4 adapter port 1 at 0b20
/sys/bus/i2c/devices/i2c-3/name:1:AMDGPU SMU
/sys/bus/i2c/devices/i2c-4/name:1:AMDGPU DM i2c hw bus 0
/sys/bus/i2c/devices/i2c-5/name:1:AMDGPU DM i2c hw bus 1
/sys/bus/i2c/devices/i2c-6/name:1:AMDGPU DM i2c hw bus 2
/sys/bus/i2c/devices/i2c-7/name:1:AMDGPU DM i2c hw bus 3
/sys/bus/i2c/devices/i2c-8/name:1:AMDGPU DM aux hw bus 0
/sys/bus/i2c/devices/i2c-9/name:1:AMDGPU DM aux hw bus 1
```

jonasmalacofilho, 2022-02-12:
For hardware monitoring [and control], the `amdgpu` driver exposes a hwmon device and some other sysfs attributes.[^1][^2]  That's good!

However, it seems that there are already tools for user-friendly monitoring and control, using those same interfaces.  And I don't see why would liquidctl do better... are those tools particularly limited in some way?

Back to LED control, it probably uses one of those exposed `AMDGPU` I²C devices.  But are there any tools that already implement that feature?  To work on this we would need to figure out the protocol, and, as I mentioned before,[^3] that could be hard to do.

[^1]: https://wiki.archlinux.org/title/AMDGPU#Monitoring
[^2]: https://www.kernel.org/doc/html/v4.20/gpu/amdgpu.html#gpu-power-thermal-controls-and-monitoring
[^3]: https://github.com/liquidctl/liquidctl/issues/299#issuecomment-788663127

Nolij, 2022-02-12:
Yeah I was more asking for RGB, its stock settings are more than enough in terms of performance and cooling for me. And monitoring already works. Just want RGB control.

jonasmalacofilho, 2022-02-12:
Related: https://gitlab.com/CalcProgrammer1/OpenRGB/-/issues/2216

jonasmalacofilho, 2022-02-14:
Oh, that's you too, right?  Is there any reason why you would still prefer liquidctl over OpenRGB?

I ask because given their more substantial experience with graphics cards, their specific focus on RGB control, and the fact that I don't have access to an AMD GPU right now, OpenRGB is much more likely to support this card in the next year.

In fact, unless someone really wants liquidctl to support this card, enough to figure out how to capture the traffic (and, likely, build a tool just for doing that), I don't see any progress happening on this issue here.

Nolij, 2022-02-14:
I'm aware of both, so I made a feature request for both, hoping to improve my odds. I understand that this isn't really the scope of this project though, so I'll focus on OpenRGB for now. Thank you for trying.

jonasmalacofilho, 2022-02-14:
You're welcome.  And I've subscribed to the OpenRGB issue and will keep an eye on it (and on any work on sniffing AMD I²C traffic).

But I'm close this issue since we're not really expecting to work on it, at least in the near future.

closed 2022-02-14 by jonasmalacofilho