liquidctl/liquidctl #427: Corsair H115i Elite pump speeds

samweisgamdschie, 2022-02-15:
After installing the main branch of this repo I got liquidctl working with my Corsair H115i Elite Capellix cooler. I read the docs from [here](https://github.com/liquidctl/liquidctl/blob/main/docs/corsair-commander-core-guide.md) and played around with it. In contrast to this link I found a few things out:
* pump speeds can be set to 60, 70-100 (with +5 steps).
* pump speed at 65 throws following exception:
`ERROR: Unexpected error with Corsair Commander Core (experimental): liquidctl.error.ExpectationNotMet: device returned incorrect data type`.
  
  But maybe it is caused from a too fast update of the pump speed (every 3 seconds) because I cant reproduce it without my bash for loop (`# for eh in {100..10..5}; do echo "Setting pump speed to $eh"; liquidctl set pump speed $eh; sleep 3; done`).
  
  Or it has something to do with the parallel running `# watch 'liquidctl status'`
* after the first set speed commands the RGB LEDs from both the pump and the fans had a regular hickup (short turnoff) with a period of about 3 seconds (after one rainbow color circle).
* So it seems setting color modes are in anyway possible.

Maybe you can add the pump speed info to the docs. Also I first installed it from pypi and found out that 1.8.1 does not support setting speeds. Only the version of the main branch does support it.
---

Additional information about the system:

- Affected device(s): Corsair H115i Elite Capellix
- Operating system and version: Debian GNU/Linux 11 (bullseye)
- Liquidctl installation method: git clone inside a pipenv environment
- Output of `liquidctl --version`: liquidctl v1.8.1 (fedf566415b4)

jonasmalacofilho, 2022-02-15:
Hi @samweisgamdschie,

> * pump speed at 65 throws following exception:
>       `ERROR: Unexpected error with Corsair Commander Core (experimental): liquidctl.error.ExpectationNotMet: device returned incorrect data [...]

Thanks for reporting this issue.

>     * after the first set speed commands the RGB LEDs from both the pump and the fans had a regular hickup (short turnoff) with a period of about 3 seconds (after one rainbow color circle).

This one as well.  Also, it might be related to the previous one.

@ParkerMc, I'm pinging you since you probably want to know about these.

---

The other stuff aren't really issues:

> [...] In contrast to this link I found a few things out:
> 
>     * pump speeds can be set to 60, 70-100 (with +5 steps).

Reading the docs, I don't think they say otherwise.  Still, can you suggest a change that would improve them?

Note that is important to avoid being too specific, particularly while the driver is experimental.

>     * So it seems setting color modes are in anyway possible.

Well, in the sense that the device/firmware supports it, yes.  But in the sense of what the driver currently supports, then [no](https://github.com/liquidctl/liquidctl/blob/fedf566415b4d4140b986e8dce8b1a1ca09a6ba2/liquidctl/driver/commander_core.py#L109-L110), that's just an artifact.

> Maybe you can add the pump speed info to the docs. [...]

I don't think it's time to document a bug in an unreleased version; we should instead try to fix it.

> [...] Also I first installed it from pypi and found out that 1.8.1 does not support setting speeds. Only the version of the main branch does support it.

We don't currently add "Changed/added in version x" notes to the documentation.

But, if you followed the "Documentation" link from PyPI, you would arrive at the documentation that corresponds to the particular version you were looking at.  And for 1.8.1, the docs correctly [don't](https://github.com/liquidctl/liquidctl/blob/v1.8.1/docs/corsair-commander-core-guide.md) mention any support for fan/pump speed control.

Still, thanks anyway.

ParkerMc, 2022-02-15:
> * pump speed at 65 throws following exception:
>       `ERROR: Unexpected error with Corsair Commander Core (experimental): liquidctl.error.ExpectationNotMet: device returned incorrect data [...]

When we do anything with the commander core we have to put it in software mode to change settings or get status and running something else in parallel will set it to software mode again and reset everything.  
Can you reproduce this bug without running the watch status in parallel? If so can you run the set speed script with `-debug` added to the end?

> * after the first set speed commands the RGB LEDs from both the pump and the fans had a regular hickup (short turnoff) with a period of about 3 seconds (after one rainbow color circle).

This is related to the other issue, whenever we set it to software mode the lights automatically get set to software mode and they turn off because software mode resets itself on power off. The only way around it is to use software mode that we can't do as it will reset itself to hardware mode if it doesn't receive data after about 30 seconds.

jonasmalacofilho, 2022-07-03:
Triage: @samweisgamdschie, have you been able to do the tests suggested by ParkerMc?

jonasmalacofilho, 2022-08-11:
Closing in favor of #455 (and due to lack of feedback here).

closed 2022-08-11 by jonasmalacofilho