liquidctl/liquidctl #432: OSError: [Errno 71] Protocol error

panmarco83, 2022-03-03:
When running liquidctl as user (not root), the output is ```OSError: [Errno 71] Protocol error```
The problem persists even after adding and updating the udev rules for my specific AIO ( Corsair Hydro H115i ).
This was an attempt at updating an older installation that was working in userspace (I believe it was 1.7.2)

(as user, not root) ```$ liquidctl list```

```
$ liquidctl list --debug
[DEBUG] (cli) (main): liquidctl: 1.8.1.post126+gcc0050f
[DEBUG] (cli) (main): platform: Linux-5.16.8-051608-generic-x86_64-with-glibc2.29
[DEBUG] (cli) (_log_requirements): python: 3.8.10 (default, Nov 26 2021, 20:14:08) 
[GCC 9.3.0]
[DEBUG] (cli) (_log_requirements): colorlog: 6.6.0
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.11.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): smbus: 1.1.post2
[DEBUG] (smbus) (find_devices): searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] (smbus) (find_devices): found I²C bus i2c-3
[DEBUG] (smbus) (find_devices): found I²C bus i2c-1
Traceback (most recent call last):
  File "/usr/local/bin/liquidctl", line 8, in <module>
    sys.exit(main())
  File "/home/source/.local/lib/python3.8/site-packages/liquidctl/cli.py", line 414, in main
    selected = list(find_liquidctl_devices(**opts))
  File "/home/source/.local/lib/python3.8/site-packages/liquidctl/driver/__init__.py", line 59, in find_liquidctl_devices
    for dev in bus_cls().find_devices(**kwargs):
  File "/home/source/.local/lib/python3.8/site-packages/liquidctl/driver/smbus.py", line 67, in find_devices
    yield from i2c_bus.find_devices(drivers, **kwargs)
  File "/home/source/.local/lib/python3.8/site-packages/liquidctl/driver/smbus.py", line 93, in find_devices
    yield from drv.probe(self, **kwargs)
  File "/home/source/.local/lib/python3.8/site-packages/liquidctl/driver/ddr4.py", line 152, in probe
    eeprom = smbus.load_eeprom(spd_addr)
  File "/home/source/.local/lib/python3.8/site-packages/liquidctl/driver/smbus.py", line 170, in load_eeprom
    eeprom = self._i2c_dev.joinpath(dev, 'eeprom').read_bytes()
  File "/usr/lib/python3.8/pathlib.py", line 1230, in read_bytes
    return f.read()
OSError: [Errno 71] Protocol error
```

Additional information about the system:

- Affected device(s): Corsair Hydro H115i
- Operating system and version: Ubuntu Linux 20.04.3 LTS
- Liquidctl installation method: github; checkout master / v1.7.2 / v1.8.1
- Output of `liquidctl --version`: liquidctl v1.8.1.post126+gcc0050f (Linux-5.16.8-051608-generic-x86_64-with-glibc2.29)

jonasmalacofilho, 2022-03-03:
This is caused by a kernel bug revealed in 5.16.3.  More information in: #416.

closed 2022-03-03 by jonasmalacofilho

jonasmalacofilho, 2022-03-03:
If your distro doesn't yet ship the fix, a workaround may be to pass `--bus hid` or `--bus <usb number>`: when that's the case, liquidctl will, if possible, only probe devices on that specific bus.

panmarco83, 2022-03-03:
Would a newer kernel possibly have the bug repaired ?
Mainline kernel lists 5.16.9 to 12 available that could be installed, I just have not felt the need yet

panmarco83, 2022-03-03:
Looks like it is, I just installed 5.16.12 and booted over to it and liquidctl seems to work fine in userspace now.
Thanks for the help with this :)

jonasmalacofilho, 2022-03-03:
Yes, my patch was released as part of 5.16.10.

: )