liquidctl/liquidctl #433: EVGA HYBRID Kit for EVGA GeForce RTX 3090/3080 Ti/3080 XC3

mdeguzis, 2022-03-03:
Device type:  AIO
Product page:  https://www.evga.com/products/productlist.aspx?type=18&family=Cooling&chipset=GPU+HYBRID+Cooler (I specifically have the version without the RGB fans.
First-party software:  
Other commonly used tools: 

The device appears to support the following features:

 - monitoring of temperatures or other environment sensors:  yes
 - monitoring of fan or pump speeds or duty cycles: yes
 - monitoring of voltages, currents or power: 
 - configurable fan or pump speeds: 
 - configurable voltages or current/power limits: 
 - configurable lighting of embedded LEDs: 
 - configurable lighting of accessories like RGB fans or LED strips: 

The device is physically connected to the host system using:

- USB: 
- PCI-E: yes
- onboard the motherboard: 
- other: 

The device uses which protocols to communicate with the host system:

- HID: 
- USB: 
- I²C: 
- SMBus: 
- other: 

Additional information about the device:

Finally, I can help with:

- testing changes on Linux: yes
- testing changes on Windows: 
- testing changes on Mac OS: 
- attempting to capture USB traffic: yes
- attempting to capture I²C/SMBus traffic: tes
- analyzing traffic data: 
- documenting the protocol: yes
- implementing the driver: yes

assuming documentation and/or assistance will be provided.

jonasmalacofilho, 2022-03-03:
This is very likely related to #244.

Based on what OpenRGB has done, there's a particular issue that will need to be addressed for Linux, related to the use of `i2c_smbus_read_i2c_block_data()` and `i2c_smbus_write_i2c_block_data()`.[^1]

If you can, try to join the OpenRGB Discord thread: https://discord.com/channels/699861463375937578/941201973192065125

When things get a little to liquidctl-specific when can switch over to the liqudictl Discord, but I think it would be nice to first help figure out how to make that OpenRGB controller work on Linux.

[^1]: https://gitlab.com/CalcProgrammer1/OpenRGB/-/merge_requests/825

mdeguzis, 2022-03-03:
I wasn't able to pull up that thread. Can you give me an invite link from the channel and just note the name of the thread?

jonasmalacofilho, 2022-03-03:
OpenRGB: https://discord.gg/Dn9nqV77 -> #development -> EVGA Ampere on Linux

liquidctl: https://discord.gg/GyCBjQhqCd

jonasmalacofilho, 2022-07-03:
Triage: closing as the consensus is that this is not possible to implement with the current NVIDIA Linux drivers. Having followed the discussions on OpenRGB, I still have a tiny bit of hope that a non-obvious I2C transfer type might work, but various people looking into this haven't found it yet...

closed 2022-07-03 by jonasmalacofilho