liquidctl/liquidctl #434: Support for listing available channels?

mdeguzis, 2022-03-03:
I found what channels my Kraken Z AIO cooler supports by checking https://github.com/liquidctl/liquidctl/blob/main/liquidctl/driver/kraken3.py, but it would be nice to have an option `--list-channels` so I don't need to guess / check code. I looked here only because the ring color example did not work for me, now I know why.

```
# Available color channels and IDs for model Z coolers
_COLOR_CHANNELS_KRAKENZ = {
    'external': 0b001,
}
```

mdeguzis, 2022-03-03:
I guess you can disregard due to docs/ like https://github.com/liquidctl/liquidctl/blob/main/docs/kraken-x3-z3-guide.md, but would be nice to list them without checking the docs.

jonasmalacofilho, 2022-03-03:
I see your point.

What we could realistically do would be considerably limited, probably really only specifying the channel names (not modes, options, number of colors, etc.).  But it's worth _considering..._

---

Regarding the documentation, you should also be able to find this time or information in the `liquidctl(8)` man page.