liquidctl/liquidctl #435: Direct mode for RGB Fusion 2.0 controllers

jonasmalacofilho, 2022-03-03:
It has come up in a talk with OpenRGB developers that our Fusion 2.0 driver doesn't yet support Direct mode.

I'm not sure exactly what the device supports, and if there are any limitations, but it could be something to add to our driver at some point.  Possibly as an API-only mode (but that choice really depends on how exactly does it work).

For reference, the OpenRGB code for this controller is in [`OpenRGB/Controllers/GigabyteRGBFusion2USBController`](https://gitlab.com/CalcProgrammer1/OpenRGB/-/tree/master/Controllers/GigabyteRGBFusion2USBController).

---

@pde, this could be useful for the script you contributed, right?  If so, would you be willing to take a look into it?

@CaseySJ, as the original author of our driver, do you have any comments about this?  But feel free to ignore this, I saw that you're pretty advanced on a driver for the Aura USB controller... so I suspect you might not using a Gigabyte board anymore.

Chr1sNo, 2022-03-03:
Don't mind me, just throwing a comment up so as to follow the conversation.

Having said that: Of interest might be [the packet structure](https://gitlab.com/CalcProgrammer1/OpenRGB/-/blob/master/Controllers/GigabyteRGBFusion2USBController/GigabyteRGBFusion2USBController.h#L128) for the ITE chip.

CaseySJ, 2022-03-06:
@jonasmalacofilho 

If _Direct Mode_ on Gigabyte RGB Fusion 2.0 functions in a similar manner to the _Asus Aura (USB)_ controller I'm working on, then it requires a continuous flow of USB commands to the controller in order to address each LED individually for color and duration. This is conducive to a _service_ that stay running in the background, but not conducive to our fire-and-forget model of operation. The fire-and-forget model puts the controller in charge of dynamically modulating the LEDs.

I'm sure you know this already, but mentioning it anyway can provide context for others who're reading this.

I see you discovered my work-in-progress on the Aura LED (USB) controller! Maybe I can briefly hijack this thread to make a couple of points:

- There are apparently two versions of the Asus Aura controller on Z690 boards. My ProArt Z690-Creator uses `0x19af` whereas the Asus Z690 Formula uses `0x18f3` 
- I do not own the Z690 Formula so I am relying on someone else to test compatibility and provide WireShark captures. The controller seems to have some differences because the color modes that work on my board don't all seem to work on the _Formula_. 

Questions/Comments:

- I would like to submit a pull request because owners of ProArt Z690-Creator and other boards with `0x19af` can benefit right away. For the `0x18f3` controller, should I just remove it from the driver or keep it there, but clearly mark it as _Experimental_ in the docs?
- I had some trouble running the unit tests. Will try again. If I encounter errors I'll return and describe them in proper detail.

jonasmalacofilho, 2022-03-06:
> This is conducive to a service that stays running in the background, but not conducive to our fire-and-forget model. The fire-and-forget model puts the controller in charge of dynamically modulating the LEDs.

While the CLI works in fire-and-forget mode, our APIs don't assume that (though the ones exposed through the CLI do still need to support that usage too).  For example, there's an experimental software fan/pump speed control service in [`extra/yoda`](https://github.com/liquidctl/liquidctl/blob/main/extra/yoda) (and IIRC it too uses some APIs not exposed on the CLI, when those are available by the specific drivers).

So it might still make sense to implement these types of "special" APIs, particularly if anyone is interested in using them right now.  I guess it will depend if @pde thinks it would make the script he just contributed ([`extra/contrib/fusion_rgb_cycle.py`](https://github.com/liquidctl/liquidctl/blob/main/extra/contrib/fusion_rgb_cycle.py)) a few weeks ago.

jonasmalacofilho, 2022-03-06:
@CaseySJ,

_(Answering the rest in a separate comment, so that in a few days I can hide it as off-topic if it's starting getting in the way of discussing the Gigabyte thing)._

> I do not own the Z690 Formula so I am relying on someone else to test compatibility and provide WireShark captures. This has slowed progress, which is why my fork has been idle for several days. This controller seems to have some differences because the color modes that work on my board don't all seem to work on the Formula.

One other thing you could try is to take a look at OpenRGB's code and issues, and see how differently do they handle those controllers.

> I would like to submit a pull request because owners of ProArt Z690-Creator and other boards with 0x19af can benefit right away.

Great!

> For the 0x18f3 controller, should I just remove it from the driver or keep it there, but clearly mark it as Experimental in the docs?

I think it's simpler to remove it.  Marking it as experimental isn't enough, your 0x19af will already be marked that way: all devices go through at least one release, often two, before they we say that they stable.

What you could do instead is feature gate it with an `--unsafe` flag.  But that's a bit of work to do correctly, since in some APIs you have to return `None` or `[]` and warn, while in others you need to raise an exception.  Still, if you think the code already is somewhat useful for the 0x18f3 controller, and that having more people testing that would be useful, then take a look at the nvidia and/or ddr4 modules and drivers to see how they use these `--unsafe` features gates.

P.S. In those drivers, the `smbus` unsafe feature is a permanent requirement for I²C/SMBus devices; the other unsafe features are the ones being used to specifically gate devices that we're not comfortable enabling by default.

> I had some trouble running the unit tests. Will try again. If I encounter errors I'll return and describe them in proper detail.

Ok, feel free to open an issue, or reach out through Discord or email.  And we can also talk in the PR thread, once that's opened.

jonasmalacofilho, 2022-07-03:
Triage: closing as this would be a somewhat niche feature for most of our users, and there doesn't seem to be much interest in it.

closed 2022-07-03 by jonasmalacofilho