liquidctl/liquidctl #436: Corsair H100i not recognised

Englishman100, 2022-03-04:
I have dual-boot Arch Linux / Windows 11 setup and trying to get fan control, particularly for the H100i V2, working under Arch.

Unfortunately nothing is listed when typing 'liquidctl list' in a terminal window. I have attached the output of 'liquidctl list --debug' and 'lsusb -vt', both of which seem to show that '1b1c:0c04 Corsair Link Cooling Node' has been detected. 

First time trying to use liquidctl so could be a step I've missed?

[Liquidctl Debug.txt](https://github.com/liquidctl/liquidctl/files/8185673/Liquidctl.Debug.txt)

[Lsusb.txt](https://github.com/liquidctl/liquidctl/files/8185676/Lsusb.txt)

jonasmalacofilho, 2022-03-04:
According to those USB IDs, you have a CoolIT-based design which predates the models we currently support.  There is some initial work on a driver for these coolers, but it's stalled and help is need.  You can find more info about this in #142.

---

On a minor note, you device can't be a H100i v2 with those device IDs (and the USB descriptor saying "Cooling Node" also indicates to me that's a CoolIT model, because of how that protocol works).

At the same time, I think there's an error in our documentation, because I think your device may have been marketed as a H100i GT... if I'm right about that, we currently say we support it, when we don't.

So I'll keep this issue open for now with a "documentation" tag, but as soon as I fix that let's keep the rest of the discussion in the existing issue (#142).

Englishman100, 2022-03-04:
Here is a photo of the cooler I have just taken, definitely a Corsair H100i! Maybe the OS is incorrectly reporting it?

[IMG_20220304_122402025_HDR](https://user-images.githubusercontent.com/61411798/156763354-5c577249-843c-45f5-9b7e-06124cb687e7.jpg)
?

jonasmalacofilho, 2022-03-04:
CoolIT is one possible OEM manufacturer for the cooler, the other (main) one being Asetek.  Corsair has sourced (and still sources) coolers from both.  (The same happens with NZXT, EVGA, Fractal, MSI, etc... none of them actually manufacturers these coolers).

So yeah, yours is a H100i (probably GT, but could also be the no-suffix one).  Just not a H100i v2 or GTX.

jonasmalacofilho, 2022-03-04:
Oh, and regarding what the ~~OS~~ device reports though the USB string descriptors, that's expected for a CoolIT model of that time: they used a single VID:PID for a bunch of devices (coolers and fan controllers, mostly), and device identification was done in the upper level protocol.

closed 2022-03-04 by jonasmalacofilho

Englishman100, 2022-03-04:
Thanks for the explanation. Just to complete this discussion, attached is the model as reported by Corsair Link, H100i. It wasn't that long ago I bought it and because of that I assumed it was a V2, my apologies on that. I will follow the existing issue (#142) in the hope that it gets fixed (I'm almost entirely over to Arch Linux as my main platform, really just fan and RGB control outstanding, neither of which is essential, but nice to have)
[image]

commit 54629a7f (2022-03-04): docs: readme: list each Corsair GT/GTX/v2 model individually

The naming scheme used at time was very confusing: for example, the H80i
GT was a newer design made by Asetek, while the H100i GT was older and
made by CoolIT.

To avoid suggesting that we support coolers like the H100i GT, revert
from a simplified list back to individual SKUs.

Closes: #436 (partially, the rest is left for #142)