liquidctl/liquidctl #437: Aquacomputer LEAKSHIELD

lhupfeldt, 2022-03-07:
Device type: Loop pressure monitoring
Product page: https://shop.aquacomputer.de/product_info.php?products_id=4063
First-party software: Aquacomputer Aquasuite
Other commonly used tools: 

The device appears to support the following features:

 - monitoring of temperatures or other environment sensors: Pressure
 - monitoring of fan or pump speeds or duty cycles: 
 - monitoring of voltages, currents or power: 
 - configurable fan or pump speeds: 
 - configurable voltages or current/power limits: 
 - configurable lighting of embedded LEDs: 
 - configurable lighting of accessories like RGB fans or LED strips: 
 - Integrated OLED display: Yes

The device is physically connected to the host system using:

- USB: Yes
- PCI-E: No
- onboard the motherboard: 
- other: 

The device uses which protocols to communicate with the host system:

- HID: 
- USB: 
- I²C: 
- SMBus: 
- other: 

Additional information about the device:

Finally, I can help with:

- testing changes on Linux: Yes
- testing changes on Windows: 
- testing changes on Mac OS: 
- attempting to capture USB traffic: On Linux
- attempting to capture I²C/SMBus traffic: 
- analyzing traffic data: Possibly
- documenting the protocol: Possibly
- implementing the driver: Possibly

assuming documentation and/or assistance will be provided.

I do not own this device yet.
I'm a programmer (Python) so will be happy to assist. I have no experience in USB protocol development/debugging though.

jonasmalacofilho, 2022-03-07:
Hi @lhupfeldt,

It looks like it would be a very interesting (and probably straightforward) device to support.

After the reading the product page, I think a few more functions are supported than what you had already listed, so I added them to the list.  Also, after checking the rest, I removed the confirmed "no" entries, since they aren't necessary anymore.

> attempting to capture USB traffic: **On Linux**

Are you able to capture the USB traffic between the device and the (Windows-only) first-party tool using a Windows VM?

> I'm a programmer (Python) so will be happy to assist. I have no experience in USB protocol development/debugging though.

Great!  The USB protocol stuff can be very easy sometimes (although it depends on the device), and we're also here to help.

lhupfeldt, 2022-03-07:
Hi @jonasmalacofilho,
I have just ordered one. I will let you know when it arrives.

TheSkyHeart, 2022-03-07:
Hi Guys,

I was looking for a tools to support Aqua Computer Devices on Linux but found none, but I also found this  [Coolero](https://gitlab.com/codifryed/coolero) which is a front end UI to liquidctl, I even opened a feature request and the developer asked me to put a pull request here.

first I would like to thank you guys for the nice tools, and I would like to request for adding support for Aqua Computer Devices, Especially:
[OCTO](https://shop.aquacomputer.de/product_info.php?products_id=3832) 
[Quadro](https://shop.aquacomputer.de/product_info.php?products_id=3773)
[farbwerk 360](https://shop.aquacomputer.de/product_info.php?products_id=3798)
[D5 NEXT](https://shop.aquacomputer.de/product_info.php?products_id=3785)

I'm sure the Water Cooling  community will appreciate that, since I would say 60% of Professional Custom Water Cool PC uses one of those devices and the manufacturer application run's only on Windows.

I can contribute with any information needed since I have all the listed devices above, I can even contribute with providing one of those devices to the team if anyone resides in EU.

jonasmalacofilho, 2022-03-07:
@TheSkyHeart ,

Can you create a couple of new issues for those devices?  Otherwise we'll quickly derail this thread.

I think that, at least for now, you can group the three controllers in a single issue.  And the pump should get its own issue.

Also, those devices already have have mainline, out-of-tree, or WIP hwmon drivers:

- [aquacomputer_octo-hwmon](https://github.com/detrophy/aquacomputer_octo-hwmon)
- [aquacomputer_quadro-hwmon](https://github.com/leoratte/aquacomputer_quadro-hwmon)
- [aquacomputer_d5next](https://www.kernel.org/doc/html/latest/hwmon/aquacomputer_d5next.html)
- [(aquacomputer_d5next) Add support for Aquacomputer Farbwerk 360](https://patchwork.kernel.org/project/linux-hwmon/patch/20220227205625.8775-1-savicaleksa83@gmail.com/)

It's good to mention those in the new issues... and our drivers should try to use them, when possible.

> I can even contribute with providing one of those devices to the team if anyone resides in EU.

Sadly, AFAIK none of us are currently residing in the EU.

P.S. Thanks for the complements.

TheSkyHeart, 2022-03-07:
Hi @jonasmalacofilho 

Thanks a lot for the reply, I will open 2 new issues with all the information you posted.

jdarmetzki, 2022-11-05:
I guess this is a lot more complicated than it looks on first sight. The most interesting operational mode imo is the one with dynamic pressure. In this mode the vacuum is dynamically adjusted to fit the current pump rpm and flow. 
I don't know whether this is implemented in the driver or the firmware, but even if it is in the firmware the data has to be fed by an external application.

Finally there are quite a few settings related to the reservoir size, pump etc., and there are lots of values that can be read.

nyabinary, 2025-01-02:
any updates on this :3

lhupfeldt, 2025-01-02:
Sorry, no progress. I have the device, but didn't get the time to finish my build, so it is still not installed. It will be a while before I get back to it.

aleksamagicka, 2025-01-03:
Some parts of it are already reverse engineered in the [aquacomputer_d5next](https://github.com/aleksamagicka/aquacomputer_d5next-hwmon/blob/main/aquacomputer_d5next.c#L319-L350) driver.

lhupfeldt, 2025-01-03:
If anybody from the team is in the EU now and want to have a go at it, I will be happy to lend out my LEAKSHILD.