liquidctl/liquidctl #438: Aqua Computer OCTO, Quadro, Farbwerk 360

TheSkyHeart, 2022-03-07:
Device type:  Fan Controller, RGB Controller, Temp Sensor, Flow Sensor
Product page: 
[OCTO](https://shop.aquacomputer.de/product_info.php?products_id=3832)
[Quadro](https://shop.aquacomputer.de/product_info.php?products_id=3773)
[farbwerk 360](https://shop.aquacomputer.de/product_info.php?products_id=3798)
First-party software: AquaSuite
Other commonly used tools:  not tested

The device appears to support the following features:

 - monitoring of temperatures or other environment sensors: Yes, also Flow Sensors
 - monitoring of fan or pump speeds or duty cycles:  Yes
 - monitoring of voltages, currents or power:  Yes
 - configurable fan or pump speeds:  Yes
 - configurable voltages or current/power limits: 
 - configurable lighting of embedded LEDs: Yes
 - configurable lighting of accessories like RGB fans or LED strips: Yes

The device is physically connected to the host system using:

- USB:  it can be connected with normal USB-A using a specific Cable
- PCI-E: 
- onboard the motherboard: Yes Internal USB header
- other: 

The device uses which protocols to communicate with the host system:

- HID: 
- USB: 
- I²C: 
- SMBus: 
- other: 

Additional information about the device:

as mentioned by @jonasmalacofilho in issue #437:
those devices already have have mainline, out-of-tree, or WIP hwmon drivers:
[aquacomputer_octo-hwmon](https://github.com/detrophy/aquacomputer_octo-hwmon)
[aquacomputer_quadro-hwmon](https://github.com/leoratte/aquacomputer_quadro-hwmon)

Finally, I can help with:

- testing changes on Linux:  Yes
- testing changes on Windows:  Yes
- testing changes on Mac OS: 
- attempting to capture USB traffic:  Yes
- attempting to capture I²C/SMBus traffic: Yes if it's guided 
- analyzing traffic data: 
- documenting the protocol: 
- implementing the driver: 

assuming documentation and/or assistance will be provided.

aleksamagicka, 2022-03-26:
Hi! Author of the aquacomputer_d5next driver here. Octo support is currently in process of being reviewed/in iteration for mainline, and a Quadro is on its way to me. I'll look at the code in this repo (not necessarily immediately, because college) and see what I can do. 😄 Same goes for the Farbwerk 360, Octo, Quadro, since I know most of their usable register offsets.

jonasmalacofilho, 2022-03-27:
Oh, that'll be great, @aleksamagicka.  Let me know if you need any help with the liquidctl-specific stuff.

jonasmalacofilho, 2022-08-11:
There's now initial monitoring support for the Octo and the Farbwerk 360 n the latest `main` branch.  Please give it a try.