liquidctl/liquidctl #440: Corsair H115i Pro XT insufficient permissions inspite of setting udev rules

FlameSky-S, 2022-03-09:
I did the following setups. However, when running `liquidctl -m corsair initialize` the insufficient permissions error poped out. Am I missing something?

- liquidctl installed via pip, version is 1.8.1. 
- Accessing through remote ssh session. 
- udev rules placed under `/etc/udev/rules.d`. 
- Edited H115i Pro XT line as follows:
  ```bash
  SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c21", GROUP="plugdev", MODE="0660"
  ```
- User added to `plugdev` group
- Reloaded udev rules using `udevadm control --reload-rules`

`sudo liquidctl` works fine, but I'd like to make it work for unpreviledged users. 

Debug output:
```bash
$ liquidctl -m corsair initialize -g
[DEBUG] (cli) (main): version: liquidctl v1.8.1
[DEBUG] (cli) (main): platform: Linux-5.13.0-30-generic-x86_64-with-glibc2.10
[DEBUG] (smbus) (find_devices): searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] (smbus) (find_devices): ignoring 0-0052, cannot infer bus number
[DEBUG] (smbus) (find_devices): found I²C bus i2c-3
[DEBUG] (smbus) (find_devices): found I²C bus i2c-10
[DEBUG] (smbus) (find_devices): ignoring 0-0050, cannot infer bus number
[DEBUG] (smbus) (find_devices): found I²C bus i2c-1
[DEBUG] (smbus) (find_devices): found I²C bus i2c-8
[DEBUG] (smbus) (find_devices): ignoring 0-0037, cannot infer bus number
[DEBUG] (smbus) (find_devices): found I²C bus i2c-6
[DEBUG] (smbus) (find_devices): ignoring 0-0053, cannot infer bus number
[DEBUG] (smbus) (find_devices): found I²C bus i2c-4
[DEBUG] (smbus) (find_devices): ignoring 0-0051, cannot infer bus number
[DEBUG] (smbus) (find_devices): found I²C bus i2c-2
[DEBUG] (smbus) (find_devices): found I²C bus i2c-0
[DEBUG] (smbus) (find_devices): found I²C bus i2c-9
[DEBUG] (smbus) (find_devices): found I²C bus i2c-7
[DEBUG] (smbus) (find_devices): ignoring 0-0036, cannot infer bus number
[DEBUG] (smbus) (find_devices): found I²C bus i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus (CommanderCore, CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] (usb) (find_devices): found HID device 048d:5702
[DEBUG] (usb) (find_devices): found HID device 048d:5702
[DEBUG] (usb) (find_devices): found HID device 1b1c:0c21
[DEBUG] (hydro_platinum) (probe): instantiated HydroPlatinum driver for Corsair Hydro H115i Pro XT
[DEBUG] (usb) (find_devices): searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc, CorsairAsetekProDriver)
[DEBUG] (usb) (find_devices): found USB device 1d6b:0003
[DEBUG] (usb) (find_devices): found USB device 8087:0026
[DEBUG] (usb) (find_devices): found USB device 1b1c:0c21
[DEBUG] (usb) (find_devices): found USB device 05e3:0608
[DEBUG] (usb) (find_devices): found USB device 048d:5702
[DEBUG] (usb) (find_devices): found USB device 05e3:0608
[DEBUG] (usb) (find_devices): found USB device 1d6b:0002
[DEBUG] (cli) (main): device: Corsair Hydro H115i Pro XT
[INFO] (cli) (log_error): open failed
Traceback (most recent call last):
  File "/home/mhs/anaconda3/lib/python3.8/site-packages/liquidctl/cli.py", line 422, in main
    with dev.connect(**opts):
  File "/home/mhs/anaconda3/lib/python3.8/site-packages/liquidctl/driver/hydro_platinum.py", line 181, in connect
    ret = super().connect(**kwargs)
  File "/home/mhs/anaconda3/lib/python3.8/site-packages/liquidctl/driver/usb.py", line 118, in connect
    self.device.open()
  File "/home/mhs/anaconda3/lib/python3.8/site-packages/liquidctl/driver/usb.py", line 384, in open
    self.hiddev.open_path(self.hidinfo['path'])
  File "hidraw.pyx", line 125, in hidraw.device.open_path
OSError: open failed
[ERROR] (cli) (log_error): Error: could not open Corsair Hydro H115i Pro XT, possibly due to insufficient permissions
```

---

Additional information about the system:

- Affected device(s):  Corsair Hydro H115i Pro XT
- Operating system and version: Ubuntu 20.04
- Liquidctl installation method:  PyPI
- Output of `liquidctl --version`: liquidctl v1.8.1

jonasmalacofilho, 2022-03-09:
Can I ask why did you edit the udev rule?  Ubuntu 20.04 is systemd-based, no?

---

Anyway, did you restart the system, or, alternatively (and IIRC), did you reload the `usbhid` kernel module?

`udevadm control --reload-rules` is _not_ enough, it just reloads the rules, but doesn't trigger them.  In theory, you could use `udevadm trigger` to trigger them... but that still wouldn't be enough for a HID with its hidraw device is already present; this is weird, I not something I've seen documented, but the permissions appear to set _only_ when creating the device, but not if it's already there.

This is why you either need to reboot or force the recreation of the hidraw device (by reloading the suitable kernel driver).

jonasmalacofilho, 2022-03-09:
Hm, from the way the `uaccess` tag is described in the [Arch Wiki](https://wiki.archlinux.org/title/Users_and_groups#Pre-systemd_groups),

> [...] [udev](https://wiki.archlinux.org/title/Udev) marking the devices with a `uaccess` [tag](https://github.com/systemd/systemd/blob/main/src/login/70-uaccess.rules.in) and `logind` assigning the permissions to users dynamically via [ACLs](https://wiki.archlinux.org/title/ACL) according to which session is currently active.

it's possible that just logging off and on again might be enough.

I though a reboot might be necessary because it would force the recreating of the hidraw device; but it seems that it's `logind` that applies the permissions set by udev, so maybe the device doesn't need to be recreated.

FlameSky-S, 2022-03-09:
Thank you for your quick response. 

> Can I ask why did you edit the udev rule?

I'm using liquidctl through remote ssh sessions and I saw these comments in the rules file.

```bash
# These rules also assume that only physical sessions (or "seats") need
# unprivileged access to the devices.  In case headless sessions are also
# expected to interactively run liquidctl, using GROUP and MODE might be
# preferable to TAG+="uaccess".
```

Should I edit the udev rules or not?

> it's possible that just logging off and on again might be enough.

Well, it's good news because the server has multiple users logged in and is really busy. However, I'm not quite sure how to log off and on since all users log in through ssh. Simply opening another ssh session doesn't work.

jonasmalacofilho, 2022-03-09:
> I'm using liquidctl through remote ssh sessions and I saw these comments in the rules file.
> Should I edit the udev rules or not?

In that case, I think so.

> Well, it's good news because the server has multiple users logged in and is really busy. However, I'm not quite sure how to log off and on since all users log in through ssh. Simply opening another ssh session doesn't work.

Yeah, I don't think that works outside of `uaccess`.  You probably do need `GROUP` and `MODE`.

Also, trigger the rules now with `udevadm trigger` (as root), just so that step isn't forgotten.

Next, in order to not reboot, you need to get the hidraw device to receive the correct/new ownership and permissions.  I see three ways for that:

- **if you're sure that you don't have any other critical or in-use HIDs on that server**, you should be able to get away with simply `modprobe -r usbhid ; modprobe usbhid`... _but there's a risk that I'm wrong or that you forget about some HID that's in use, and you get your users mad;_
- alternatively, you can unbind `usbhid` from that _specific_ device, and then rebind it (the process looks something like [this](https://github.com/liquidctl/liquidctl#device-not-listed-linux));
- finally, you can just `chown` and `chmod` the necessary hidraw device for now, and after the next boot just check that the rules did their job that time.

jonasmalacofilho, 2022-03-09:
Glad to hear!