liquidctl/liquidctl #442: Cannot list Corsair H100i RGB Pro XT

hibby50, 2022-03-12:
I don't think this is a case of bad corsair naming (at least I hope not) driver in lsusb is usbhid.

`[DEBUG] (cli) (main): version: liquidctl v1.8.1
[DEBUG] (cli) (main): platform: Linux-5.16.11-1-default-x86_64-with-glibc2.34
[DEBUG] (smbus) (find_devices): searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] (smbus) (find_devices): found I²C bus i2c-3
[DEBUG] (smbus) (find_devices): found I²C bus i2c-10
[DEBUG] (smbus) (find_devices): found I²C bus i2c-1
[DEBUG] (smbus) (find_devices): found I²C bus i2c-17
[DEBUG] (smbus) (find_devices): found I²C bus i2c-8
[DEBUG] (smbus) (find_devices): found I²C bus i2c-15
[DEBUG] (smbus) (find_devices): found I²C bus i2c-6
[DEBUG] (smbus) (find_devices): found I²C bus i2c-13
[DEBUG] (smbus) (find_devices): found I²C bus i2c-4
[DEBUG] (smbus) (find_devices): found I²C bus i2c-11
[DEBUG] (smbus) (find_devices): found I²C bus i2c-2
[DEBUG] (smbus) (find_devices): found I²C bus i2c-0
[DEBUG] (smbus) (find_devices): found I²C bus i2c-9
[DEBUG] (smbus) (find_devices): found I²C bus i2c-16
[DEBUG] (smbus) (find_devices): found I²C bus i2c-7
[DEBUG] (smbus) (find_devices): found I²C bus i2c-14
[DEBUG] (smbus) (find_devices): found I²C bus i2c-5
[DEBUG] (smbus) (find_devices): found I²C bus i2c-12
[DEBUG] (usb) (find_devices): searching HidapiBus (CommanderCore, CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] (usb) (find_devices): found HID device 046d:4069
[DEBUG] (usb) (find_devices): found HID device 046d:4069
[DEBUG] (usb) (find_devices): found HID device 046d:4069
[DEBUG] (usb) (find_devices): found HID device 046d:4069
[DEBUG] (usb) (find_devices): found HID device 046d:4069
[DEBUG] (usb) (find_devices): found HID device 046d:4069
[DEBUG] (usb) (find_devices): found HID device 046d:4069
[DEBUG] (usb) (find_devices): found HID device 046d:4069
[DEBUG] (usb) (find_devices): found HID device 046d:4069
[DEBUG] (usb) (find_devices): found HID device 046d:4069
[DEBUG] (usb) (find_devices): found HID device 046d:4069
[DEBUG] (usb) (find_devices): found HID device 046d:4069
[DEBUG] (usb) (find_devices): found HID device 046d:4069
[DEBUG] (usb) (find_devices): found HID device 046d:4069
[DEBUG] (usb) (find_devices): found HID device 046d:4069
[DEBUG] (usb) (find_devices): found HID device 046d:4069
[DEBUG] (usb) (find_devices): found HID device 046d:c52b
[DEBUG] (usb) (find_devices): found HID device 046d:c52b
[DEBUG] (usb) (find_devices): found HID device 046d:c52b
[DEBUG] (usb) (find_devices): found HID device 1b1c:0c1a
[DEBUG] (usb) (probe): instantiated CommanderPro driver for Corsair Lighting Node Core
[DEBUG] (usb) (find_devices): found HID device 1b1c:0c2d
[DEBUG] (usb) (find_devices): found HID device 1462:7d30
[DEBUG] (usb) (find_devices): found HID device 1462:7d30
[DEBUG] (usb) (find_devices): found HID device 0db0:005a
[DEBUG] (usb) (find_devices): found HID device 2f68:0042
[DEBUG] (usb) (find_devices): found HID device 2f68:0042
[DEBUG] (usb) (find_devices): found HID device 2f68:0042
[DEBUG] (usb) (find_devices): found HID device 2f68:0042
[DEBUG] (usb) (find_devices): found HID device 2f68:0042
[DEBUG] (usb) (find_devices): found HID device 2f68:0042
[DEBUG] (usb) (find_devices): found HID device 2f68:0042
[DEBUG] (usb) (find_devices): searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc, CorsairAsetekProDriver)
[DEBUG] (usb) (find_devices): found USB device 1d6b:0003
[DEBUG] (usb) (find_devices): found USB device 2f68:0042
[DEBUG] (usb) (find_devices): found USB device 046d:085b
[DEBUG] (usb) (find_devices): found USB device 0db0:005a
[DEBUG] (usb) (find_devices): found USB device 1462:7d30
[DEBUG] (usb) (find_devices): found USB device 8087:0032
[DEBUG] (usb) (find_devices): found USB device 1b1c:0c2d
[DEBUG] (usb) (find_devices): found USB device 1b1c:0c1a
[DEBUG] (usb) (find_devices): found USB device 05e3:0608
[DEBUG] (usb) (find_devices): found USB device 05e3:0608
[DEBUG] (usb) (find_devices): found USB device 046d:c52b
[DEBUG] (usb) (find_devices): found USB device 1d6b:0002
Device #0: Corsair Lighting Node Core
├── Vendor ID: 0x1b1c
├── Product ID: 0x0c1a
├── Release number: 0x0009
├── Serial number: 04E00213E01888AFA69E3660001C00F5
├── Bus: hid
├── Address: /dev/hidraw8
└── Driver: CommanderPro
[DEBUG] (cli) (_list_devices_human): hierarchy: UsbHidDriver, BaseUsbDriver, BaseDriver
`
`╰─ lsusb -vt                                   
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/9p, 20000M/x2
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 2: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 1462:7d30 Micro Star International 
    |__ Port 3: Dev 3, If 6, Class=Audio, Driver=snd-usb-audio, 480M
        ID 0db0:005a Micro Star International 
    |__ Port 3: Dev 3, If 4, Class=Audio, Driver=snd-usb-audio, 480M
        ID 0db0:005a Micro Star International 
    |__ Port 3: Dev 3, If 2, Class=Audio, Driver=snd-usb-audio, 480M
        ID 0db0:005a Micro Star International 
    |__ Port 3: Dev 3, If 0, Class=Audio, Driver=snd-usb-audio, 480M
        ID 0db0:005a Micro Star International 
    |__ Port 3: Dev 3, If 7, Class=Human Interface Device, Driver=usbhid, 480M
        ID 0db0:005a Micro Star International 
    |__ Port 3: Dev 3, If 5, Class=Audio, Driver=snd-usb-audio, 480M
        ID 0db0:005a Micro Star International 
    |__ Port 3: Dev 3, If 3, Class=Audio, Driver=snd-usb-audio, 480M
        ID 0db0:005a Micro Star International 
    |__ Port 3: Dev 3, If 1, Class=Audio, Driver=snd-usb-audio, 480M
        ID 0db0:005a Micro Star International 
    |__ Port 4: Dev 4, If 3, Class=Audio, Driver=snd-usb-audio, 480M
        ID 046d:085b Logitech, Inc. 
    |__ Port 4: Dev 4, If 1, Class=Video, Driver=uvcvideo, 480M
        ID 046d:085b Logitech, Inc. 
    |__ Port 4: Dev 4, If 2, Class=Audio, Driver=snd-usb-audio, 480M
        ID 046d:085b Logitech, Inc. 
    |__ Port 4: Dev 4, If 0, Class=Video, Driver=uvcvideo, 480M
        ID 046d:085b Logitech, Inc. 
    |__ Port 5: Dev 5, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        ID 2f68:0042  
    |__ Port 5: Dev 5, If 2, Class=Human Interface Device, Driver=usbhid, 12M
        ID 2f68:0042  
    |__ Port 5: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 2f68:0042  
    |__ Port 10: Dev 6, If 2, Class=Human Interface Device, Driver=usbhid, 12M
        ID 046d:c52b Logitech, Inc. Unifying Receiver
    |__ Port 10: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 046d:c52b Logitech, Inc. Unifying Receiver
    |__ Port 10: Dev 6, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        ID 046d:c52b Logitech, Inc. Unifying Receiver
    |__ Port 11: Dev 7, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 05e3:0608 Genesys Logic, Inc. Hub
    |__ Port 13: Dev 8, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 05e3:0608 Genesys Logic, Inc. Hub
        |__ Port 4: Dev 11, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            ID 1b1c:0c2d Corsair 
        |__ Port 2: Dev 10, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            ID 1b1c:0c1a Corsair 
    |__ Port 14: Dev 9, If 0, Class=Wireless, Driver=btusb, 12M
        ID 8087:0032 Intel Corp. AX210 Bluetooth
    |__ Port 14: Dev 9, If 1, Class=Wireless, Driver=btusb, 12M
        ID 8087:0032 Intel Corp. AX210 Bluetooth
`

jonasmalacofilho, 2022-03-12:
Hi @hibby50,

I think you're right, this doesn't look like another instance of bad naming by Corsair.  It's possible that there's a new product ID around...

Can you show me the output of `lsusb -v -d1b1c:0c2d`?

---

Can you also confirm the marketing branding of your device (i.e. what's on the box)?  I'm guessing it's indeed "Corsair H100i RGB Pro XT", but if you got that string from somewhere else, it can be a lot be a lot less reliable.  Alternatively, if you can find the the `CW-********-**` SKU number on the box or or on the device, that'll work too.

hibby50, 2022-03-12:
Hey @jonasmalacofilho appreciate you looking into this and helping me out

Unfortunately, I got this PC from a vendor, so I do not have any retail packaging. The PC was just built this month though, so it is possibly there is a new product ID. FWIW on my PO from the vendor 'FA-WATER-517    CORSAIR HYDRO iCUE H100I PRO XT RGB 240MM LIQUID CPU COOLING SYSTEM W/ COPPER COLD PLA`

I believe FA-WATER-517 is their internal SKU

```
╰─ sudo lsusb -v -d1b1c:0c2d
[sudo] password for root: 

Bus 001 Device 011: ID 1b1c:0c2d Corsair H100iRGBPROXT
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c2d 
  bcdDevice            1.00
  iManufacturer           1 Corsair Inc.
  iProduct                2 H100iRGBPROXT
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               7
Device Status:     0x0001
  Self Powered
```

jonasmalacofilho, 2022-03-12:
It seems that there's a whole batch of Pro XT coolers with new USB PIDs.

I don't see any reference that there has been a protocol change, but it's probably too soon to know that...

Anyway, I added the new IDs I found (including yours) to the driver in a test branch.  Do you want to just give it a try?

Alternatively, if you can capture some traffic between iCue (perhaps on a Windows VM) and your device, that would allow me to have a better idea of what (if anything) has changed.

---

The untested branch with (just) the new IDs is: [`new-pro-xt-pids`](https://github.com/liquidctl/liquidctl/tree/new-pro-xt-pids).

hibby50, 2022-03-12:
Many thanks, I'll give this a go later today and report back

hibby50, 2022-03-12:
It seems to be working just fine on that branch!

```
╰─ liquidctl status
Corsair Hydro H100i Pro XT
├── Liquid temperature    28.4  °C
├── Fan 1 speed            418  rpm
├── Fan 1 duty              22  %
├── Fan 2 speed            385  rpm
├── Fan 2 duty              22  %
├── Pump speed            1624  rpm
└── Pump duty               55  %
```

I'm new to this software, anything you'd like me to test specifically?

jonasmalacofilho, 2022-03-12:
First, don't forget to `initialize` the device.

Afterward, you can also try changing the lighting, setting fan speeds and changing the pump mode.  Our docs show how to do these in more detail: <https://github.com/liquidctl/liquidctl/blob/main/docs/corsair-platinum-pro-xt-guide.md>.

---

Can you also show me the output from `initialize` and `status --debug`?

hibby50, 2022-03-12:
I was able to successfully set each fans' speed and toggle the LEDs off, I pasted the output you requested below. Let me know if there is anything else I can do!

```
╰─ liquidctl -n 1 initialize
Corsair Hydro H100i Pro XT
└── Firmware version    2.0.6  
```
```
╰─ liquidctl -n 1 initialize --debug
[DEBUG] (cli) (main): liquidctl: 1.9.0.dev145+gba1aadd
[DEBUG] (cli) (main): platform: Linux-5.16.11-1-default-x86_64-with-glibc2.34
[DEBUG] (cli) (_log_requirements): python: 3.8.12 (default, Aug 31 2021, 01:23:42) [GCC]
[DEBUG] (cli) (_log_requirements): colorlog: 6.6.0
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.11.0.post2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1
[DEBUG] (cli) (_log_requirements): smbus: 1.1.post2
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-10
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-17
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-15
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-13
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-11
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-16
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-14
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-12
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: CommanderCore, CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1a (usage_page=0xffc0 usage=0x0001)
[DEBUG] (usb) (probe): found CommanderPro: Corsair Lighting Node Core
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c2d (usage_page=0xff00 usage=0x0001)
[DEBUG] (hydro_platinum) (probe): instantiated HydroPlatinum driver for Corsair Hydro H100i Pro XT
[DEBUG] (cli) (main): device: Corsair Hydro H100i Pro XT
[DEBUG] (keyval) (__init__): data in /run/user/1000/liquidctl/vid1b1c_pid0c2d/loc9
[DEBUG] (keyval) (store): stored leds_enabled=0 (in /run/user/1000/liquidctl/vid1b1c_pid0c2d/loc9/leds_enabled)
[DEBUG] (keyval) (store): stored pump_mode=1 (in /run/user/1000/liquidctl/vid1b1c_pid0c2d/loc9/pump_mode)
[DEBUG] (keyval) (load): no data (file) found for fan1_mode
[DEBUG] (hydro_platinum) (_missing_): falling back to FIXED_DUTY for _FanMode(None)
[DEBUG] (keyval) (load): no data (file) found for fan1_duty
[... 89 lines cut ...]
[DEBUG] (usb) (find_devices): USB device: 1b1c:0c2d
[DEBUG] (usb) (find_devices): USB device: 1b1c:0c1a
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 046d:c52b
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: Corsair Lighting Node Core
[DEBUG] (keyval) (__init__): data in /run/user/1000/liquidctl/vid1b1c_pid0c1a/loc8
[DEBUG] (commander_pro) (get_status): only Commander Pro and Obsidian 1000D report status
[DEBUG] (cli) (main): device: Corsair Hydro H100i Pro XT
[DEBUG] (keyval) (__init__): data in /run/user/1000/liquidctl/vid1b1c_pid0c2d/loc9
[... cut ...]
[DEBUG] (usb) (read): read 64 bytes: ff:18:20:06:00:03:00:ef:17:00:02:ff:e8:03:ff:6c:09:02:ff:e8:03:ff:64:09:01:bf:00:00:bf:48:03:00:00:00:00:00:2d:01:04:00:05:2d:32:37:41:33:20:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:9e
Corsair Hydro H100i Pro XT
├── Liquid temperature    23.9  °C
├── Fan 1 speed           2412  rpm
├── Fan 1 duty             100  %
├── Fan 2 speed           2404  rpm
├── Fan 2 duty             100  %
├── Pump speed             840  rpm
└── Pump duty               75  %
```

jonasmalacofilho, 2022-03-12:
Almost everything looks good.

However, there's something weird going on with the pump mode/duty/speed:

```
# before `initialize`, cooler state unknown
├── [ ... ]
├── Pump speed            1624  rpm
└── Pump duty               55  %
```
```
# after `initialize`, which should have reset the pump mode to "balanced"
├── [ ... ]
├── Pump speed             840  rpm
└── Pump duty               75  %
```

Any chance the pump speed rises after a while, making it match the supposedly higher duty cycle?

---

P.S. I also noticed that the firmware version is 2.x.x, which isn't something we've seen before on these devices.

hibby50, 2022-03-12:
Yes it actually shipped with 2.0.9 but I did a force fw upgrade that brought it to 2.0.6 last night due to an unrelated issue. The other interesting thing with this one is I read on the doc you linked me

>Animations are not supported at the hardware level, and require successive invocations of the commands shown above, or use of the liquidctl APIs.

This is not the case on this model/firmware. I set an animation in icue on windows and it persists in bios/linux. So to speak to our first couple comments, this is indeed another case of corsair bad naming :) 

I rebooted and reproduced, that is indeed an issue.

```
$ liquidctl status
Corsair Hydro H100i Pro XT
├── Liquid temperature    27.1  °C
├── Fan 1 speed            321  rpm
├── Fan 1 duty              20  %
├── Fan 2 speed            288  rpm
├── Fan 2 duty              20  %
├── Pump speed            1623  rpm
└── Pump duty               55  %

$ liquidctl -n 1 initialize
```
wait 4 minutes

```
╰─ liquidctl status
Corsair Hydro H100i Pro XT
├── Liquid temperature    24.6  °C
├── Fan 1 speed           2457  rpm
├── Fan 1 duty             100  %
├── Fan 2 speed           2443  rpm
├── Fan 2 duty             100  %
├── Pump speed             843  rpm
└── Pump duty               75  %

```

hibby50, 2022-03-12:
Almost seems to be inverse??

```
$ liquidctl -n 1 initialize --pump-mode extreme
Corsair Hydro H100i Pro XT
└── Firmware version    2.0.6  

$  liquidctl status                             
Corsair Hydro H100i Pro XT
├── Liquid temperature    23.8  °C
├── Fan 1 speed           2443  rpm
├── Fan 1 duty             100  %
├── Fan 2 speed           2443  rpm
├── Fan 2 duty             100  %
├── Pump speed               0  rpm
└── Pump duty              100  %

$ liquidctl -n 1 initialize --pump-mode quiet  
Corsair Hydro H100i Pro XT
└── Firmware version    2.0.6  

$ liquidctl status                           
Corsair Hydro H100i Pro XT
├── Liquid temperature    22.8  °C
├── Fan 1 speed           2444  rpm
├── Fan 1 duty             100  %
├── Fan 2 speed           2443  rpm
├── Fan 2 duty             100  %
├── Pump speed            1540  rpm
└── Pump duty               55  %
```

jonasmalacofilho, 2022-03-12:
> This is not the case on this model/firmware. I set an animation in icue on windows and it persists in bios/linux.

That's probably what #193 is about.  I guess the documentation could be fixed to say that some firmware versions do support it, but we don't yet.  I might do that later today.  Nonetheless, thanks for reporting the discrepancy.

---

> I rebooted and reproduced, that is indeed an issue.

Yeah, and that seems like a significant issue.

Can you try capturing the traffic between iCue and the device?  We have a few guides on how to that in our `docs/` folder.

> Almost seems to be inverse??

Indeed, it partially seems to be the case.  But on top of that there's clearly something wrong with the speed vs. duty cycle being reported.

> Yes it actually shipped with 2.0.9 but I did a force fw upgrade that brought it to 2.0.6 last night due to an unrelated issue.

Can I ask what was this unrelated issue?  Even if it doesn't matter now, it could in the future.

By the way, is iCue reporting a sensible pump speed vs. duty cycle?  While I think the issue is more likely in liquidctl (either a bug or something that changed with the v2 firmware/protocol), it could also, at least hypothetically, be a firmware bug.

hibby50, 2022-03-12:
> Can you try capturing the traffic between iCue and the device? We have a few guides on how to that in our docs/ folder.
 Sure thing, see attached [h100i-fw2x.tar.gz](https://github.com/liquidctl/liquidctl/files/8238475/h100i-fw2x.tar.gz)

>Can I ask what was this unrelated issue? Even if it doesn't matter now, it could in the future.
Totally silly, one of the fans was at max RPM and I couldn't get it to slow down, ended up being a loose connection in the back of the case :)

>By the way, is iCue reporting a sensible pump speed vs. duty cycle? 
I didn't see duty cycle in icue, but numbers seem mostly reasonable. certainly never states 0 RPM as it did in the output above

jonasmalacofilho, 2022-03-12:
> Sure thing, see attached [h100i-fw2x.tar.gz](https://github.com/liquidctl/liquidctl/files/8238475/h100i-fw2x.tar.gz)

Thanks, I'll take a look as soon as I can.

---

> I didn't see duty cycle in icue, but numbers seem mostly reasonable. certainly never states 0 RPM as it did in the output above

I missed that, but that's not immediately concerning to me... probably just an artifact, perhaps an extra incoming packet that we need to ignore.

What's looks to be concerning is how a higher duty cycle can translate to a lower rotational speed:

| reported duty cycle | reported rotational speed |
| --: | --: |
| `55%` | `1623 ± 1 rpm` |
| `75%` | `842 ± 2 rpm` |

jonasmalacofilho, 2022-04-08:
@hibby50,

First, I'm sorry it took me a while to actually look at your capture in detail.

It seems that you filtered/sent me the data for the wrong device (the capture shows data for the Lighting Node Core).  Can you redo the capture for the H100i Pro XT?  And since that will be necessary, can you also make sure to change some fan & pump settings while the capture is going on?

Thanks!

hibby50, 2022-04-08:
No apology necessary, sure thing! I'm a bit tied up, leaving on a business trip for a week Monday. But if not this weekend then next

hibby50, 2022-06-16:
Hopefully I filtered correctly this time @jonasmalacofilho
[capture2.pcapng.gz](https://github.com/liquidctl/liquidctl/files/8918387/capture2.pcapng.gz)

vincentcheongws, 2022-11-04:
> Yes it actually shipped with 2.0.9 but I did a force fw upgrade that brought it to 2.0.6 last night due to an unrelated issue. The other interesting thing with this one is I read on the doc you [...]

Hey @hibby50, I now face the same issue as you where the pump speed is lower at a higher pump duty. I have also noticed that when my pump profile is set to 'extreme' the pump fails and requires a reset. I can't help but suspect that all these issues are due to the firmware. Any chance you have gotten your firmware reverted back to v2.0.9?

mikelolasagasti, 2024-11-23:
I've a `Corsair Hydro H100i PRO XT`, device `1b1c:0c2d`, with Firmware version `2.0.9`. I've been able to detect it using `new-pro-xt-pids` branch and configure fan speeds.

What would be required in order to get the branch merged for next release?