liquidctl/liquidctl #445: 1.9.0 hwmon duty reading error - no pwm1_input file

codifryed, 2022-04-05:
Hi, first thanks for the new release :+1: 

When trying to get the status with liquidctl 1.9.0 for my nzxtsmart2 device through hwmon it throws an error.
`ERROR: unexpected OS error with NZXT RGB & Fan Controller: FileNotFoundError(2, 'No such file or directory')`

```
liquidctl list
Device #0: NZXT RGB & Fan Controller
Device #1: NZXT Kraken X (X53, X63 or X73)

liquidctl initialize all
NZXT RGB & Fan Controller
└── Firmware version    1.13.0

NZXT Kraken X (X53, X63 or X73)
├── Firmware version       2.1.0
├── Pump Logo LEDs      detected
└── Pump Ring LEDs      detected

# shortened for pertinent info:
liquidctl status --debug
[DEBUG] (cli) (main): device: NZXT RGB & Fan Controller
[INFO] (smart_device) (get_status): bound to nzxt_smart2 kernel driver, reading status from hwmon
[DEBUG] (hwmon) (get_string): read fan1_input: 839
[INFO] (cli) (log): detailed error: unexpected OS error with NZXT RGB & Fan Controller: FileNotFoundError(2, 'No such file or directory')
Traceback (most recent call last):
File "/usr/lib/python3.10/site-packages/liquidctl/cli.py", line 463, in main
status = dev.get_status(**opts)
File "/usr/lib/python3.10/site-packages/liquidctl/driver/smart_device.py", line 570, in get_status
return self._get_status_from_hwmon()
File "/usr/lib/python3.10/site-packages/liquidctl/driver/smart_device.py", line 552, in _get_status_from_hwmon
ret.append((f'Fan {n} duty', self._hwmon.get_int(f'pwm{n}_input') * 100. / 255, '%')),
File "/usr/lib/python3.10/site-packages/liquidctl/driver/hwmon.py", line 36, in get_int
return int(self.get_string(name))
File "/usr/lib/python3.10/site-packages/liquidctl/driver/hwmon.py", line 31, in get_string
value = (self.path / name).read_text().rstrip()
File "/usr/lib/python3.10/pathlib.py", line 1132, in read_text
with self.open(mode='r', encoding=encoding, errors=errors) as f:
File "/usr/lib/python3.10/pathlib.py", line 1117, in open
return self._accessor.open(self, mode, buffering, encoding, errors,
FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/hidraw/hidraw4/device/hwmon/hwmon4/pwm1_input'
[ERROR] (cli) (log): unexpected OS error with NZXT RGB & Fan Controller: FileNotFoundError(2, 'No such file or directory')
[DEBUG] (cli) (main): device: NZXT Kraken X (X53, X63 or X73)
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (read): read 64 bytes: 75:02:3a:00:18:00:13:51:48:48:30:33:30:31:01:21:03:b7:05:2c:2c:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
NZXT Kraken X (X53, X63 or X73)
├── Liquid temperature    33.3  °C
├── Pump speed            1463  rpm
└── Pump duty               44  %

# just to verify:
cat /sys/class/hidraw/hidraw4/device/hwmon/hwmon4/pwm1
128
```

Dev TLDR:
At first look it appears to be a small oversight, in several places in the code I see the use of "pwm{n}_input" when trying to get fan/pump duty from hwmon. This file of course does not exist and the standard is "pwm{n}"

At quick glance I see it used in the following places:
liquidctl/driver/smart_device.py:552
liquidctl/driver/kraken3.py:248
liquidctl/driver/kraken3.py:257
liquidctl/driver/kraken3.py:261

---

Additional information about the system:

- I'm running the 5.17 kernel so the kernel driver for my nzxtsmart2 device is loaded. I have not loaded the kernel driver for my kraken3 device yet, so that is using the normal driver.
- Affected device(s): NZXT RGB & Fan Controller, NZXT Kraken X (X53, X63 or X73)
- Operating system and version: Arch Linux
- Liquidctl installation method: tested with both liquidctl-git AUR package and python package liquidctl == 1.9.0
- Output of `liquidctl --version`: liquidctl v1.10.0.dev2+g3271625 (Linux-5.17.1-zen1-1-zen-x86_64-with-glibc2.35)

jonasmalacofilho, 2022-04-05:
Yeah, I messed up there.  And thanks for the PR to fix it, taking a look at it now...

jonasmalacofilho, 2022-04-05:
Will make a patch release with this fix later today or tomorrow.

In the meantime (especially for others landing here), adding `--direct-access` should work around the problem.

closed 2022-04-05 by jonasmalacofilho

jonasmalacofilho, 2022-04-05:
For 1.10.0 I want to start using hwmon for setting fan/pump speeds as well.

commit a7250b65 (2022-04-05): hwmon: remove _input suffix from pwmN attributes (cont.)

Closes: #446
Fixes: #445