liquidctl/liquidctl #448: Corsair Comander Core

debian-user-france1, 2022-04-11:
The RGB lights flash _once_ per request (eg. `liquidctl status`).

Additional information about the system:

- Operating system and version: Manjaro Linux
- Liquidctl installation method: yay
- version: 1.8.1

jonasmalacofilho, 2022-04-11:
Cc: @ParkerMc

jonasmalacofilho, 2022-04-11:
I'm pretty sure this is a known problem, but it seems we still didn't have an issue open for it.  Thanks for reporting it.

ParkerMc, 2022-04-11:
That is correct. It is unknown at the moment if it fixable, but it wouldn't hurt to keep an issue open for now to track it.
Quick background, it happens when we "wake up" the controller, that puts it in software mode and somewhere in that it turns off the leds that get turned back on when it falls back to hardware mode.
It is possible that the issue will go away for all but the first time if we copy the hardware mode RGB to the software RGB, but it will likely restart the animation.

ParkerMc, 2022-06-01:
There is currently no path to fixing this as liquidctl only supports hardware mode.

jonasmalacofilho, 2022-06-01:
On Linux, specifically, and _if_ someone writes the kernel driver for it, a hwmon driver for this device could be a solution: it would both keep device in software mode, as well as already provide status information in a convenient form.

(Liquidctl can be made to fetch info from hwmon, when thats's available... we're already doing that on other devices that have those kernel drivers).

jonasmalacofilho, 2022-07-03:
Triage: closing as, like ParkerMc said, there's currently no path to fixing this _in liquidctl._

closed 2022-07-03 by jonasmalacofilho

Viktini, 2023-07-12:
Would this affect software that puts the lights in software mode? Apparently [OpenRGB supports the Commander Core controllers](https://gitlab.com/CalcProgrammer1/OpenRGB/-/issues/3008) so I thought I'd ask.

arch-user-france1, 2023-07-13:
It would not, I think.

ParkerMc, 2023-12-04:
> Would this affect software that puts the lights in software mode? Apparently [OpenRGB supports the Commander Core controllers](https://gitlab.com/CalcProgrammer1/OpenRGB/-/issues/3008) so I thought I'd ask.

I also implemented the OpenRGB controller. And no it will not. But the OpenRGB controller has to always be running.(I also wrote that one)

matteodevenuto, 2026-10-12:
For anyone who finds this later: a fix inside liquidctl isn't possible, but there is a workaround (details in #753).

Commander CORE firmware 2.x refuses sensor reads in hardware mode. In software mode its LEDs only show what the host sends. Once the host goes quiet, the controller returns to hardware mode on its own within about 30 s, and the hardware effect restarts. That restart is the flash, so it happens even if liquidctl skips the SLEEP.

What works is a long-running process that:
- wakes the controller once
- keeps writing LED frames (endpoint `0x22`, OpenRGB's format) every few seconds as a keepalive
- serves telemetry from the same session

OpenRGB does the lighting side of this, as ParkerMc mentioned above. Mine also caches status for liquidctl-style consumers: https://github.com/matteodevenuto/omarchy-corsair-cooling.