liquidctl/liquidctl #449: NZXT H1 V2

thecosmonad, 2022-04-11:
I've recently upgraded my NZXT H1 to the V2 revision due airflow concerns and sadly discovered that the AIO cooling solution is now bound to an internal smart hub instead of the usual fan/pump motherboard headers. The hub is connected via a USB 2.0 header an powered using a SATA connector. It does have three headers, two for the fans (a 140 mm rad fan and a 92 mm GPU exhaust one) and one for the AIO pump. Tested it using NZXT CAM and apparently only fans speeds can be be changed, the pump is greyed out. Instead readings for both fans and pump speeds are available. I've tried sniffing the USB traffic using a Windows VM but didn't get a lot out of it: opening CAM spits out some initialization stuff but then there was nothing I could do in the interface. The various options did not apply and no USB traffic was sent at all... I think CAM went nuts due to both CPU and GPU temps being unknown in the VM. I might try installing Windows on bare metal later.

Anyway, I did a quick guess and added the device to the Smart Hub V2s as
```
(0x1e71, 0x2015, None, 'NZXT H1 V2', {
     'speed_channel_count': 2,
     'color_channel_count': 0
})
```
then prevented the initialization from parsing LEDs as there are none
```
parsers = {b'\x11\x01': parse_firm_info}
if self.product_id not in [0x2015]:
    parsers[b'\x21\x03']: parse_led_info
self._read_until(parsers)
```
Seems to be working fine, it does initialize and I can write/read both fan speeds. This is just a quick PoC, it should probably have its own implementation as it is a mix of both a Smart Hub V2 and an AIO cooler (a rebranded Kraken I guess?).
```
$ liquidctl initialize
NZXT H1 V2
└── Firmware version    1.10.0

$ liquidctl status
NZXT H1 V2
├── Fan 1 control mode     PWM  
├── Fan 1 duty              60  %
├── Fan 1 speed           1282  rpm
├── Fan 2 control mode     PWM  
├── Fan 2 duty              60  %
└── Fan 2 speed           1293  rpm
```

Device type: Smart hub with AIO and fan headers
Product page: https://nzxt.com/product/h1
First-party software: NZXT CAM
Other commonly used tools: None

The device appears to support the following features:

 - monitoring of temperatures or other environment sensors: Maybe?
 - monitoring of fan or pump speeds or duty cycles: Yes
 - monitoring of voltages, currents or power: No
 - configurable fan or pump speeds: Yes
 - configurable voltages or current/power limits: No
 - configurable lighting of embedded LEDs: No
 - configurable lighting of accessories like RGB fans or LED strips: No

The device is physically connected to the host system using:

- USB: No
- PCI-E: No
- onboard the motherboard: USB 2.0 header
- other: No

The device uses which protocols to communicate with the host system:

- HID: No
- USB: Yes
- I²C: No
- SMBus: No
- other: No

```
$ lsusb -d 1e71:2015 -v
Bus 001 Device 004: ID 1e71:2015 NZXT NZXT H1 V2
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x2015 
  bcdDevice            1.0a
  iManufacturer           1 NZXT, Inc.
  iProduct                2 NZXT H1 V2
  iSerial                 3 0D022013ABABBC855F8F1894
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     236
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      0
Device Status:     0x0001
  Self Powered
```

Finally, I can help with:

- testing changes on Linux: Yes
- testing changes on Windows: Yes
- testing changes on Mac OS: No
- attempting to capture USB traffic: Yes<sup>1</sup>
- attempting to capture I²C/SMBus traffic: Yes<sup>1</sup>
- analyzing traffic data: Yes<sup>1</sup>
- documenting the protocol: Yes
- implementing the driver: Yes

<sup>1</sup>assuming documentation and/or assistance will be provided.

jonasmalacofilho, 2022-04-11:
Hi,

> Tested it using NZXT CAM and apparently only fans speeds can be be changed, the pump is greyed out.

It's odd that the pump is grayed out instead of just not listed.  Does NZXT has anything to say about that?

> Anyway, I did a quick guess and added the device to the Smart Hub V2s [...] Seems to be working fine, it does initialize and I can write/read both fan speeds. This is just a quick PoC, it should probably have its own implementation as it is a mix of both a Smart Hub V2 and an AIO cooler (a rebranded Kraken I guess?).

if the protocol is indeed the same (just with no LEDs to control), I think just changing the `SmartDevice2` driver would still be the better option.

Our class hierarchy is mostly about protocols, and a small conditional seems better than duplicating the rest of the initialization method; I would only make the conditional depend on something more abstract, like the absence of LED channels, instead of a specific product ID.

thecosmonad, 2022-04-11:
Hi Jonas,

> It's odd that the pump is grayed out instead of just not listed. Does NZXT has anything to say about that?

Definitely. I'll try it on bare metal first, could be related to CAM misbehaving within the VM.

> I would only make the conditional depend on something more abstract, like the absence of LED channels, instead of a specific product ID.

Got it, I'll keep you updated. Thanks for the help!

thecosmonad, 2022-04-13:
Hi @jonasmalacofilho,

I've managed to get the pump speed reading to work, it's basically the same as a Kraken one.
While reading the device two messages are received, one being the same as the usual Smart Device V2 and another one being a Kraken-like one missing the liquid temperature readings.
I do confirm that there's no way to change the pump speed in CAM running on bare metal. Due to the case size constraints NZXT must have opted to disable such option to prevent overheating.

Anyway, here's the two messages:
```
Fans (\x67\x02):
[103, 2, 19, 32, 2, 13, 133, 188, 171, 171, 148, 24, 143, 95, 2, 255, 2, 2, 0, 0, 0, 0, 0, 0, 208, 4, 21, 5, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 60, 60, 0, 0, 0, 0, 0, 0, 60, 60, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]

Pump (\x75\x02):
[117, 2, 19, 32, 2, 13, 133, 188, 171, 171, 148, 24, 143, 95, 1, 0, 0, 186, 15, 0, 50, 2, 2, 233, 4, 60, 60, 2, 21, 5, 60, 60, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 5]
```
The pump speed lies within `msg[17:18]` and can be extracted using `msg[18] << 8 | msg[17]`.

I do get these results:
```
$ liquidctl status
NZXT H1 V2
├── Fan 1 control mode     PWM  
├── Fan 1 duty              25  %
├── Fan 1 speed            662  rpm
├── Fan 2 control mode     PWM  
├── Fan 2 duty              30  %
├── Fan 2 speed            750  rpm
└── Pump speed            4026  rpm

```
Which are coherent with what CAM reports on bare metal (I've mimicked the silent profile in liquidctl)
[image]

The solution I've opted for is an extension of the SmartDevice2 class, as follows:
```
class H1V2(SmartDevice2):
    SUPPORTED_DEVICES = [
        (0x1e71, 0x2015, None, 'NZXT H1 V2', {
            'speed_channel_count': 2,
            'color_channel_count': 0
        }),
    ]

    def initialize(self, direct_access=False, **kwargs):
        self.device.clear_enqueued_reports()

        # initialize fan reporting
        if self._speed_channels:
            if self._hwmon and not direct_access:
                _LOGGER.info('bound to %s kernel driver, assuming it is already initialized',
                             self._hwmon.module)
            else:
                if self._hwmon:
                    _LOGGER.warning('forcing re-initialization despite %s kernel driver',
                                    self._hwmon.module)
                update_interval = (lambda secs: 1 + round((secs - .5) / .25))(.5)  # see issue #128
                self._write([0x60, 0x02, 0x01, 0xe8, update_interval, 0x01, 0xe8, update_interval])
                self._write([0x60, 0x03])

        # request static firmware info
        self._write([0x10, 0x01])
        ret = []

        def parse_firm_info(msg):
            fw = f'{msg[0x11]}.{msg[0x12]}.{msg[0x13]}'
            ret.append(('Firmware version', fw, ''))

        self._read_until({b'\x11\x01': parse_firm_info})
        return sorted(ret)

    def _get_status_directly(self):
        ret = []

        def parse_fan_info(msg):
            mode_offset = 16
            rpm_offset = 24
            duty_offset = 40
            raw_modes = [None, 'DC', 'PWM']

            for i, _ in enumerate(self._speed_channels):
                mode = raw_modes[msg[mode_offset + i]]
                ret.append((f'Fan {i + 1} speed', msg[rpm_offset + 1] << 8 | msg[rpm_offset], 'rpm'))
                ret.append((f'Fan {i + 1} duty', msg[duty_offset + i], '%'))
                ret.append((f'Fan {i + 1} control mode', mode, ''))
                rpm_offset += 2

        def parse_pump_info(msg):
            pump_offset = 18
            ret.append(('Pump speed', msg[pump_offset] << 8 | msg[pump_offset - 1], 'rpm'))

        self.device.clear_enqueued_reports()
        self._read_until({b'\x75\x02': parse_pump_info, b'\x67\x02': parse_fan_info})
        return sorted(ret)
```
I've skipped all LEDs related stuff as there are none and added the pump extras. Don't know about hwmon though.
What do you think? Any suggestions?

thecosmonad, 2022-04-13:
I did quite a lot of edits to my previous comment, I'm sorry if you may have received many notifications, hope I haven't caused too much confusion.

jonasmalacofilho, 2022-04-13:
That's fine, I only managed to look at it now, after all the edits : )

But can you put it in a pull request so it's easier for me to review and track any additional changes that we decide to do?

Regarding hwmon, we should probably wait until the kernel driver is changed to support this device, because I can see that happening just as easily with the pump as `fan1` or `fanN+1`.  And I don't think we need to resort to grepping the label for "pump" if such kernel driver doesn't even exist yet.

One final note: ~4000 rpm seems quite high for these pumps.  I'm tempted to think that it's the classic 8 vs. 4 poles issue that also makes bare bones pumps report duplicated speeds on normal motherboard _fan_ headers.  At the same time, I would have expected NZXT to compensate for it in CAM.  Can you reach out to their support and ask if ~4000 rpm pump speed is really accurate?

thecosmonad, 2022-04-13:
Hi Jonas, got the pull request ready as a draft. I'll get in touch with NZXT asap for the pump speed, CAM not compensating for that issue it's weird indeed.

> Regarding hwmon, we should probably wait until the kernel driver is changed to support this device, because I can see that happening just as easily with the pump as fan1 or fanN+1. And I don't think we need to resort to grepping the label for "pump" if such kernel driver doesn't even exist yet.
I do agree 100%.

thecosmonad, 2022-04-14:
Hi @jonasmalacofilho,
NZXT replied and told me that's the expected pump's behaviour. So apparentely 4K rpm is fine ¯\\\_(ツ)\_/¯

jonasmalacofilho, 2022-04-15:
@thecosmonad,

> NZXT replied and told me that's the expected pump's behaviour. So apparentely 4K rpm is fine ¯\_(ツ)_/¯

Well, ok then...  : )

---

@amezin,

Pinging you since you might be interested to know that this device shares behavior with the Smart Device V2 controllers, and so may want to add support for this device in `nzxt-smart2`.

thecosmonad, 2022-04-15:
Hi @jonasmalacofilho,

Pull request is ready if you'd like to check it out.

I discovered that there was no need to parse both messages 0x67 and 0x75, the latter one already contains all the information about the fans speed, duty and control mode, just in different places.

Given the message
```
\x75\x02:
[117, 2, 19, 32, 2, 13, 133, 188, 171, 171, 148, 24, 143, 95, 1, 0, 0, 160, 15, 0, 50, 2, 2, 132, 2, 30, 30, 2, 249, 6, 100, 100, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 5]
```
Fan 1 control mode: `msg[21]`
Fan 1 speed: `msg[24] << 8 | msg[23]`
Fan 1 duty: `msg[25]`
Fan 2 control mode: `msg[22]`
Fan 2 speed: `msg[29] << 8 | msg[28]`
Fan 2 duty: `msg[30]`
Pump speed: `msg[18] << 8 | msg[17]`

This will result in:
```
NZXT H1 V2
├── Fan 1 control mode     PWM  
├── Fan 1 duty              30  %
├── Fan 1 speed            644  rpm
├── Fan 2 control mode     PWM  
├── Fan 2 duty             100  %
├── Fan 2 speed           1785  rpm
└── Pump speed            4000  rpm
```
Both duty and rpm offsets are 5 bytes apart instead of the 2 used by the Smart Device V2.

jonasmalacofilho, 2022-04-17:
@thecosmonad,

> Pull request is ready if you'd like to check it out.

Thanks!  I haven't been able to take a look yet, but I'll try to get to it soon!

PR #451 (2022-05-05): Support for NZXT H1 V2 case Smart Device
This adds support for the smart device that is currently supplied within the new NZXT H1 V2 case. It does behave the same as a Smart Device V2 with the addition of a pump reading for the included AIO minus LED controls. I've managed to get it working by extending the `smart_device` driver.

Fixes: #449 
Closes:
Related:

---

Checklist:

- [x] Adhere to the [development process]
- [x] Conform to the [style guide]
- [x] Verify that the changes work as expected on real hardware
- [x] Add automated tests cases
- [x] Verify that all (other) automated tests  [... cut]
closed 2022-05-05 by jonasmalacofilho