liquidctl/liquidctl #457: Hydro H115i Pro Error: USBError(16, 'Resource Busy")

theamazingnorad, 2022-05-09:
Trying to get my Hydro H115i Pro working on an Arch based distro and I think there is an issue, possibly with Python 3.10 and/or one of the USB libraries based on some initial searching.  TLDR: It throws an USB Busy error when I attempt to initialize the device. I am only about six months into my Linux journey, and fairly new to Git, so bear with me on this...I am not a Python programmer so I am a little lost where to troubleshoot from here. 

```
$ liquidctl list -v
Device #0: Corsair RM850i
├── Vendor ID: 0x1b1c
├── Product ID: 0x1c0c
├── Release number: 0x0002
├── Bus: hid
├── Address: /dev/hidraw1
└── Driver: CorsairHidPsu

Device #1: Corsair Lighting Node Core
├── Vendor ID: 0x1b1c
├── Product ID: 0x0c1a
├── Release number: 0x0009
├── Serial number: 3780030A601C75AF3ED3D65E071C00F5
├── Bus: hid
├── Address: /dev/hidraw2
└── Driver: CommanderPro

Device #2: Corsair Commander Pro
├── Vendor ID: 0x1b1c
├── Product ID: 0x0c10
├── Release number: 0x0009
├── Serial number: 0105000293060317
├── Bus: hid
├── Address: /dev/hidraw0
└── Driver: CommanderPro

Device #3: Corsair Hydro H115i Pro
├── Vendor ID: 0x1b1c
├── Product ID: 0x0c13
├── Release number: 0x0100
├── Serial number: 7289_2.0
├── Bus: usb1
├── Address: 6
├── Port: 2.2
└── Driver: HydroPro
```
 And this is the error I get when I try to initialize the device. Sudo does not matter and the udev rules appear to be installed (under /usr/lib/udev).  I tried re-installing Python, and the dependencies for manual building listed on the main page, with no luck;  I admit, however, I might have missed one since I had to translate the names to the AUR. 

'''
$ liquidctl -n 3 initialize --debug
[DEBUG] (cli) (main): liquidctl: 1.9.1
[DEBUG] (cli) (main): platform: Linux-5.17.5-zen1-1-zen-x86_64-with-glibc2.35
[DEBUG] (cli) (_log_requirements): python: 3.10.4 (main, Mar 23 2022, 23:05:40) [GCC 11.2.0]
[DEBUG] (cli) (_log_requirements): colorlog: 6.5.0
[DEBUG] (cli) (_log_requirements): docopt: 0.6.2
[DEBUG] (cli) (_log_requirements): hidapi: 0.11.2
[DEBUG] (cli) (_log_requirements): pyusb: 1.2.1.post37+g3678fc1
[DEBUG] (cli) (_log_requirements): smbus: 1.1
[DEBUG] (cli) (_log_requirements): libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: CommanderCore, CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 1b1c:1c0c (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (__init__): has kernel driver: corsair_psu (/sys/class/hidraw/hidraw1/device/hwmon/hwmon5)
[DEBUG] (usb) (probe): found CorsairHidPsu: Corsair RM850i
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1a (usage_page=0xffc0 usage=0x0001)
[DEBUG] (usb) (probe): found CommanderPro: Corsair Lighting Node Core
[... 89 lines cut ...]
  File "/usr/lib/python3.10/site-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python3.10/site-packages/usb/core.py", line 234, in setup_request
    self.managed_claim_interface(device, intf)
  File "/usr/lib/python3.10/site-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python3.10/site-packages/usb/core.py", line 180, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/usr/lib/python3.10/site-packages/usb/backend/libusb1.py", line 833, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/usr/lib/python3.10/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 16] Resource busy
[ERROR] (cli) (log): unexpected OS error with Corsair Hydro H115i Pro: USBError(16, 'Resource busy')

'''

Additional information about the system:

- Affected device(s): Corsair Hydro H115i Pro
- Operating system and version:  EndeavourOS
- Liquidctl installation method: official Arch Linux repository 
- Output of `liquidctl --version`: liquidctl v1.9.1 (Linux-5.17.5-zen1-1-zen-x86_64-with-glibc2.35)

Thanks for the help!

jonasmalacofilho, 2022-05-18:
At a glance, it seems that something else has claimed the relevant USB interface ahead of liquidctl.

Can you enable PyUSB and LibUSB debugging as well?

```
$ PYUSB_DEBUG=debug LIBUSB_DEBUG=4 liquidctl -n 3 initialize --debug
```

(You can also manually `export` each of those variables).

Also, what kernel drivers are attached to that interface?  A simple way to check it is with lsusb:

```
$ lsusb -vvt
```

closed 2022-05-28 by jonasmalacofilho

Nikita-Portnoy, 2023-05-18:
Hello @jonasmalacofilho,

Not sure if this is still relevant or not. I am running into the same issue in Arch with H150i Pro. Here is the initialize debug output:
```
[INFO] (cli) (log): detailed error: Corsair Hydro H150i Pro: unexpected OS error: USBError(16, 'Resource busy')
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/liquidctl/cli.py", line 463, in main
    status = dev.initialize(**opts)
  File "/usr/lib/python3.10/site-packages/liquidctl/driver/asetek_pro.py", line 109, in initialize
    self._post([_CMD_WRITE_PUMP_MODE, _PUMP_MODES.index(pump_mode)], read_length=5)
  File "/usr/lib/python3.10/site-packages/liquidctl/driver/asetek_pro.py", line 98, in _post
    self.device.write(_WRITE_ENDPOINT, data)
  File "/usr/lib/python3.10/site-packages/liquidctl/driver/usb.py", line 326, in write
    return self.usbdev.write(endpoint, data, timeout=timeout)
  File "/usr/lib/python3.10/site-packages/usb/core.py", line 986, in write
    intf, ep = self._ctx.setup_request(self, endpoint)
  File "/usr/lib/python3.10/site-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python3.10/site-packages/usb/core.py", line 229, in setup_request
    self.managed_claim_interface(device, intf)
  File "/usr/lib/python3.10/site-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python3.10/site-packages/usb/core.py", line 178, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/usr/lib/python3.10/site-packages/usb/backend/libusb1.py", line 829, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/usr/lib/python3.10/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 16] Resource busy
[ERROR] (cli) (log): Corsair Hydro H150i Pro: unexpected OS error: USBError(16, 'Resource busy')
```
Here are the attached kernel drivers:
```
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    /sys/bus/usb/devices/usb6  /dev/bus/usb/006/001
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    /sys/bus/usb/devices/usb5  /dev/bus/usb/005/001
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    /sys/bus/usb/devices/usb4  /dev/bus/usb/004/001
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    /sys/bus/usb/devices/usb3  /dev/bus/usb/003/001
    |__ Port 7: Dev 2, If 0, Class=Wireless, Driver=btusb, 12M
        ID 0b05:185c ASUSTek Computer, Inc. 
        /sys/bus/usb/devices/3-7  /dev/bus/usb/003/002
    |__ Port 7: Dev 2, If 1, Class=Wireless, Driver=btusb, 12M
        ID 0b05:185c ASUSTek Computer, Inc. 
        /sys/bus/usb/devices/3-7  /dev/bus/usb/003/002
    |__ Port 8: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 0b05:1872 ASUSTek Computer, Inc. 
        /sys/bus/usb/devices/3-8  /dev/bus/usb/003/003
    |__ Port 9: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 05e3:0608 Genesys Logic, Inc. Hub
        /sys/bus/usb/devices/3-9  /dev/bus/usb/003/004
        |__ Port 2: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            ID 1b1c:0c0b Corsair Lighting Node Pro
            /sys/bus/usb/devices/3-9.2  /dev/bus/usb/003/006
    |__ Port 10: Dev 5, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 05e3:0608 Genesys Logic, Inc. Hub
        /sys/bus/usb/devices/3-10  /dev/bus/usb/003/005
        |__ Port 2: Dev 8, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            ID 1b1c:0c1a Corsair 
            /sys/bus/usb/devices/3-10.2  /dev/bus/usb/003/008
        |__ Port 4: Dev 10, If 0, Class=Vendor Specific Class, Driver=usbfs, 12M
            ID 1b1c:0c12 Corsair 
            /sys/bus/usb/devices/3-10.4  /dev/bus/usb/003/010
    |__ Port 13: Dev 7, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 1b1c:1b34 Corsair 
        /sys/bus/usb/devices/3-13  /dev/bus/usb/003/007
    |__ Port 13: Dev 7, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        ID 1b1c:1b34 Corsair 
        /sys/bus/usb/devices/3-13  /dev/bus/usb/003/007
    |__ Port 14: Dev 9, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 1b1c:1b49 Corsair 
        /sys/bus/usb/devices/3-14  /dev/bus/usb/003/009
    |__ Port 14: Dev 9, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        ID 1b1c:1b49 Corsair 
        /sys/bus/usb/devices/3-14  /dev/bus/usb/003/009
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    /sys/bus/usb/devices/usb2  /dev/bus/usb/002/001
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    /sys/bus/usb/devices/usb1  /dev/bus/usb/001/001
```

Any help with this is appreciated. Let me know if I can provide any other information.

Thank you!