liquidctl/liquidctl #458: Channel name consistency

jmarucha, 2022-05-17:
Currently I work with the scripts using liquidctl CLI and I notice each driver processes the channel names differently. They also behave differently when all of them.

It may be beneficial to decide on implementing one particular behavior for these things, as the logic in most of them perform is similar (tho differently coded) and can be refactored into other class.

I provide the table with current state of code
| Driver               | Function               | `fan`  | `fans` | `sync` | `fanN` | `pump` | Exception                                                           |
|----------------------|------------------------|--------|--------|--------|--------|--------|---------------------------------------------------------------------|
| `asetek.py`          | python map             | Yes    | No     | No     | No     | Yes    | KeyError                                                            |
| `asetek_pro.py`      | `_fan_indexes`         | All    | No     | No     | Yes    | No*    | `NotSupportedByDevice`, `ValueError(f'unknown channel: {channel}')` |
| `commander_core.py`  | `_parse_channels`      | No     | All    | No     | Yes    | Yes    | `ValueError(f'unknown channel, should be one of {CHANNELS}')`       |
| `commander_pro.py`   | `_get_hw_fan_channels` | No     | No     | All    | Yes    | No     | `ValueError` as above, `ValueError` if no fans are there            |
| `corsair_hid_psu.py` |                        |        |        |        |        |        | Free for all, `channel` isn't parsed                                |
| `hydro_platinium.py` | `_get_hw_fan_channels` | All    | No     | No     | Yes    | No     | `ValueError` as above                                               |
| `kraken2.py`         | python map             | Yes    | No     | No     | No     | Yes    | KeyError                                                            |
| `kraken3.py`         | python map             | Yes/No | No     | No     | No     | Yes    | KeyError                                                            |
| `smart_device.py`    | python map             | No     | No     | All    | Yes    | No     | KeyError                                                            |

jonasmalacofilho, 2022-05-18:
Related: #315

jmarucha, 2022-05-18:
I'm glad more people recognize the issue.

If you asked me about my personal preference, I'd go for:
- `fans` being alias to all fans of particular device.
- `fan` being applicable to one fan, with multiple fans defaulting to `fans` behavior but with warning
- `fanN` referencing particular fan (numbering starts from one)
- `fan1` being alias of `fan` for single fan channel
- `sync` available but depreciated and removed in next major release (in accordance with Semantic versioning). The name only makes sense for RGB channels

About errors:
- If fan channel or pump channel not supported at all return `NotSupportedByDevice`
- If `fanN` is specified correctly, but fan number N does not exist, `ValueError(f'unknown channel, should be one of {CHANNELS}')`
- Value error as above in all other cases (and channel name always validated, not like in corsair hydro).

About implementation:
- Add to each driver class a Dict describing available channels, something like
```python
available_speed_channels = {
    fan: {
       count: 4, # if not set, defaults to 1
       supported_modes: ["fixed", "profile"]
    },
    pump: {
        supported_modes: ["fixed"]
} 
```
- Add a function for parsing speed channels, probably in `util.py` in some function
```python
def parse_channels(channel_string, available_speed_channels, mode="any"):
    ...
    return (channel_name, fan_numbers) # like return ("fan", [1]) or ("pump",) or ("fan", [0,1,2,3])
```
The good side effect is that immediately one can write a cli function that lists supported speed channels.
```sh
# liquidctl speed channels
NZXT Kraken X (X42, X52, X62 or X72)
├── fans         (modes: "fixed", "profile")
｜       ├── fan1
｜        └── fan2
└── pump      (modes: "fixed")
```
If the team likes my idea, I can go on and start writing.

jonasmalacofilho, 2022-05-27:
First, thanks for looking into this!

Please wait for @MarshallAsch to also give his thoughts on this, before starting your implementation, but here are my comments.

---

_Regarding the channel names..._

> * `fans` being alias to all fans of particular device.

Assuming fan _channels,_ agreed. 

That's probably obvious to us, but some devices come with 2 or 3 fans but a single fan channel, and some users find that surprising.  So when applicable our UI and docs should be clear that we're talking about fan _channels,_ not actual fans.

> * `fan` being applicable to one fan, with multiple fans defaulting to `fans` behavior but with warning

I'm not sure I understand what you mean by "multiple fans defaulting to `fans`".  Can you give me an example?

> * `fanN` referencing particular fan (numbering starts from one)

Agreed.

> * `fan1` being alias of `fan` for single fan channel

Agreed.

> * `sync` available but depreciated and removed in next major release (in accordance with Semantic versioning). The name only makes sense for RGB channels

Agreed that it doesn't make sense for cooling channels.

On removal × deprecation, please take a look at our [stability guarantees](https://github.com/liquidctl/liquidctl/blob/main/docs/developer/process.md#stability-and-backward-compatibility).  TL;DR: let's not mention that it _will_ be removed in v2, because we don't know that yet; also, we don't have concrete plans for a v2 either.  But yes, deprecate it in favor of a better alternative.

On that subject, is some instances we may want something like `all`, for example, if pumps are also involved.  This isn't implemented yet, but on Kraken X2s it should be possible to set both `fan` and `pump` at the same time.  And there might be some device that already does something like that today with `sync`...

---

_Regarding the errors..._

> * If fan channel or pump channel not supported at all return `NotSupportedByDevice`

Agreed.

> * If `fanN` is specified correctly, but fan number N does not exist, `ValueError(f'unknown channel, should be one of {CHANNELS}')`

Agreed.
 
> * Value error as above in all other cases

Agreed.

> (and channel name always validated, not like in corsair hydro).

I'm not sure which specific Hydro driver you're talking about, but I'm guessing it's when there's a single channel.  In that case, our approach has been to automatically fix correctable user errors.

I guess a warning might be ok in these scenarios... but I'm divided, as we would also annoy some users for no good reason.

(If, however, the driver you're talking about does something beyond accepting any name for a single [and thus unambiguous] cooling channel, then that driver needs to be fixed).

---

_Regarding the implementation..._

> * Add to each driver class a Dict describing available channels, something like

Makes sense.

> * Add a function for parsing speed channels, probably in `util.py` in some function

Same.

> The good side effect is that immediately one can write a cli function that lists supported speed channels.

Yes, but let's not conflate that with the other changes, as it would be a public-facing addition, and I'm not sure it covers all of our future needs.  And once it's in there, we wouldn't be able remove or drastically change it, due to our stability guarantees.

But you can add something like it in a separate PR, either in `--debug` mode or behind an `--unsafe` feature gate.  Please also clearly mark it as experimental/not to be depended on (yet) by downstream projects.

jmarucha, 2022-05-28:
>I'm not sure I understand what you mean by "multiple fans defaulting to fans". Can you give me an example?

You running a device with *two* fan channels, what should happen when you type `set fan speed 100`?
- Throw an error, better set nothing than all fans (breaks stability guarantee)
- Throw a warning that there are multiple fans, apply setting to both fan channels (my favorite)
- Set all fans to 100%, without saying anything
it's about the choice i guess

jonasmalacofilho, 2022-05-28:
Oh, I see...

While I see the benefit of catching potential errors by issuing a warning on `fan` when there are multiple channels, I also think `set fan speed` reads more naturally than `set fans speed`.  So applying the setting to all fan channels without issuing a warning is also an alternative to be considered.

But the difference between both alternatives is tiny, and we can experiment with it.  If the warning is (still) your preferred choice, I'm ok with trying it out.

---

P.S. Setting all fans to 100% would also be a breaking change.

MarshallAsch, 2022-05-29:
I agree with all of this (I'm currently on my phone so I can't see what I'm responding too while I'm writing so sorry about that). I would suggest this gets split into 3 separate tasks to avoid doing too much at once.

1. Rename / mark fan names as deprecated in favor of the consistent naming 
2. Add the channel names to the driver configuration block
3. Add the api to get the channel names ( mark this as experimental / internal then maybe make it stable