liquidctl/liquidctl #461: Corsair H150i Pro Not Detected

PatrickGHall, 2022-05-31:
I have a refurbished Corsair H150i Pro, which I suspect is part of my problem here. In any case, liquidctl can't see the device and it has no driver loaded. Attempting to manually bind the `usbhid` driver gives me an error "No such device"

I have also tried adding the following to the provided `71-liquidctl.rules` to try and force the OS to do it: 
```
# Corsair Hydro H150i Pro
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c12", TAG+="uaccess", PROGRAM="/bin/sh -c 'echo -n $id:1.0 >/sys/bus/usb/drivers/usbhid/bind;\
echo -n $id:1.1 >/sys/bus/usb/drivers/usbhid/bind'"
```

but I haven't had any luck. 

Debugging with `lsusb` I think I might have a clue as to what is going on, but it might not be relevant. The device shows the correct vendor and product IDs, but calls itself a "Platinum" rather than a "Pro" 
The relevant output from `$ lsusb -v`:
```
Bus 003 Device 006: ID 1b1c:0c12 Corsair H150i Platinum
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c12 
  bcdDevice            1.00
  iManufacturer           1 Corsair
  iProduct                2 H150i Platinum
  iSerial                 3 7289_2.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower               64mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
[... 104 lines cut ...]
├── Fan 5 control mode     PWM  
└── Fan 6 control mode     PWM  

[DEBUG] (cli) (main): device: Corsair Hydro H150i Pro
[DEBUG] (usb) (open): selected interface: 0
[DEBUG] (asetek) (_configure_flow_control): set clear to send = True
[DEBUG] (usb) (ctrl_transfer): sending control transfer with (64, 2, 2), {}
[DEBUG] (usb) (write): writting 2 bytes: 32:01
[DEBUG] (usb) (read): read 3 bytes: 32:12:34
[DEBUG] (usb) (release): explicitly release interface
[DEBUG] (usb) (release): explicitly release interface
```

---

Additional information about the system:

- Affected device(s): Corsair H150i Pro
- Operating system and version: Ubuntu 22.04 LTS
- Liquidctl installation method: installed through pip 
- Output of `liquidctl --version`: 
```
$ liquidctl --version
liquidctl v1.9.1 (Linux-5.15.0-33-generic-x86_64-with-glibc2.35)
```<!-- e.g. liquidctl v1.9.0 (Linux-5.16.16-arch1-1-x86_64-with-glibc2.35) -->

PatrickGHall, 2022-05-31:
Also, in case it is relevant, the USB 2.0 header for the H150i is connected to one of the two passthrough ports on the Commander

jonasmalacofilho, 2022-05-31:
> Attempting to manually bind the `usbhid` driver gives me an error "No such device"

Oh, forgot to reply to this part.

Yeah, that shouldn't work since this particular model isn't a HID.  But that's ok, we can still communicate with it using PyUSB/LibUSB.

jonasmalacofilho, 2022-05-31:
On a related note, would

```
[DEBUG] (usb) (probe): HydroPro recognized: Corsair Hydro H150i Pro
```

or 

```
[DEBUG] (usb) (probe): HydroPro identified: Corsair Hydro H150i Pro
```

make it more obvious that from that point in the execution onward liquidctl  knows about that device and will (try to) handle it appropriately?

(Also pinging @MarshallAsch for his opinion on this).

MarshallAsch, 2022-05-31:
> (Also pinging @MarshallAsch for his opinion on this).

I like option 2. `[DEBUG] (usb) (probe): HydroPro identified: Corsair Hydro H150i Pro`

Ill take a look at the rest of this thread tonight

PatrickGHall, 2022-05-31:
I was confused by it not showing up after the call to `initialize all` and I initially couldn't see it in `list` but after tinkering with things on the `rgb_vengeance` stuff I now see it in `list` and am able to control it. I wish I knew at what point it came alive but there was a reboot in there somewhere. I think it's safe to close this one out as PEBK

While I have you here, can I ask a question about the vengenace RGB (also probably PEBK)? 
I have 4 sticks and I _assume_ they correlate to `/dev/i2c-{0-3}` and given that they appear there, I _assume_ `i2c-dev` is loaded although I can't see it listed in `lsmod` even after attempting `modprobe i2c-dev` which doesn't output any error.
I have myself added to the `i2c` group and did a reboot:

```
$ ll /dev/ | grep i2c
crw-rw----   1 root    i2c      89,     0 May 31 13:15 i2c-0
crw-rw----   1 root    i2c      89,     1 May 31 13:15 i2c-1
crw-rw----   1 root    i2c      89,     2 May 31 13:15 i2c-2
crw-rw----   1 root    i2c      89,     3 May 31 13:15 i2c-3
```
```
$ id `whoami`
uid=1000(patrick) gid=1000(patrick) groups=1000(patrick),4(adm),24(cdrom),27(sudo),30(dip),46(plugdev),122(lpadmin),134(lxd),135(sambashare),137(i2c)
```
My processor is AMD, but I think the SMBus comment is about the spec itself being written by Intel, not that the architecture must be Intel to use it. When executing the commands to set the colors for the vengeance sticks, I keep getting this:

```
$ liquidctl --match vengeance set led color fixed fa11f2 --unsafe=smbus,vengeance_rgb
ERROR: no device matches available drivers and selection criteria
patrick@cloud2gnd:~$ sudo -E env "PATH=$PATH" liquidctl --match vengeance set led color fixed fa11f2 --unsafe=smbus,vengeance_rgb
[sudo] password for patrick: 
ERROR: no device matches available drivers and selection criteria
```
What might I be missing in the setup?

PatrickGHall, 2022-05-31:
Also I think the weird issue with the device not showing in the list command was because I had previously installed through apt, which was 1.5 and then purged and installed through pip to get the latest. When I rebooted it must have sorted itself

jonasmalacofilho, 2022-06-03:
> My processor is AMD, but I think the SMBus comment is about the spec itself being written by Intel, not that the architecture must be Intel to use it.

Actually, we really are limited right now to Intel platforms, because of some kernel driver differences.  Specifically, we expect to be able to read the both pages of the DDR4 SPD EEPROM through the kernel utilities (to really check that we know how to "talk" to that specific device), but IIRC only the Intel driver sets that up, at least automatically.

> I found some directions on the OpenRGB page for modifying the kernel params which loaded more I2C devices and got the RAM sticks to show up in OpenRGB, but still getting the same error on `liquidctl`

Can you share those?  They could be related to what I just mentioned above.

---

> Yet another unrelated thing now that the 150i Pro is listed, when setting the alert colors it seems to not take the values given at least for the "good" state and defaults to green.

I'm not the author of this driver, but based on what the work I did for the previous generation of Asetek devices, there's a chance that our documentation might be wrong, and that alerts device just don't work that way on the device.  Sorry about that...

Any chance you can investigate this a bit further?  Maybe see how it works on other software, both first party (CorsairLink _and_ iCue) and third party (OpenRGB, OpenCorsairLink, etc.)?

You can also take a look at [our 5-th gen docs](https://github.com/liquidctl/liquidctl/blob/main/docs/asetek-690lc-guide.md#lighting-modes) and see if how your device responds to the alert settings seems to match that better.

PatrickGHall, 2022-06-06:
The OpenRGB kernel parameter instructions are here: https://gitlab.com/CalcProgrammer1/OpenRGB#kernel-parameters

I can take a crack at loading up windows + iCue to see if the alerts are supposed to work for this device

jonasmalacofilho, 2022-07-03:
Triage: @PatrickGHall, have you managed to look into the problem with alert mode?

jonasmalacofilho, 2022-07-07:
It's possible that there are non-trivial interactions between normal and the `alert` modes.  And without having this feature exposed through iCue, it would require some experimentation, including exhaustively testing all imaginable states/combinations...

Are you interested in working on this, since you have the device?

Otherwise I'm tempted to simply remove the mode, since right now it is neither complete nor functional.

PatrickGHall, 2022-07-08:
I can look into it. My pace will undoubtedly be slow. I'll take a look at the repo for the tips on how to get setup for that, I think I remember seeing documentation for it when I was first perusing.

MacGyverNL, 2022-07-24:
As mentioned in the #484, I *think* this may be caused by what seems to be a typo in selecting the temperatures from the 3-tuple, leading to a duplicate temperature being passed as mid- and high-point. I get the feeling the software in these devices simply hasn't been built to cope with that situation.

@PatrickGHall You can try if applying that single edit fixes your colours. Since it's Python, if you're careful about it, you can do so on the system-installed package if you remove the corresponding `asetek_pro.cpython-3*.pyc` cache file.

jonasmalacofilho, 2022-08-11:
Closing probably as fixed by PR #484.  Please provide feedback in case the issue still remains with the latest code in `main`.

closed 2022-08-11 by jonasmalacofilho